esphome/issues #2867: Cant add a Delay to Script without getting an error

DaveTiff, 2021-12-22:
### The problem

Trying to add a delay to a script, place in the place according to 

[https://esphome.io/guides/automations.html?highlight=script](url) delay action

See below my Yamal & Error message

### Which version of ESPHome has the issue?

2021.10.3 

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2021.12.4

### What platform are you using?

ESP8266

### Board

Sonoff POW

### Component causing the issue

Script

### Example YAML snippet

```yaml
script:
  - id: relayoff
    then:
      - delay: 1ms       
      if:
        condition:
          or:
            # If any of the Psudo switches are On (acc1, acc2, acc3)
            - switch.is_on: acc1
            - switch.is_on: acc2
            - switch.is_on: acc3            
        then:
  
        else:
          - switch.turn_off: relay
```

### Anything in the logs that might be useful for us?

```txt
ERROR Error while reading config: Invalid YAML syntax:

while parsing a block collection
  in "/config/esphome/sonoff-pow-aux.yaml", line 119, column 7:
          - delay: 1ms       
          ^
expected <block end>, but found '?'
  in "/config/esphome/sonoff-pow-aux.yaml", line 120, column 7:
          if:
          ^
```

jesserockz, 2021-12-22:
The issue is not the delay, the issue is that your `if` is not indented and needs a `-` to be a part of the action list properly.

```yaml
script:
  - id: relayoff
    then:
      - delay: 1ms       
      - if:
          condition:
            or:
              # If any of the Psudo switches are On (acc1, acc2, acc3)
              - switch.is_on: acc1
              - switch.is_on: acc2
              - switch.is_on: acc3            
          then:
            ...
          else:
            - switch.turn_off: relay
```

DaveTiff, 2021-12-22:
Thank you, It compiled and work fine without the delay entry. So I assume it was that. 

The sample code I found on the web for the if statement did not have the - before the if.

Just tried it and now as per your code and working as expected.

Thank You so much

closed 2021-12-22 by DaveTiff

jesserockz, 2021-12-22:
When you only have a single action (like the `if`) then it doesn't need the `-` to show a list as ESPHome can auto convert a single item to a list, but as soon as you add more actions it is needed.

Glad I could help