esphome/issues #2868: Time zone from Home Assistant wrong (with ESP-IDF)

Bascht74, 2021-12-22:
### The problem

The time zone from HA is reportet UTC0, but I set GTM+1 (Berlin) in HA. 

### Which version of ESPHome has the issue?

2021.12.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2021.12.4

### What platform are you using?

ESP32-IDF

### Board

m5stack-atom

### Component causing the issue

time

### Example YAML snippet

```yaml
external_components:
  - source: github://Jopand/esphome_components@develop
    refresh: 0s
  - source: github://esphome/esphome@dev
    refresh: 0s
    components: [ modbus ]
  - source: github://pr#2916
    components: [ modbus_controller ]

substitutions:
  upper_devicename: Nilan
  devicename: nilan
  device_description: "Niland Modbus connection"
  updates: 30s

esphome:
  name: $devicename
  comment: "${device_description}"

esp32:
  board: m5stack-atom
  framework:
    type: esp-idf
    version: recommended

wifi:
  reboot_timeout: 2min
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  power_save_mode: none

api:
  reboot_timeout: 9999 min
  encryption: 
    key: !secret encryption_key

ota:
  password: !secret ota_password

uart:
  rx_pin: GPIO32   # GPIO22 -> RS485 Kit, GPIO32 -> RS485 Extension
  tx_pin: GPIO26   # GPIO19 -> RS485 Kit, GPIO26 -> RS485 Extension
  parity: EVEN
  baud_rate: 19200
  id: uart_modbus
  stop_bits: 1
  
modbus:
  uart_id: uart_modbus
  id: modbus_id
  send_wait_time: 200ms
  
modbus_controller:
  - id: nilan_modbus_controller
    address: 30
    modbus_id: modbus_id
    command_throttle: 100ms
    setup_priority: -100
    update_interval: 30s
    
time:
  - platform: homeassistant
    id: homeassistant_time  

nilan:
```

### Anything in the logs that might be useful for us?

```txt
[16:15:14][C][homeassistant.time:010]: Home Assistant Time:
[16:15:14][C][homeassistant.time:011]:   Timezone: 'UTC0'
```

### Additional information

This is set in HA: 
[image]

I changed from arduino to ESP-IDF. Before that change I cannot remember a wrong time zone...
Might be helpful or not...

jesserockz, 2021-12-22:
The timezone that you are seeing in the logs comes from the host system timezone via python. I have not looked into how that is linked to the user selected timezone in HA though.

As a quick fix, you can manually set the timezone in the yaml file

Bascht74, 2021-12-26:
@jesserockz 
Yes, I applied this workaround. 

But I tried the following: 

My Docker x86 Add-on log output (because I cannot compile ESP-IDF there): 
```
INFO Reading configuration /config/nilan.yaml...
INFO Updating https://github.com/Jopand/esphome_components.git@develop
INFO Updating https://github.com/esphome/esphome.git@dev
INFO Detected timezone 'Etc/UTC'
INFO Generating C++ source...
INFO Compiling app...
[10:41:03][C][homeassistant.time:010]: Home Assistant Time:
[10:41:03][C][homeassistant.time:011]:   Timezone: 'UTC0'
```

esphome log output HA Add-on  after compiling with x86: 
```
[11:42:50][C][homeassistant.time:010]: Home Assistant Time:
[11:42:50][C][homeassistant.time:011]:   Timezone: 'UTC0'
```

After I changed the timezone of my x86 docker component to `Europe/Berlin` it still is UTC0 in the log...

```
INFO Detected timezone 'Europe/Berlin'
[11:55:37][C][homeassistant.time:010]: Home Assistant Time:
[11:55:37][C][homeassistant.time:011]:   Timezone: 'UTC0' 
```

So it seems that the timezone is "baked in" the compilation file and not automatically inferred on the time zone of the computer that is "running" ESPHome, but this might not always be accurate.

Should the documentation be more clear about that or might it be better to get the timezone as well from the HA instance where ESPhome gets the time from...? 
I think the latter would be more consistent as environment for compilations might have (as mine) a wrong time zone or maybe the device is pre-compiled.

jesserockz, 2021-12-29:
> or might it be better to get the timezone as well from the HA instance where ESPhome gets the time from...?

The addon has no idea and no connection to Home Assistant at all. The time is a different story as that asks the api client for the time, and doesnt actually specifically need to be HA.

The data is loaded using [tzdata](https://pypi.org/project/tzdata/) and [tzlocal](https://pypi.org/project/tzlocal/).

Not sure if you are running the cli inside docker manually, but if you can, then try adding `-v` -> `esphome -v run ....yaml` to get the DEBUG line like below.

```shell
$ esphome -v config example.yaml 
INFO Reading configuration example.yaml...
INFO Detected timezone 'Pacific/Auckland'
DEBUG  -> TZ string NZST-12NZDT,M9.5.0,M4.1.0/3
```

31337Ghost, 2021-12-30:
I have exactly the same problem, for the temporary fix, I changed the platform from `homeassistant` to `sntp`:
```yaml
time:
  - platform: sntp
```

Bascht74, 2021-12-30:
@jesserockz 
I would expected that if I use 

```
time:
  - platform: homeassistant
```

that esphome gets the timezone "live" from the homeassistant time component. 
Right now it seams that it gets the time, but not the timezone. The timezone seems to be backed into the binary (i uses the timezone of the compilation environment) and is not dynamic.

That is why I wrote: 

> or might it be better to get the timezone as well from the HA instance where ESPhome gets the time from...?

I would like that the time component gets the timezone from HA via the api like the sntp seems to get it from the timeserver, if 31337Ghost is right about his workaround with `platform: smtp` ...

nightdemonx, 2022-01-04:
I confirm this problem. My time difference is 10 hours. For the correct time, I had to choose a different time zone. 

```
time:
  - platform: homeassistant
    id: esptime
    timezone: America/Rio_Branco
#timezone: Asia/Yekaterinburg
```
I don't remember when it changed, but initially it worked correctly.

anas-ivs, 2022-01-28:
Having same problem on NSPanel and TTGO Display with time difference of 4 hours. 

```
time:
  - platform: homeassistant
    id: esptime
    timezone: Asia/Baku
    #timezone: Asia/Kuching
```
 
 Like @nightdemonx, did not recall when this started to break but was working and did not need to spesify TZ before (on TTGO).

Bascht74, 2022-04-09:
For me it seems that the timezone is backed in the firmware rather requested by the api via the homeassistant plattform. If I don't provide the timezone in the YAML it uses the "wrong" timezone from the docker image. If I change the timezone, it is using again that timezone, if I see it right. It (=time component) never requests the timezone to use via the homeassistant platform, but I don't know why it is done that way...
If I don't provide a timezone, I want the one to be used where I get my time from. If I provide a time zone, I want that one to be used. I never want to use the timezone of my compiling environment...

jesserockz, 2022-04-20:
The timezone is set in the firmware at compile time based on the system timezone that is running ESPHome (If you do not specify it in yaml)

https://github.com/esphome/esphome/blob/7a778f3f330ca607f2772501eac82a043d0346d3/esphome/components/time/__init__.py#L260-L268

Bascht74, 2022-04-21:
So then „works as designed“ —> feature request.