esphome/issues #2869: "Required key not available" when starting the docker service

wsw70, 2021-12-23:
### The problem

When starting the docker container I get the following error in the log:

```
2021-12-23 06:56:23,994 INFO Starting dashboard web server on http://0.0.0.0:6052 and configuration dir /config...
2021-12-23 06:56:24,002 WARNING Error with socket 10 (('192.168.20.0', 5353))): [Errno 126] Required key not available
Traceback (most recent call last):
  File "/usr/lib/python3.9/asyncio/selector_events.py", line 1056, in sendto
    self._sock.sendto(data, addr)
OSError: [Errno 126] Required key not available
```

This is a non-blocking error (ESPHome runs fine)

### Which version of ESPHome has the issue?

2021.12.1

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

N/A

### What platform are you using?

ESP8266

### Board

N/A

### Component causing the issue

N/A

### Additional information

The IP referenced in the error log (`192.168.20.0`) is not a live IP. I do have a `192.168.20.0/24` network used by Wireguard (`192.168.20.1` is the VPN endpoint and the IPs in that network grow from there) and it has no relationship with the ESPHome installation (ESPHome is in the docker `172.19.0.0/16` network, and docker itself runs on the `192.168.10.2` host (part of `192.168.10.0/24`).

This is to say that I do not know why ESPHome is even attempting to bind to `192.168.20.0`.

It knows about the network because of the docker configuration (`host`) but that's all

```
esphome:
    container_name: esphome
    image: esphome/esphome
    volumes:
      - /etc/docker/data/esphome:/config
    network_mode: host
    restart: unless-stopped
```

OttoWinter, 2022-01-18:
Yeah that's an issue with the zeroconf library (used for mDNS) afaik, and would need to be fixed upstream.

fly3rman, 2022-10-10:
I can confirm the problem.
```
Attaching to esphome
esphome    | 2022-10-10 18:46:43,749 INFO Starting dashboard web server on http://0.0.0.0:6052 and configuration dir /config...
esphome    | 2022-10-10 18:46:43,762 WARNING Error with socket 16 (('10.10.2.11', 5353))): [Errno 126] Required key not available
esphome    | Traceback (most recent call last):
esphome    |   File "/usr/lib/python3.9/asyncio/selector_events.py", line 1056, in sendto
esphome    |     self._sock.sendto(data, addr)
```
with 10.10.2.11 beeing an wireguard interface.

flipreverse, 2023-02-27:
@OttoWinter Any chance there is a workaround for the Home Assistant add-on? Since I'm using HASS, I cannot edit my docker config for that container.

Koli0842, 2025-09-10:
Running it via Podman, still an issue, and the workaround also still works

ssieb, 2025-09-10:
It's just a warning, but doesn't cause any problems.  Is it actually blocking something from working for you?