esphome/issues #2871: H-bridge "Fan" decay_mode not implemented as expected.

FaBjE, 2021-12-23:
### The problem

I'm currently making a new component to control motorized valve and linear actuators using an HBridge.
Since there are already two components (light and fan) containing a (partial) HBridge implementation it seemed like a good idea to me to implement one generic HBridge driver class and let the specific components use it to fulfill there specific application.

On porting the hbridge.fan component I encountered something unexpected.
It has a "decay_mode" which you can set to either slow (braking) or fast (coasting) mode. _[This naming seems reversed to me, but anyway...]_

The coasting mode is as I would expect. but the braking mode applies the A and B input signals as follows:
`set_hbridge_levels_(1.0f, 1.0f - speed)`
This applies a full-duty cycle on A and a partial duty cycle on B.

This causes the motor to be shorted (braked) the first part of the cycle and energized the last part of the cycle (when B is not active). This signal is put on continuously (not only for speed reducing)
In my opinion this is wrong a puts a lot of stress on the motor/components. 

You can either fully brake or gradually brake by having a low duty cycle the same on both A and B and gradually increasing it.
Slowing down til a certain speed is achieved requires some kind of (encoder) feedback signal that is not available in a HBridge alone.

My proposal is to either remove this "decay_mode" parameter as it has no applicable function.

Alternatively we could use it to determine if speed is set to 0 whether to coarse or to (gradually) brake. (not between speed changes) again, we have no feedback if the motor has stopped or not.

Docs on HBridge fan:
https://esphome.io/components/fan/hbridge.html

Source it is about: https://github.com/esphome/esphome/blob/72c6bfaa50e54f58e5dae494bc71040281d4d0a4/esphome/components/hbridge/fan/hbridge_fan.cpp#L67

I will submit a pull request with the new/updated component in the coming future. But this issue required further discussion outside of the pull request.

### Which version of ESPHome has the issue?

2022.1.0-dev

### What type of installation are you using?

Docker

### What platform are you using?

ESP8266

### Board

nodemcu

### Component causing the issue

fan.hbridge

WeekendWarrior1, 2022-02-20:
> On porting the hbridge.fan component I encountered something unexpected.
It has a "decay_mode" which you can set to either slow (braking) or fast (coasting) mode. [This naming seems reversed to me, but anyway...]

[image]
`Slow` and `Fast` decay are standard H-bridge / inductive load terminology that describes how the 'residual' current within a motor circuit is dissipated. 

> The coasting mode is as I would expect. but the braking mode applies the A and B input signals as follows:
set_hbridge_levels_(1.0f, 1.0f - speed)
This applies a full-duty cycle on A and a partial duty cycle on B.

> This causes the motor to be shorted (braked) the first part of the cycle and energized the last part of the cycle (when B is not active). This signal is put on continuously (not only for speed reducing)
In my opinion this is wrong a puts a lot of stress on the motor/components.

No, this is correct and quite common in H-bridge motor control, and can often help provide extra motor torque when PWM speed controlling with a large load on your DC motor..
You will find this discussed in plenty of H-bridge libraries, and H-bridge datasheets.
Here is a good page on some of the benefits of Slow decay switching:
https://learn.adafruit.com/improve-brushed-dc-motor-performance/current-decay-mode

And to show the more linear speed control provided by Slow decay:
[image]

In fact it is usually recommended that if you are new to H-Bridges + DC motors, or you are unsure which decay mode to use, that Slow Decay is the default:
https://learn.adafruit.com/improve-brushed-dc-motor-performance/choosing-decay-mode-and-pwm-frequency

FaBjE, 2022-02-21:
Issue was invalid. This was intended functionality.

Closing...

closed 2022-02-21 by FaBjE