esphome/issues #2872: Nextion Editor Simulator Connection

Londoneye02, 2021-12-24:
### The problem

I try to connect to the Nextion Editor Simulator ("Debug") feature. This is to avoid having to deal with the physical screen (to avoid any damage) and because is just faster to do it this way.

ESPHome Logs shows the following:

```
[12:10:29][W][nextion:072]: Nextion is not connected! 
[12:10:29][E][uart:015]: Reading from UART timed out at byte 0!
```

And Nextion Editor 

[image]

Connecting to a real screen with the exact same tft file works OK.

ESPHome logs show the following

```
[12:14:14][C][nextion:117]: Nextion:
[12:14:14][C][nextion:118]:   Device Model:     NX4832K035_011R
[12:14:14][C][nextion:119]:   Firmware Version: 163
[12:14:14][C][nextion:120]:   Serial Number:    ---------------------
[12:14:14][C][nextion:121]:   Flash Size:       33554432
[12:14:14][C][nextion:122]:   Wake On Touch:    True

```

I guess ESPHome tries to get this info at startup and the simulator does not provide it...

Is there a way to override this?

### Which version of ESPHome has the issue?

2021.10.3

### What type of installation are you using?

Docker

### What platform are you using?

ESP8266

### Board

Wemos D1 Mini

### Component causing the issue

Nextion

### Example YAML snippet

```yaml
esphome:
  name: esphomethermo-v0
  platform: ESP8266
  board: d1_mini_lite

# Enable logging
logger:
  baud_rate: 0     # Disable UART logging (pins GPIO1/3 are used for Nextion communication)
  
uart:
  id: uart_1
  rx_pin: GPIO3
  tx_pin: GPIO1
  baud_rate: 9600
  
 
# Enable Home Assistant API
api:

ota:
  password: !secret OTA_password

wifi:
  ssid: !secret Wifi_ssid
  password: !secret Wifi_password
  
  manual_ip:
    # Set this to the IP of the ESP
    static_ip: 192.168.1.31
    # Set this to the IP address of the router. Often ends with .1
    gateway: 192.168.1.1
    # The subnet of the network. 255.255.255.0 works for most home networks.
    subnet: 255.255.255.0

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Esphomethermo-V0"
    password: !secret OTA_password

captive_portal:

    
display:
  - platform: nextion
    id: nextion1
    update_interval: 5s
    uart_id: uart_1
    lambda: |-
      it.set_component_value("dActualTemp", 50);
      it.set_component_text("t0", "Hello World!");
```

### Anything in the logs that might be useful for us?

```txt
[12:10:29][W][nextion:072]: Nextion is not connected! 
[12:10:29][E][uart:015]: Reading from UART timed out at byte 0!
```

mikesch-1960, 2022-11-09:
It's quite old, but I'm new here and just found this. Maybe I can give you a few tips.
I'm not familiar with ESPHome, but I have some experience with the Nextion simulator. So two ideas:
1. Does the baud rate set in the simulator and in the ESP match?
2. The Nextion simulator does not send the messages
>   0x00 for Nextion has started or reset
>   and
>   0x88 for Nextion has powered up and is now initialized successfully.
> 
You have to simulate these messages yourself in the code of 'Program.s' of the HMI file when the editor is in debug mode, because the ESP might need these messages.
I use this code snippet for it:
~~~javascript
appid=1 // appid is always 0 while running in the Nexion Editor debugger!
if(appid==0) // if in debugging mode
{
  // Adjust the brightness. The simulator becomes significantly darker than a real Nextion at small values
  cfgDimActive=100
  cfgDimStandby=50
  // The simulator does not send the startup messages that Nextion devices send.
  // Since the MCU may rely on them, we do it manually here.
  //
  printh 00 00 00 ff ff ff // Power Up
  delay=100 // delay for a more realistic simulation
  printh 88 ff ff ff // Ready
}
~~~

winco206, 2023-11-22:
I have a same problem with communication. 

Any solution and advice?

cskollar, 2024-05-27:
I've found a workaround! Unfortunately to send the "power up" and "ready" state from the Editor (from Program.s) that mentioned above is not working. But if I create a dummy button in the Editor and assign a `print "comok"` command to it, pressing it will make ESPHome connect. It seems there are timing issues, which is why it doesn't work from Program.s, and that's why the dummy button needs to be pressed multiple times, but the point is, it works!

iqubik, 2024-06-12:
> I've found a workaround! Unfortunately to send the "power up" and "ready" state from the Editor (from Program.s) that mentioned above is not working. But if I create a dummy button in the Editor and [...]

Its work for me. this problem has many of time.

closed 2024-09-19 by jesserockz