esphome/issues #2873: DSMR and Improv_serial > logger baud_rate must not be 0

zuidwijk, 2021-12-24:
### The problem

I want to use `improv_serial:` in the code of DSMR. Yet DSMR requires the logger baud_rate to be 0.
Using `improv_serial` requires the logger baud_rate to be **not** 0.

Using Improv:
```
improv_serial:

logger:
#  baud_rate: 0
```

Using DSMR:
```
#improv_serial:

logger:
  baud_rate: 0
```

### Which version of ESPHome has the issue?

2021.12.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2021.12.5

### What platform are you using?

ESP8266

### Board

ESP8266-12F

### Component causing the issue

DSMR

### Example YAML snippet

```yaml
wifi:
#  networks:
#    - ssid: 'REPLACEME'
#      password: 'REPLACEME'

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: ${device_name}

captive_portal:

improv_serial:

substitutions:
  device_name: slimmelezer
  device_description: "DIY P1 module to read your smart meter"
    
esphome:
  name: ${device_name}
  comment: "${device_description}"
  platform: ESP8266
  esp8266_restore_from_flash: true
  board: d1_mini
  name_add_mac_suffix: false
  project:
    name: zuidwijk.slimmelezer
    version: "2.0"

# Enable logging
logger:
  baud_rate: 0
  level: DEBUG

# Enable Home Assistant API
api:

ota:

web_server:
  port: 80

uart:
  baud_rate: 115200
  rx_pin: D7
  rx_buffer_size: 1700

dsmr:
  max_telegram_length: 1700

sensor:
  - platform: dsmr
    energy_delivered_tariff1:
      name: "Energy Consumed Tariff 1"
    energy_delivered_tariff2:
      name: "Energy Consumed Tariff 2"
    energy_returned_tariff1:
      name: "Energy Produced Tariff 1"
    energy_returned_tariff2:
      name: "Energy Produced Tariff 2"
    power_delivered:
      name: "Power Consumed"
      accuracy_decimals: 3
    power_returned:
      name: "Power Produced"
      accuracy_decimals: 3
    electricity_failures:
      name: "Electricity Failures"
      icon: mdi:alert
    electricity_long_failures:
      name: "Long Electricity Failures"
      icon: mdi:alert
    voltage_l1:
      name: "Voltage Phase 1"
    voltage_l2:
      name: "Voltage Phase 2"
    voltage_l3:
      name: "Voltage Phase 3"
    current_l1:
      name: "Current Phase 1"
    current_l2:
      name: "Current Phase 2"
    current_l3:
      name: "Current Phase 3"
    power_delivered_l1:
      name: "Power Consumed Phase 1"
      accuracy_decimals: 3
    power_delivered_l2:
      name: "Power Consumed Phase 2"
      accuracy_decimals: 3
    power_delivered_l3:
      name: "Power Consumed Phase 3"
      accuracy_decimals: 3
    power_returned_l1:
      name: "Power Produced Phase 1"
      accuracy_decimals: 3
    power_returned_l2:
      name: "Power Produced Phase 2"
      accuracy_decimals: 3
    power_returned_l3:
      name: "Power Produced Phase 3"
      accuracy_decimals: 3
    gas_delivered:
      name: "Gas Consumed"
  - platform: uptime
    name: "SlimmeLezer Uptime"
  - platform: wifi_signal
    name: "SlimmeLezer Wi-Fi Signal"
    update_interval: 60s

text_sensor:
  - platform: dsmr
    identification:
      name: "DSMR Identification"
    p1_version:
      name: "DSMR Version"
  - platform: wifi_info
    ip_address:
      name: "SlimmeLezer IP Address"
    ssid:
      name: "SlimmeLezer Wi-Fi SSID"
    bssid:
      name: "SlimmeLezer Wi-Fi BSSID"
  - platform: version
    name: "ESPHome Version"
    hide_timestamp: false
```

oxan, 2021-12-24:
Using `dsmr` and `improv_serial` on the same UART is impossible, as both components process the bytes received over serial. They'll interfere with each other.

I'm not sure why improv_serial is coupled to the logger, but even if it wasn't, this wouldn't work.

mmakaay, 2021-12-24:
The conflict is not with the logger. The relation with the logger is that with `baud_rate: 0`, the DSMR reader can use hardware UART for RX. The conflict is, like @oxan says, the fact that both `improv_serial` and `dsmr` make use of RX, and RX is only available on a single hardware UART on ESP8266. If both must be used, I'd say that using ESP32 would be the way to go.

zuidwijk, 2021-12-24:
The only ESP32 suitable for < 250mA would be the C3... and the support for that isn't 100% ready (at least, not a few weeks ago). 

But I'm using D7 (gpio13) as RX, so the main RX/TX isn't used. Than can still be used for Improv, right?

jesserockz, 2021-12-27:
GPIO13 is the alternate RX pin and still uses the one and only hardware UART that the esp8266 has so it wont work.

To get multiple UART receivers to work, ESPHome will need to have some sort of actual UART bus that forwards all received packets to all listening components for decoding.

OttoWinter, 2022-01-18:
Note that connecting multiple devices on the RX side is not a good idea even if it were supported.

First, if the dsmr sends data at the same time as your PC using improv_serial (which can happen because neither is aware of the other's existence), both integrations will just end up with garbled data. Considering the amount of data dsmr sends that's not unlikely to happen.

Secondly, you'd also need to be careful with the wiring. If the DSMR is pulling the line high and your PC low you might end up damaging the components if they don't have special circuitry to prevent this.