esphome/issues #2874: Latest Dev build doesn't start

tungmeister, 2021-12-26:
### The problem

Just updated my dev build instance on hass and the below error is produced:
```
[10:06:12] INFO: Waiting for dashboard to come up...
[10:06:12] INFO: Starting ESPHome dashboard...
Traceback (most recent call last):
  File "/usr/bin/esphome", line 33, in <module>
    sys.exit(load_entry_point('esphome', 'console_scripts', 'esphome')())
  File "/usr/bin/esphome", line 22, in importlib_load_entry_point
    for entry_point in distribution(dist_name).entry_points
  File "/usr/lib/python3.9/importlib/metadata.py", line 524, in distribution
    return Distribution.from_name(distribution_name)
  File "/usr/lib/python3.9/importlib/metadata.py", line 187, in from_name
    raise PackageNotFoundError(name)
importlib.metadata.PackageNotFoundError: esphome
```

### Which version of ESPHome has the issue?

Latest Dev

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP8266

TAMCTec, 2021-12-26:
Same issue here, also install with Home Assistant Add-on

molnarti, 2021-12-27:
i have the same error for a quite a few days. i am on the 'latest' tag not dev.

tungmeister, 2021-12-27:
@molnarti  until someone commits a change to correct it you'll still have the issue, there hasn't been a commit for 5 days as devs are probably enjoying the xmas period.

jesserockz, 2021-12-27:
#2856

molnarti, 2021-12-27:
@tungmeister i am just wondering that a completely broken build gets released and stays relatively unnoticed for over a week.

ohumi, 2021-12-28:
Same for me: latest and 2021.12.2 are broken. 
dev is working

kbickar, 2021-12-29:
This is the same problem that https://github.com/esphome/esphome/pull/2936 was fixing.

Running `pip3 install --no-cache-dir -e /esphome` doesn't work for some reason, but `pip3 install --no-cache-dir /esphome` does, so probably something should be updated in the setup

jesserockz, 2021-12-29:
Looks like I simply missed the hassio version in the Dockerfile, made esphome/esphome#2964 to fix it

jesserockz, 2021-12-29:
Once this workflow has completed, you should be able to rebuild the dev addon
https://github.com/esphome/esphome/actions/runs/1635472804

closed 2021-12-29 by jesserockz

jesserockz, 2021-12-29:
Confirmed working on my HA OS install now. Sorry about that folks