esphome/issues #2875: Incorrect number set when setting large numbers

johanericandersson, 2021-12-27:
### The problem

When I try to set a number-entity, from Home Assistant, so it increases or decreases with another digit when dealing with numbers larger than 7 digits, the digits after the 7th gets all 0.
This happens only when the number increases or decreases with a digit, after that I can set the correct number and even change the number to a different one as long as the number of digits doesn't change.

E.g. changing from 9015153 to 90151534 results in 90151530. Changing from 9015153454 results in 9015153000.

### Which version of ESPHome has the issue?

2021.12.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2021.12.5

### What platform are you using?

ESP8266

### Board

lolin nodemcu v3

### Component causing the issue

number

### Example YAML snippet

```yaml
number:
  - platform: template
    name: ${device_name}_counter_lane1
    id: ${device_name}_counter_lane1
    optimistic: true
    min_value: -1
    max_value: 9000000000000000
    step: 1
    initial_value: -1

binary_sensor:
  - platform: gpio
    name: ${device_name}_lane1
    id: ${device_name}_lane1
    pin:
      number: GPIO14
      inverted: true
      mode:
        input: true
        pullup: true
    filters:
      - delayed_on: 100ms
      - delayed_off: 100ms
    on_press:
      lambda: |-
        if (id(${device_name}_counter_lane1).state >= 0) {
          auto call = id(${device_name}_counter_lane1).make_call();
          call.set_value(id(${device_name}_counter_lane1).state + 1);
          call.perform();
        }
```

martgras, 2021-12-27:
This happens because the underlying data type is float (32 bits).
A 32-bit float has about 7 digits of precision.  That means for very large numbers you can increase by 1 anymore and must use larger steps

johanericandersson, 2021-12-28:
That's a bit disappointing.
Is there any other "number-component" that uses double as data type?
Because I really need bigger numbers that I can increase by 1.
The work-around would be to split the number into multiple floats and combine them in HA.

martgras, 2021-12-28:
HA doesn't provide an alternative entity type that esphome could use (like Integer ... ) 
https://developers.home-assistant.io/docs/core/entity/number
If HA adds something like for example LargeNumber or BigInteger then esphome could take advantage of it.

afarago, 2022-10-02:
+1
I am debugging my esphome config for 12 hours, and finding the answer myself lead to this thread. Just would vote up this scenario.

My use case is to transfer **unix epoch timestamp** for sending "last updated" information.
The reason:  checking on whether the device is regularly updated unix timstamp is a very useful candidate as it clearly shows whether there are bigger "spots" using a simple graph.

I just came to realise that numbers like "1664424242" can easily gets "truncated".

```
sensor:
  - platform: template
    id: lastupdate_ts
    name: ${systemName}.Last Update Timestamp

time:
  - platform: homeassistant
    timezone: Europe/Berlin
    on_time_sync:
      then:
        - lambda: |-
            auto time_isvalid = id(homeassistant_time).now().is_valid();
            auto time_text = id(homeassistant_time).now().strftime("%Y-%m-%d %H:%M:%S");
            long long int time_ts = id(homeassistant_time).now().timestamp;

            if (time_isvalid) {
              id(lastupdate).publish_state(time_text);
              id(lastupdate_ts).publish_state(time_ts); id(lastupdate_ts).update();
            }
```

Currently I have no real workaround, would appreciate any ideas...