esphome/issues #2876: got error: multi-line comment, but no comments are used

Bascht74, 2021-12-27:
### The problem

I get the following error: 

```
Compiling .pioenvs/nilan/src/main.o
/config/nilan.yaml:460:3: error: multi-line comment [-Werror=comment]
           controller->queue_command(set_value_command);
   ^
cc1plus: some warnings being treated as errors
*** [.pioenvs/nilan/src/main.o] Error 1
```

### Which version of ESPHome has the issue?

2021.12.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

./.

### What platform are you using?

ESP32-IDF

### Board

./.

### Component causing the issue

Compiler

### Example YAML snippet

```yaml
- platform: template
    name: "${upper_devicename} Set user function"
    id: nilan_set_user_function
    optimistic: true
    update_interval: 60s
    entity_category: config
    options:
       - "None"
       - "Extend"
       - "Inlet"
       - "Exhaust"
       - "External heater offset"
       - "Ventilate"
       - "Cooker Hood"
    set_action:  
      - lambda: |-
          modbus_controller::ModbusController *controller = id(nilan_modbus_controller);
          std::vector<uint16_t> value; 
          uint16_t reg = 601;
          if (0 == x.compare(std::string{"None"}))        {value.push_back(0);}
          if (0 == x.compare(std::string{"Extend"}))      {value.push_back(1);}
          if (0 == x.compare(std::string{"Inlet"}))       {value.push_back(2);}
          if (0 == x.compare(std::string{"External heater offset"})) {value.push_back(3);}
          if (0 == x.compare(std::string{"Erase"}))       {value.push_back(4);}
          if (0 == x.compare(std::string{"Cooker Hood"})) {value.push_back(6);}
          if (0 == x.compare(std::string{"Ventilate"}))   {value.push_back(5);}
          modbus_controller::ModbusCommandItem set_value_command = modbus_controller::ModbusCommandItem::create_write_multiple_command(controller, reg, 1, value);
          controller->queue_command(set_value_command);
```

### Anything in the logs that might be useful for us?

```txt
automation_20->add_actions({lambdaaction_4});
  // select.template:
  //   platform: template
  //   name: Nilan Set user function
  //   id: nilan_set_user_function
  //   optimistic: true
  //   update_interval: 60s
  //   entity_category: config
  //   options:
  //   - None
  //   - Extend
  //   - Inlet
  //   - Exhaust
  //   - External heater offset
  //   - Ventilate
  //   - Cooker Hood
  //   set_action:
  //     then:
  //     - lambda: !lambda "modbus_controller::ModbusController *controller = id(nilan_modbus_controller);\n<cont>
  //         std::vector<uint16_t> value; \nuint16_t reg = 601;\nif (0 == x.compare(std::string{\"<cont>
  //         None\"}))        {value.push_back(0);}\nif (0 == x.compare(std::string{\"Extend\"<cont>
  //         }))      {value.push_back(1);}\nif (0 == x.compare(std::string{\"Inlet\"}))<cont>
  //         \       {value.push_back(2);}\nif (0 == x.compare(std::string{\"External heater<cont>
  //         \ offset\"})) {value.push_back(3);}\nif (0 == x.compare(std::string{\"Erase\"<cont>
  //         }))       {value.push_back(4);}\nif (0 == x.compare(std::string{\"Cooker Hood\"<cont>
  //         })) {value.push_back(6);}\nif (0 == x.compare(std::string{\"Ventilate\"})) <cont>
  //         \  {value.push_back(5);}\nmodbus_controller::ModbusCommandItem set_value_command\
  //         \ = modbus_controller::ModbusCommandItem::create_write_multiple_command(controller,\
  //         \ reg, 1, value);\ncontroller->queue_command(set_value_command);"
  //       type_id: lambdaaction_5
  //     trigger_id: trigger_14
  //     automation_id: automation_21
  //   disabled_by_default: false
  //   initial_option: None
  nilan_set_user_function = new template_::TemplateSelect();
  nilan_set_user_function->set_update_interval(60000);
  nilan_set_user_function->set_component_source("template.select");
  App.register_component(nilan_set_user_function);
  App.register_select(nilan_set_user_function);
  nilan_set_user_function->set_name("Nilan Set user function");
  nilan_set_user_function->set_disabled_by_default(false);
  nilan_set_user_function->set_entity_category(ENTITY_CATEGORY_CONFIG);
  nilan_set_user_function->traits.set_options({"None", "Extend", "Inlet", "Exhaust", "External heater offset", "Ventilate", "Cooker Hood"});
  nilan_set_user_function->set_optimistic(true);
  nilan_set_user_function->set_initial_option("None");
  automation_21 = new Automation<std::string>(nilan_set_user_function->get_set_trigger());
  lambdaaction_5 = new LambdaAction<std::string>([=](std::string x) -> void {
      #line 449 "/config/nilan.yaml"
      modbus_controller::ModbusController *controller = nilan_modbus_controller;
      std::vector<uint16_t> value;
      uint16_t reg = 601;
      if (0 == x.compare(std::string{"None"}))        {value.push_back(0);}
      if (0 == x.compare(std::string{"Extend"}))      {value.push_back(1);}
      if (0 == x.compare(std::string{"Inlet"}))       {value.push_back(2);}
      if (0 == x.compare(std::string{"External heater offset"})) {value.push_back(3);}
      if (0 == x.compare(std::string{"Erase"}))       {value.push_back(4);}
      if (0 == x.compare(std::string{"Cooker Hood"})) {value.push_back(6);}
      if (0 == x.compare(std::string{"Ventilate"}))   {value.push_back(5);}
      modbus_controller::ModbusCommandItem set_value_command = modbus_controller::ModbusCommandItem::create_write_multiple_command(controller, reg, 1, value);
      controller->queue_command(set_value_command);
  });
```

### Additional information

I added a part of main.cpp

I don't know how the code produces the error. 

If I delete two if lines, it compiles...

As I am not a programmer, but it seems to me like something that needs to be corrected or at least recognized and shown as error in the editor if the cause is on my side. 
I don't have a clue what is going on here and producing the error...

Sebastian

martgras, 2021-12-27:
This should be fixed in dev already and will probably be in the next release : https://github.com/esphome/esphome/pull/2891

Bascht74, 2021-12-28:
I will check in the morning…

Bascht74, 2021-12-28:
I manually changed the file of the pull request and after that the issue is gone!
So I can close this...
Thx!

closed 2021-12-28 by Bascht74

Bascht74, 2022-01-08:
@martgras I tried the latest Release 2021.12.3 which came out December 30, but the issue it still there. Do I have to wait until 2022.01? It seems like the fix is not in 2021.12.3...

martgras, 2022-01-08:
It's scheduled for 2021.12.4  or the next release