esphome/issues #2877: NeoPixelBus Light: Issue with multiple (GPIO) outputs on a single ESP32

Fitski, 2021-12-28:
### The problem

In my kitchen, I want led strips (SK6812) on top of the kitchen cupboards, as well underneath them to shine on the counter.
For this situation, I intend to use a single ESP32E configured with two NeoPixelBus platforms each having different GPIO output for each led strip. Unfortunately, I am not getting the desired behavior.

In HA, when I turn on "Kitchen Light Counter" (GPIO13).

- led strip GPIO16 (Kitchen Up) turns on.
- led strip GPIO13 (Kitchen Counter) stays off.

The same thing happens when I turn on "Kitchen Light Up" which is, in this case, a good thing since that is the desired strip tied to that entity.
After some troubleshooting such as: replacing ESP32 boards and using different GPIO pins, I tried the following scenario:

Instead of making two NeoPixelBus light components, I made just one. In this case, I only used "Kitchen Light Counter" (GPIO13) since it never lighted up in the previous scenario.
To my surprise:

- led strip GPIO16 (Kitchen Up) turns on.
- led strip GPIO13 (Kitchen Counter) **turns on**.

While there was technically one GPIO output configured, both led strips on different GPIO pins were turning on. 

Finally, I tried to use the FastLed platform to see if the problem persists, but this worked like a charm (however, here I cannot use the white-led channel on the SK6812). Each led strip was controllable per HA entity.

At this point, ssieb (member on discord who helped me with most of the troubleshooting steps) stated that there might be an issue with the NeoPixelBus.

(these are one of my first filed issues on GitHub, feedback welcome)

### Which version of ESPHome has the issue?

2021.12.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2021.12.5

### What platform are you using?

ESP32-IDF

### Board

espressif ESP32E development board.

### Component causing the issue

NeoPixelBus Light

### Example YAML snippet

```yaml
esphome:
  name: esp32node-kitchen

esp32:
  board: esp32dev
  framework:
    type: arduino

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:
  password: "eda3fd2f0c5db9bc5e49f268d1f1b662"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Esp32Node-Kitchen"
    password: "OyjiSYtd123Z"

captive_portal:

light:
  
- platform: neopixelbus
  variant: SK6812
  pin: GPIO13
  num_leds: 165
  type: GRBW
  id: kitchen_counter
  name: "Kitchen Light Counter"
  
- platform: neopixelbus
  variant: SK6812
  pin: GPIO16
  num_leds: 260
  type: GRBW
  id: kitchen_up
  name: "Kitchen Light Up"
```

maik1607, 2022-01-05:
try to add a method with different channels, it helped me

Fitski, 2022-01-05:
> try to add a method with different channels, it helped me

Hi @maik1607, thanks for your reply. Can you elaborate on what you mean by "adding a method with different channels"?
Perhaps you could share your configuration. Thank you in advance!

maik1607, 2022-01-06:
Sorry for the short reply. Here is part of my configuration:

```
light:
  - platform: neopixelbus
    type: RGB
    variant: WS2811
    method:
      type: esp32_rmt
      channel: 0
    pin: ${neopix_room1}
    num_leds: 20
    id: nxroom1
    internal: true
  - platform: neopixelbus
    type: RGB
    variant: WS2811
    method:
      type: esp32_rmt
      channel: 1
    pin: ${neopix_room2}
    num_leds: 20
    id: nxroom2
    internal: true
```

Fitski, 2022-01-08:
Thank you @maik1607, This workaround works for me as well!

spali, 2022-09-26:
can confirm this behavior still occurs in 2022.9.1.
Configuring two light's with neopixelbus with different pins controls always the first defined light.
And the workaround with the channels also still works.

raulgbcr, 2022-11-12:
2022.10.2 seeing the same behaviour. Wondering if this is something wrong with neopixelbus and not ESPHome.

bigbangus, 2022-12-20:
This fix works for me as well with HA and ESPHome 2022.12.

```
light:
  - platform: neopixelbus
    type: GRBW
    variant: SK6812
    pin: GPIO22
    method:
      type: esp32_rmt
      channel: 0
    num_leds: 84
    name: "Light 1"

  - platform: neopixelbus
    type: GRBW
    variant: SK6812
    pin: GPIO23
    method:
      type: esp32_rmt
      channel: 1
    num_leds: 54
    name: "Light 2"
```

SparkyCola, 2023-01-03:
I can confirm both the issue and workaround (pins GPIO5 and GPIO32 on ESP32)

fkollmann, 2023-11-30:
For those using the NeoPixelBus directly, the following change is required to support multiple busses:

original code:

```cpp
NeoPixelBus<NeoRgbFeature, Neo800KbpsMethod> pixels(CAR_LIGHTS_COUNT, CAR_LIGHTS_PIN);
```

updated code:

```cpp
static NeoPixelBus<NeoRgbFeature, NeoEsp32Rmt0800KbpsMethod> carPixels(CAR_LIGHTS_COUNT, CAR_LIGHTS_PIN);
static NeoPixelBus<NeoRgbFeature, NeoEsp32Rmt1800KbpsMethod> riverPixels(RIVER_LIGHTS_COUNT, RIVER_LIGHTS_PIN)
```

For supporting multiple busses, make sure to have separate methods for each instance (the number after `Rmt`):

- `NeoEsp32Rmt0800KbpsMethod`
- `NeoEsp32Rmt1800KbpsMethod`
- etc.

Please keep in mind the feature (`Rgb`, `Rgbw`, etc.) and method (`800Kbps`, `400Kbps`, etc.) must match your neo pixel type.

ssieb, 2023-12-01:
If it's an ESP32, you're better off using the RMT component instead.

Eric-IOT, 2024-02-13:
> Thank you @maik1607, This workaround works for me as well! using Lolin S3mini Home assistant and ESPHome 2023.12.9

brooksben11, 2024-08-21:
Holy cow, just spent an hour plus troubleshooting what I assumed must be a wiring/hardware issue only to find this thread. How has this not been fixed (or at least a note in the docs?)!?

ssieb, 2024-08-21:
You should be using the RMT component instead.

jkosharek, 2025-02-07:
The method doesn't seem to be working on 2024.12.4. Is there another way to use 2 GPIO with addressable WS2812 LEDs?

```
esp32:
  board: esp32dev
  framework:
    type: arduino

light:
# OnBoard
  - platform: neopixelbus
    name: "Sys-LED Onboard"
    id: sys_led_onboard
    type: GRB
    variant: WS2812
    method:
      type: esp32_rmt
      channel: 0
    pin: GPIO16
    num_leds: 1
    effects:
      - pulse:
          name: Pulse Effect
          transition_length: 550ms
          update_interval: 550ms
      - strobe:
          name: Strobe Effect
          colors:
            - state: true
              brightness: 100%
              red: 100%
              green: 0%
              blue: 0%
              duration: 70ms
            - state: false
              duration: 25ms
            - state: true
              brightness: 100%
              red: 0%
              green: 0%
              blue: 100%
              duration: 150ms
            - state: false
              duration: 25ms
# OnBoard
# Strip
  - platform: neopixelbus
    name: "Sys-LED-Status"
    id: sys_led_status
    type: GRB
    variant: WS2812
    method:
      type: esp32_rmt
      channel: 1
    pin: GPIO18
    num_leds: 6
# Strip
```