esphome/issues #2878: Cannot compile for ESP8266 using Docker image, linking step fails

djryanj, 2021-12-28:
### The problem

I have been unable to compile firmware for an ESP8266 board, the linking step fails consistently with: `xtensa-lx106-elf-g++: fatal error: -fuse-linker-plugin, but liblto_plugin.so not found`

### Which version of ESPHome has the issue?

2021.12.1, 2022.1.0-dev

### What type of installation are you using?

Docker

### What platform are you using?

ESP8266

### Board

Wemos D1 Mini, also esp01_1m

### Component causing the issue

Unknown

### Example YAML snippet

```yaml
esphome:
  name: outdoorplug

esp8266:
  board: d1_mini

wifi:
  ssid: "ssid"
  password: "password"
  ap:
    ssid: "d1-hotspot"
    password: "password"

captive_portal:

logger:

ota:
  password: ""

switch:
  - platform: gpio
    name: "Plug"
    pin: D6
    id: relay

status_led:
  pin:
    number: D4
    inverted: true

mqtt:
  topic_prefix: outdoors/plug
  discovery: false
  broker: mqtt.server
```

### Anything in the logs that might be useful for us?

```txt
ryan@hostname:/$ docker run --rm -v "${PWD}":/config -it esphome/esphome run outdoor.yml
INFO Reading configuration outdoor.yml...
INFO Generating C++ source...
[... 176 lines cut ...]
Compiling .pioenvs/outdoorplug/FrameworkArduino/libb64/cdecode.cpp.o
Compiling .pioenvs/outdoorplug/FrameworkArduino/libb64/cencode.cpp.o
Compiling .pioenvs/outdoorplug/FrameworkArduino/libc_replacements.cpp.o
Compiling .pioenvs/outdoorplug/FrameworkArduino/sntp-lwip2.cpp.o
Compiling .pioenvs/outdoorplug/FrameworkArduino/spiffs/spiffs_cache.cpp.o
Compiling .pioenvs/outdoorplug/FrameworkArduino/spiffs/spiffs_check.cpp.o
Compiling .pioenvs/outdoorplug/FrameworkArduino/spiffs/spiffs_gc.cpp.o
Compiling .pioenvs/outdoorplug/FrameworkArduino/spiffs/spiffs_hydrogen.cpp.o
Compiling .pioenvs/outdoorplug/FrameworkArduino/spiffs/spiffs_nucleus.cpp.o
Compiling .pioenvs/outdoorplug/FrameworkArduino/spiffs_api.cpp.o
Compiling .pioenvs/outdoorplug/FrameworkArduino/sqrt32.cpp.o
Compiling .pioenvs/outdoorplug/FrameworkArduino/time.cpp.o
Compiling .pioenvs/outdoorplug/FrameworkArduino/uart.cpp.o
Compiling .pioenvs/outdoorplug/FrameworkArduino/umm_malloc/umm_info.c.o
Compiling .pioenvs/outdoorplug/FrameworkArduino/umm_malloc/umm_integrity.c.o
Compiling .pioenvs/outdoorplug/FrameworkArduino/umm_malloc/umm_local.c.o
Compiling .pioenvs/outdoorplug/FrameworkArduino/umm_malloc/umm_malloc.cpp.o
Compiling .pioenvs/outdoorplug/FrameworkArduino/umm_malloc/umm_poison.c.o
Archiving .pioenvs/outdoorplug/libFrameworkArduino.a
Linking .pioenvs/outdoorplug/firmware.elf
xtensa-lx106-elf-g++: fatal error: -fuse-linker-plugin, but liblto_plugin.so not found
compilation terminated.
*** [.pioenvs/outdoorplug/firmware.elf] Error 1
============================================= [FAILED] Took 72.60 seconds =============================================
```

OttoWinter, 2022-01-18:
Try removing the `.esphome` folder in the current directory ($PWD) and trying again. If that does not work, please say what sort of filesystem (ext4, ntfs) the current directory is in.

djryanj, 2022-01-18:
@OttoWinter - deleting the `.esphome` folder was successful. It compiles fine.

Even though that was successful, it may be worth noting that my environment technically is Docker for Windows and the filesystem is NTFS, but mounted into WSL which is where I was running Docker from. Perhaps a source of the issue.

I believe I also tried running it from PowerShell, but the result is the same, however, I cannot replicate that now since I deleted `.esphome`.

Thanks for your help.

raoulbhatia, 2022-12-16:
Possible related: #3076

djryanj, 2024-12-13:
Closing this as I'm not sure it's actually an issue (see success in deleting `.esphome`).

closed 2024-12-13 by djryanj