esphome/issues #2879: on_time not triggering for several minutes after long deep sleep

scy, 2021-12-28:
### The problem

I have a battery-powered ESP32 letter box sensor. It’s configured to wake up on the lid’s GPIO, or every 6 hours. It will then run for a not pre-determined time: Since `deep_sleep.run_duration` is not flexible enough for me (the sensor should only sleep when on battery), I basically `prevent` deep sleep on boot and instead run a custom script (called `maybe_sleep`) via `on_time` configured to `/5` (every 5 seconds) that will check the prerequisites and manually `enter` deep sleep when they’re fulfilled.

The issue I’m seeing is that most of the time, the controller will run for far longer (several minutes) than it’s supposed to after waking up before going into deep sleep again. I’ve added some debug logs and found out, that the custom script seems to be not being called by the `on_time` until after the controller is running for several minutes.

The logs show the relevant excerpt of what’s happening after booting up (after OTA), going to sleep and then waking up again on timer. Timestamps are from a host watching the MQTT logs and can therefore be trusted. A few things are interesting:

* It seems to sleep a few minutes shorter than expected (12:28:23 to 18:20:55).
* When booting up after sleeping, SNTP reports `Synchronized time: 2021-12-28 18:28:26`, even though it’s only 18:20:55. Is this time really fetched from the network, or is there some internal workaround that sets it to “time before sleep plus configured sleep duration”? I’ve tried to find out in the ESPHome source, but my C++ isn’t very good.
* Next time the text sensor configured to contain the current timestamp updates, the controller’s time seems to be okay again and matches the host watching the logs. However, from the point of view of the controller, the time now has jumped about 7½ minutes into the past.
* At 18:28:30, i.e. after about 7.6 minutes of uptime, the first post-sleep run of the `maybe_sleep` finally occurs. It determines that the preconditions are met and that the controller may sleep again.

I don’t think that it’s a coincidence that the `on_time` script only starts running again after the time reported by SNTP after wake-up (18:18:26) has passed. To me, it looks like what’s happening is:

* Board goes to sleep and keeps either the current time and sleep duration around, or stores the time as it should be after waking up in RTC memory or something.
* Board wakes up after 7½ minutes less than planned, thinks it’s 18:28:23 (time before sleep plus 6 hours) or something.
* Once the board thinks it’s 18:28:25, `on_time` loop hook gets triggered, calls `maybe_sleep`. Notes down that its next time to run is 18:28:30.
* `maybe_sleep` determines that it’s not time to sleep (because `flap_settled` is false on boot).
* SNTP syncs the internal time and moves it back to 18:20:55 or something.
* Every time the `on_time` hook is triggered, it skips `maybe_sleep` because it’s not 18:28:30 yet.
* Once the time has reached 18:28:30, `maybe_sleep` gets triggered.

I’ve noticed that #573 introduced code that is trying to notice clock jumps, but since it’s targeting time zone changes, it won’t act if the jump is less than 15 minutes (which it isn’t, in my case).

As a workaround, I might use the “Current Timestamp” text sensor’s `on_value` to trigger `maybe_sleep` instead of relying on `on_time`.

### Which version of ESPHome has the issue?

2021.12.2

### What type of installation are you using?

pip

### What platform are you using?

ESP32

### Board

esp32-devkitlipo

### Component causing the issue

deep_sleep

### Example YAML snippet

```yaml
# …

substitutions:
  iso8601_lambda: |-
    char str[21];
    time_t now = id(node_time).now().timestamp;
    strftime(str, sizeof(str), "%Y-%m-%dT%H:%M:%SZ", gmtime(&now));
    return (std::string) str;

# …

time:
  - platform: sntp
    id: node_time
    timezone: UTC
    on_time:
      - seconds: /5
        then:
          - script.execute: maybe_sleep

deep_sleep:
  id: sleep_flap
  run_duration: 1min
  sleep_duration: 6h
  wakeup_pin: 13
  wakeup_pin_mode: INVERT_WAKEUP

text_sensor:
  # …
  - name: Current Timestamp
    id: node_ts
    platform: template
    update_interval: 5s
    lambda: ${iso8601_lambda}

# …

script:
  # …
  # go to sleep, but only on battery and with flap settled
  # disable sleep if we're not on battery
  - id: maybe_sleep
    then:
      - if:
          condition:
            binary_sensor.is_on: on_bat
          then:
            - logger.log: "maybe_sleep: on battery"
            - if:
                condition:
                  binary_sensor.is_on: flap_settled
                then:
                  - logger.log: "maybe_sleep: flap settled, entering sleep"
                  - delay: 3s
                  - deep_sleep.enter: sleep_flap
                else:
                  - logger.log: "maybe_sleep: flap not settled, not entering sleep"
          else:
            - logger.log: "maybe_sleep: not on battery"
            - deep_sleep.prevent: sleep_flap
```

### Anything in the logs that might be useful for us?

```txt
[12:28:01][D][sntp:075]: Synchronized time: 2021-12-28 12:28:00
[…]
[12:28:20][D][main:098]: maybe_sleep: on battery
[12:28:20][D][main:103]: maybe_sleep: flap settled, entering sleep
[12:28:20][D][text_sensor:067]: 'Current Timestamp': Sending state '2021-12-28T12:28:20Z'
[12:28:21][D][sensor:113]: 'Uptime': Sending state 23.63900 s with 0 decimals of accuracy
[12:28:23][I][deep_sleep:072]: Beginning Deep Sleep
[…]
[18:20:55][I][mqtt:214]: MQTT Connected!
[…]
[18:20:55][D][sntp:075]: Synchronized time: 2021-12-28 18:28:26
[18:20:56][D][sensor:113]: 'Uptime': Sending state 4.31500 s with 0 decimals of accuracy
[18:20:56][D][text_sensor:067]: 'Current Timestamp': Sending state '2021-12-28T18:20:56Z'
[…]
[18:28:26][D][text_sensor:067]: 'Current Timestamp': Sending state '2021-12-28T18:28:26Z'
[18:28:30][D][main:098]: maybe_sleep: on battery
[18:28:30][D][main:103]: maybe_sleep: flap settled, entering sleep
[…]
[18:28:31][D][sensor:113]: 'Uptime': Sending state 459.31299 s with 0 decimals of accuracy
[18:28:31][D][text_sensor:067]: 'Current Timestamp': Sending state '2021-12-28T18:28:31Z'
[18:28:33][I][deep_sleep:072]: Beginning Deep Sleep
```

jehy, 2023-03-20:
I found a workaround in thread https://github.com/esphome/issues/issues/573 - when I start using `platform: homeassistant` instead of `sntp` - everything starts working fine.

Oh. While I was writing this post, I understood that my ESP is banned from internet access. So it can't access ntp servers but can get time from HA. Likely it's the reason. I suppose there should be some kind of log entry for this....

jjmonteiro, 2024-08-19:
Afaik the ESP cannot sleep for more than 70min, so you'll have to count the wake times instead.

If you're really looking into save battery power, you might want to startup with wifi disabled and keep the runtime to an absolute minimum.

Once it wakes 6 times for a duration of 60min, then enable the wifi, do your stuff and go back to sleep another 6 times.
Also make your sensor update_interval half the run_duration, to ensure at least one valid reading.

It would be something like:

```
substitutions:
  deep_sleep_hours: '6'

deep_sleep:
  run_duration: 10s
  sleep_duration: 60min

esphome:
  name: esp-32

  on_boot:
    # This is important for both cases
    priority: 0 
    then:
      - lambda: |-
          id(wake_counter) += 1;
      - if:
          condition:
            lambda: |-
              return id(wake_counter) >= ${deep_sleep_hours};
          then:
            - lambda: |-
                id(wake_counter) = 0;
            - wifi.enable
          else:
            - deep_sleep.enter:

globals:
  - id: wake_counter
    type: int
    restore_value: yes
    # Ensure power up with WiFi enabled
    initial_value: '99'

wifi:
  # Ensure good connection at first attempt
  power_save_mode: NONE
  fast_connect: true
  # Do not enable at every boot up
  enable_on_boot: False

# Disable Safe Mode
safe_mode:
  disabled: true
```