esphome/issues #2880: Waveshare 7.5inV2 Not Working

lux4rd0, 2021-12-28:
### The problem

I have one of the newer 7.5 inch V2 e-ink devices and I'm trying to make it work with ESPHome. It seems that there have been several other users experiencing the same issue - and a few older closed out Incidents as well:

- https://github.com/esphome/issues/issues/1296
- https://community.home-assistant.io/t/waveshare-e-ink-7-5-with-waveshare-esp32-wroom-32-controler/151346/14
- https://github.com/esphome/issues/issues/2216

I've tried several different pin combinations and none of them bring to life the e-ink device. I have a Waveshare 4.2in that works just fine. I took all of those same configurations and was hoping to plugin in my 7.5inV2 and - no go. I even tried several other pin examples from posts - and those didn't work either.

### Which version of ESPHome has the issue?

v2021.12.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2021.12.5

### What platform are you using?

ESP32

### Component causing the issue

Waveshare 7.5inV2

### Example YAML snippet

```yaml
spi:
  clk_pin: GPIO18
  mosi_pin: GPIO23

display:
  - platform: waveshare_epaper
    id: esp13_epaper
    cs_pin: GPIO5
    dc_pin: GPIO19
    busy_pin: GPIO16
    reset_pin: GPIO17
    model: 7.50inV2
    update_interval: 60s
```

### Anything in the logs that might be useful for us?

```txt
Nothing in the logs...

spi:
  clk_pin: GPIO13
  mosi_pin: GPIO15

display:
  - platform: waveshare_epaper
    id: esp12_epaper
    cs_pin: GPIO12
    busy_pin: GPIO33
    reset_pin: GPIO21
    dc_pin: GPIO27
    model: 4.20in
    update_interval: 60s

This ^^^ works fine for the 4.2in product. Changing the model to 7.50inV2 should be all I need to do to make it work with the 7.5 inch V2 model... But it doesn't. I've used two brand new boards from Waveshare - assumed the first one was defective.
```

lux4rd0, 2022-02-02:
Wanted to come back around to see if I might be able to nudge somebody into assistance. Looks like a few more folks have the same issues:

https://community.home-assistant.io/t/waveshare-e-paper-7-5-inch-v2-800x480-not-supported/159449/5

It's just sitting in a box at this point... I really enjoy my 4.2in waveshare - but I need the size of the 7.5in to work for my home automation project.

Thanks!

jaca76, 2022-02-02:
Hi
I have exactly the same issue, nothing displayed, the arduino example from GxEPD2 library is working correctly with the same wiring.
I'm using ESP32

Eriner, 2022-02-04:
I added support for this a week or so ago. Resolved by esphome/esphome#3121

lux4rd0, 2022-02-05:
@Eriner That's awesome!! Thank you! I grabbed the three updated files referenced in your 3121 PR and tried them out. I used your example YAML file as well as the one that I'm using with my 4.2" and neither provided a successful result. I'm using both esp32dev and featheresp32 - along with about three different pin configurations. Unfortunately, there's nothing in the logs and the screen never flashes to life.

Not sure what else to check...

Eriner, 2022-02-05:
Is your display a 7.5in-BV2? Do you mind linking to the actual product you have? The 7.5in-BV2 is the one with red color support, 800x600. Mine had a "v3" sticker on the back, as the BV2 and BV3 variants are pretty much the same. There's a link to the waveshare product page in the PR I linked. Is it the same as the one you're using?

lux4rd0, 2022-02-05:
So - two things... :)

I assumed that when you said "I added support for the 7.5in-BV2 a week" to my issue - that perhaps I needed to use your version to get around mine not working. Alas - I don't have the version with red - just the [V2 B&W version](https://www.amazon.com/waveshare-7-5inch-HAT-Raspberry-Consumption/dp/B075R4QY3L?th=1). So I switched my settings back to model: `7.50inV2` instead of model: `7.50in-bv2` - and it lit up!! I'm like - whoa!!

And two - it lit up because of this additional setting in your example config: `reset_duration: 2ms`

I went back to the [documentation](https://esphome.io/components/display/waveshare_epaper.html#configuration-variables) and saw this:

**reset_duration** (Optional, Time): Duration for the display reset operation. Defaults to 200ms. Setting this value to 2ms may resolve issues with newer e-Paper Driver modules (e.g. Rev 2.1).

Sure enough - I have Rev 2.2 and didn't need that configuration for my other 4.2in Waveshare. Adding that configuration line now has my device working!! :)

So - thank you for your response - all roads (in this case) led to success.

Eriner, 2022-02-05:
Awesome, glad it all worked out for you! Cheers!

closed 2022-02-09 by lux4rd0