esphome/issues #2882: AM2302 only works after hard power cycle

chrismetcalf, 2021-12-29:
### The problem

I have a handfull of [AM2302/DH22](https://www.amazon.com/gp/product/B07XBRMN4P/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1) sensors wired up to ESP01 boards around my house. They work _great_ after a hard power cycle (remove USB power and plug back in) but after awhile, or after I do an `esphome run` to upload an updated program over OTA, they start reporting:

```
[00:11:21][W][dht:167]: Requesting data from DHT failed!
[00:11:21][W][dht:060]: Invalid readings! Please check your wiring (pull-up resistor, pin number).
[00:11:21][D][sensor:113]: 'Shed Temperature': Sending state nan °C with 1 decimals of accuracy
[00:11:21][D][sensor:113]: 'Shed Humidity': Sending state nan % with 0 decimals of accuracy
```
I've tried various other options for `model` but none seem to make a difference. And it works great on first boot. The AM2302s and ESP01 are powered off of 3.3v via a [little step-down board[(https://www.amazon.com/gp/product/B07CP4P5XJ/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1).

### Which version of ESPHome has the issue?

2021.12.2

### What type of installation are you using?

pip

### Which version of Home Assistant has the issue?

2021.12.4

### What platform are you using?

ESP32

### Board

ESP01

### Component causing the issue

AM2302

### Example YAML snippet

```yaml
## ESP01 Temperature Sensor
#
# Connections:
# - 5V PWR -> Conv VIN 
# - 5V GND -> Conv GND -> AM2302 - -> ESP GND
# - Conv VOUT (3.3v) -> ESP VCC -> AMC2302 +
# - ESP VCC -> 10K resistor -> ESP CHPD
# - ESP GP2 -> AM2302 Middle
# - 
esphome:
  name: shed-temp
  platform: ESP8266
  board: esp01_1m

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "shed-temp-fallback"
    password: !secret esphome_password

captive_portal:

# Enable logging
logger:

# Enable Home Assistant API
api:
  password: !secret esphome_password

ota:
  password: !secret esphome_password

web_server:
  port: 80

sensor:
  - platform: dht
    pin: GPIO2
    model: AM2302
    temperature:
      name: Shed Temperature
      force_update: true
      filters:
        - offset: -2.0667
    humidity:
      name: Shed Humidity
      force_update: true

  - platform: uptime
    name: Shed ESP Uptime

  - platform: wifi_signal
    name: Shed ESP Signal Strength

text_sensor:
  - platform: wifi_info
    ip_address:
      name: Shed ESP IP Address
```

### Anything in the logs that might be useful for us?

```txt
~/src/github.com/esphome(master*) » esphome run shed-temp.yaml
INFO Reading configuration shed-temp.yaml...
INFO Generating C++ source...
INFO Compiling app...
*********************************************************************************************************************************
Obsolete PIO Core v5.2.2 is used (previous was 5.2.4)
Please remove multiple PIO Cores from a system:
https://docs.platformio.org/page/faq.html#multiple-platformio-cores-in-a-system
*********************************************************************************************************************************
Processing shed-temp (board: esp01_1m; framework: arduino; platform: platformio/espressif8266 @ 2.6.3)
---------------------------------------------------------------------------------------------------------------------------------
HARDWARE: ESP8266 80MHz, 80KB RAM, 1MB Flash
Dependency Graph
|-- <ESPAsyncTCP-esphome> 1.2.3
|-- <ESPAsyncWebServer-esphome> 2.1.0
|   |-- <ESPAsyncTCP-esphome> 1.2.3
|   |-- <Hash> 1.0
|   |-- <ESP8266WiFi> 1.0
|-- <DNSServer> 1.1.1
|-- <ESP8266WiFi> 1.0
|-- <ArduinoJson-esphomelib> 5.13.3
|-- <ESP8266mDNS> 1.2
Compiling .pioenvs/shed-temp/src/main.cpp.o
Linking .pioenvs/shed-temp/firmware.elf
Building .pioenvs/shed-temp/firmware.bin
RAM:   [====      ]  39.4% (used 32240 bytes from 81920 bytes)
Flash: [====      ]  41.8% (used 427656 bytes from 1023984 bytes)
================================================= [SUCCESS] Took 10.81 seconds =================================================
INFO Successfully compiled program.
INFO Resolving IP address of shed-temp.local
INFO  -> 192.168.1.206
INFO Uploading .esphome/build/shed-temp/.pioenvs/shed-temp/firmware.bin (431808 bytes)
INFO Compressed to 298869 bytes
Uploading: [============================================================] 100% Done...

INFO Waiting for result...
INFO OTA successful
INFO Successfully uploaded program.
INFO Starting log output from shed-temp.local using esphome API
INFO Successfully connected to shed-temp.local
[00:07:51][I][app:102]: ESPHome version 2021.12.2 compiled on Dec 29 2021, 00:07:15
[00:07:51][C][dht:017]: DHT:
[00:07:51][C][dht:018]:   Pin: GPIO2
[00:07:51][C][dht:024]:   Model: DHT22 (or equivalent)
[00:07:51][C][dht:027]:   Update Interval: 60.0s
[00:07:51][C][dht:029]:   Temperature 'Shed Temperature'
[00:07:51][C][dht:029]:     Device Class: 'temperature'
[00:07:51][C][dht:029]:     State Class: 'measurement'
[00:07:51][C][dht:029]:     Unit of Measurement: '°C'
[00:07:51][C][dht:029]:     Accuracy Decimals: 1
[00:07:51][C][dht:030]:   Humidity 'Shed Humidity'
[00:07:51][C][dht:030]:     Device Class: 'humidity'
[00:07:51][C][dht:030]:     State Class: 'measurement'
[00:07:51][C][dht:030]:     Unit of Measurement: '%'
[00:07:51][C][dht:030]:     Accuracy Decimals: 0
...
[00:08:21][W][dht:167]: Requesting data from DHT failed!
[00:08:21][W][dht:060]: Invalid readings! Please check your wiring (pull-up resistor, pin number).
[00:08:21][D][sensor:113]: 'Shed Temperature': Sending state nan °C with 1 decimals of accuracy
[00:08:21][D][sensor:113]: 'Shed Humidity': Sending state nan % with 0 decimals of accuracy
```

Weissnix4711, 2021-12-30:
Same here. Had the issue for a while now, but just kinda forgot about it.

My logs:

```
[20:37:04][I][app:102]: ESPHome version 2021.12.2 compiled on Dec 26 2021, 10:33:51
...
[20:37:04][C][dht:017]: DHT:
[20:37:04][C][dht:018]:   Pin: GPIO4
[20:37:04][C][dht:024]:   Model: DHT22 (or equivalent)
[20:37:04][C][dht:027]:   Update Interval: 60.0s
[20:37:04][C][dht:029]:   Temperature 'Bathroom Temperature'
[20:37:04][C][dht:029]:     Device Class: 'temperature'
[20:37:04][C][dht:029]:     State Class: 'measurement'
[20:37:04][C][dht:029]:     Unit of Measurement: '°C'
[20:37:04][C][dht:029]:     Accuracy Decimals: 1
[20:37:04][C][dht:030]:   Humidity 'Bathroom Humidity'
[20:37:04][C][dht:030]:     Device Class: 'humidity'
[20:37:04][C][dht:030]:     State Class: 'measurement'
[20:37:04][C][dht:030]:     Unit of Measurement: '%'
[20:37:04][C][dht:030]:     Accuracy Decimals: 0
---
[20:37:45][W][dht:167]: Requesting data from DHT failed!
[20:37:45][W][dht:060]: Invalid readings! Please check your wiring (pull-up resistor, pin number).
[20:37:45][D][sensor:113]: 'Bathroom Temperature': Sending state nan °C with 1 decimals of accuracy
[20:37:45][D][sensor:113]: 'Bathroom Humidity': Sending state nan % with 0 decimals of accuracy
```

Also running the AM2302 off 3v3, with a 4k7 resistor between data and 3v3.

In previous versions of ESPHome, I've had the same issue, but I'd occasionally get a different error.

```
[11:26:41][W][dht:167]: Falling edge for bit 40 failed!
[11:26:41][W][dht:060]: Invalid readings! Please check your wiring (pull-up resistor, pin number).
```

D130, 2022-02-19:
Had the same problem
Resolved by using GPIO3 not GPIO2
Works a treat now