esphome/issues #2883: D1 Mini Pro different versions as a board option

Tokn59, 2021-12-29:
### The problem

When flashing a non-16 MB board the second time you get an error saying the board was flashed with the wrong size. There should be an option to select :

- Wemos D1 Mini Pro 4M
- Wemos D1 Mini Pro 16M
- Wemos D1 Mini
- Wemos D1 Mini V3

Thank You !!!

### Which version of ESPHome has the issue?

All latest builds

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

core-2021.12.3 - supervisor-2021.12.2

### What platform are you using?

ESP8266

### Board

D1 Mini Pro 4M

### Component causing the issue

Flash/OTA update

### Example YAML snippet

```yaml
Not relevant
```

### Anything in the logs that might be useful for us?

```txt
No
```

fmnamado, 2022-11-17:
I also have this problem. If i flash over USB ok, when I try OTA it gives me this error and doesn't push the new binary.

ERROR Error binary size: Error: ESP has been flashed with wrong flash size. Please choose the correct 'board' option (esp01_1m always works) and then flash over USB.

My d1 board says 4M in the PCB

fmnamado, 2022-11-17:
Problem solved

In the initial USB flash, I had the wrong board selected.

I saw here https://esphome.io/components/esp8266.html
which lead me there 
https://registry.platformio.org/platforms/platformio/espressif8266/boards?p=2

and I had the option

board: d1_mini_pro

which I found out it is for a board of 16MB

so I used the option

board: d1_mini

which has the 4MB has my board has, in my iniital flash
and now OTA updates works as expected :)

so what I understand, if we flash the first USB flash with wrong hardware config, the next OTA udpates won't work well, even if we change in YAML in the OTA updates, they will fail until a correct hardware config in first USB flash

hope helps someone

telefonmann, 2022-11-21:
@fmnamado Yes it helped someone. I flashed my 4MB pro board countless times over usb because I thought the OTA was broken for it. Thank you finding out and sharing!

And +1 for clearing up the board options.

fmnamado, 2022-11-21:
Thank you very much for your message, it helped me after a loooong day :)