esphome/issues #2884: Climate Temperature

amokrunner, 2021-12-29:
### The problem

Climate, thermostat platform, temperature_step doesn't work correctly.
Instead of 1 °F steps on the frontend, I get 17.2 °F steps

### Which version of ESPHome has the issue?

2021.12.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2021.12.6

### What platform are you using?

ESP8266

### Board

Shelly 1

### Component causing the issue

climate/thermostat

### Example YAML snippet

```yaml
# Example dual-point configuration entry
climate:
  - platform: thermostat
    name: "Fireplace"
    id: fireplace_thermostat
    sensor: temp
    default_target_temperature_low: 70 °F
    min_heating_off_time: 5s
    min_heating_run_time: 5s
    min_idle_time: 5s
    visual:
      min_temperature: 40 °F
      max_temperature: 100 °F
      temperature_step: 1 °F
    heat_action:
      - switch.turn_on: switch_id
    idle_action:
      - switch.turn_off: switch_id
    on_state:
      then:
        - logger.log: "State updated"
        - if:
            condition:
              lambda: !lambda 'return (id(fireplace_thermostat).mode == esphome::climate::ClimateMode::CLIMATE_MODE_OFF);'
            then:
              - logger.log: "OFF"
              - script.stop: timeout
            else:
              - logger.log: "HEAT"
              - script.execute: timeout
```

srnoth, 2021-12-30:
Try it without specifying the temperature unit for "temperature_step"

`temperature_step: 1`

amokrunner, 2022-01-03:
Thanks @srnoth , that does it. Maybe this should be fixed in the documentation,
https://esphome.io/components/climate/index.html
which shows "temperature_step: 0.1 °C"

closed 2022-09-13 by ssieb