esphome/issues #2885: Esphome Compile error

Lawrencezarb, 2021-12-30:
### The problem

I get the following message when compiling:

/config/esphome/bath_drawer.yaml: In lambda function:
/config/esphome/bath_drawer.yaml:101:26: warning: ‘void esphome::cover::Cover::stop()’ is deprecated (declared at src/esphome/components/cover/cover.h:146): stop() is deprecated, use make_call().set_command_stop() instead. [-Wdeprecated-declarations]
id(win_open).stop();
^
/config/esphome/bath_drawer.yaml:104:27: warning: ‘void esphome::cover::Cover::close()’ is deprecated (declared at src/esphome/components/cover/cover.h:140): close() is deprecated, use make_call().set_command_close() instead. [-Wdeprecated-declarations]
id(win_open).close();
^
/config/esphome/bath_drawer.yaml:109:26: warning: ‘void esphome::cover::Cover::stop()’ is deprecated (declared at src/esphome/components/cover/cover.h:146): stop() is deprecated, use make_call().set_command_stop() instead. [-Wdeprecated-declarations]
id(win_open).stop();
^
/config/esphome/bath_drawer.yaml:112:26: warning: ‘void esphome::cover::Cover::open()’ is deprecated (declared at src/esphome/components/cover/cover.h:134): open() is deprecated, use make_call().set_command_open() instead. [-Wdeprecated-declarations]
id(win_open).open();

### Which version of ESPHome has the issue?

2021.12.3

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2021.12.7

### What platform are you using?

ESP32

### Board

Wemos ESP32

### Component causing the issue

cover

### Example YAML snippet

```yaml
cover:
  - platform: template
    name: "drawer Control"
    id: win_open
    optimistic: true
    assumed_state: true
    open_action:
      - switch.turn_off: win2
      - switch.turn_on: win1
      - delay: 14s
      - switch.turn_off: win1
    close_action:
      - switch.turn_off: win1
      - switch.turn_on: win2
      - delay: 14s
      - switch.turn_off: win2
    stop_action:
      - switch.turn_off: win1
      - switch.turn_off: win2
      
binary_sensor:
  - platform: gpio
    pin: RX
    name: "Limit Switch"
    device_class: door
    on_release:
      then:
        - switch.turn_off: win2
        - switch.turn_off: win1
        
  - platform: gpio
    pin: D5
    name: "drawer Control Button"
    filters:
      - invert:
      - delayed_on_off: 50ms
    on_press: 
      then:
        - lambda: |
            if (id(win_open).position == cover::COVER_OPEN) {
              if (id(win1).state){
                // drawer is opening
                id(win_open).stop();
              } else {
                // drawer is open and not moving
                id(win_open).close();
              }
            } else {
              if (id(win2).state){
                // drawer is closing
                id(win_open).stop();
              } else {
                // win is closed and not moving
                id(win_open).open();
              }
            }
```

oxan, 2022-01-03:
Change `.stop()` in your lambdas to `make_call().set_command_stop()` and likewise for close/open.

Lawrencezarb, 2022-01-04:
Hi Oxan, I made the changes, and now there is no compiling errors, but the cover no longer opens or closes!

        - lambda: |
            if (id(win_open).position == cover::COVER_OPEN) {
              if (id(win1).state){
                // drawer is opening
                id(win_open).make_call().set_command_stop();
              } else {
                // drawer is open and not moving
                id(win_open).make_call().set_command_close();
              }
            } else {
              if (id(win2).state){
                // drawer is closing
                id(win_open).make_call().set_command_stop();
              } else {
                // win is closed and not moving
                id(win_open).make_call().set_command_open();
              }
            }

oxan, 2022-01-04:
Sorry, should've been `make_call().set_command_stop().perform()`.

Lawrencezarb, 2022-01-04:
Thanks Oxan, that seems to have fixed it.

Thanks for your help

closed 2022-01-04 by oxan