esphome/issues #2888: Cannot run esphome in dev container

arunderwood, 2021-12-30:
### The problem

When I pull down `esphome/esphome:dev` and let VSCode create the Dev Container (fully stock `devcontainer.json`), everything runs without error but the esphome doesn't appear to be installed correctly.  When I run esphome in the container I get this error:

```
root@839b9eb96ada:/workspaces/esphome# esphome
Traceback (most recent call last):
  File "/usr/bin/esphome", line 33, in <module>
    sys.exit(load_entry_point('esphome', 'console_scripts', 'esphome')())
  File "/usr/bin/esphome", line 22, in importlib_load_entry_point
    for entry_point in distribution(dist_name).entry_points
  File "/usr/lib/python3.9/importlib/metadata.py", line 524, in distribution
    return Distribution.from_name(distribution_name)
  File "/usr/lib/python3.9/importlib/metadata.py", line 187, in from_name
    raise PackageNotFoundError(name)
importlib.metadata.PackageNotFoundError: esphome
```

I can fix it by running `pip install .` to install esphome in non-editable mode, but it would be quicker if the `pip3 install -e .` command that runs in the [setup script](https://github.com/esphome/esphome/blob/607601b3a4be5f8e5941d59744f438177c015815/script/setup#L8) worked as intended.

## Steps to reproduce

1. `git clone https://github.com/esphome/esphome.git`
2. Open the directory in VSCode, let it spin up the dev container
3. In the dev container shell, run `esphome`

### Which version of ESPHome has the issue?

dev

### What type of installation are you using?

Docker

### What platform are you using?

ESP32

arunderwood, 2021-12-30:
It sounds like [this](https://github.com/esphome/issues/issues/2874#issuecomment-1002786970) should have fixed the issue but it still doesn't seem to be working correctly for me.

jesserockz, 2021-12-31:
Hmm yeah I guess the container is the same, but removing the editable flag defeats the purpose of editing in a devcontainer.

I will be looking at a better fix after I am back to work next year/week.

agners, 2022-01-02:
I tried to use older lint container by replacing the `image` `esphome/esphome-lint:dev` with `2021.10.0-dev20210920` in `.devcontainer/devcontainer.json`. However, even there I wasn't able to use `esphome` in the dev container. Also using older versions somehow did not fix the problem.

I dig a bit deeper and it seems that when using ediatble mode a file `/usr/lib/python3.9/site-packages/esphome.egg-link` is placed which links to the installation directory. It seems that Debian doesn't use that particular location for site packages (the location `/usr/lib/python3.9/site-packages/`). 

E.g. `site` is used to try to find esphome, but it doesn't search `/usr/lib/python3.9/site-packages/` location. It seems that Debian is not using that location or something along those lines.
```
root@9bb2b8bdb86e:/workspaces/esphome# python3 -c "import site; print(site.getsitepackages())"
['/usr/local/lib/python3.9/dist-packages', '/usr/lib/python3/dist-packages', '/usr/lib/python3.9/dist-packages']
```

Installing other packages with `-e` seems to properly work (bins go to `/usr/local/bin`, and the `egg-link` file gets created in `/usr/local/lib/python3.9/dist-packages/`.

I then started to nail down the difference between a simple [hello world package](https://github.com/gdamjan/hello-world-python-package) and esphome, and ended up with `pyproject.toml`: Deleting that file makes ESPHome installation in editable mode properly work again. I am not very familiar with Python packaging options and `pyproject.toml`.

agners, 2022-01-02:
This seems a relevant discussion in the Python community:
https://discuss.python.org/t/pip-19-1-and-installing-in-editable-mode-with-pyproject-toml/1553

From what I can tell we don't really use `pyproject.toml` for anything build relevant, so probably using `--no-use-pep517` is the way to go for us. I'll create a PR.

closed 2022-01-03 by jesserockz