esphome/issues #2889: UART Break: how can I send it?

pruwait, 2021-12-30:
### The problem

I need seend uart break for my device want to listen uart packet 19200 8n1.
I can do it with adruino IDE:
1. serial.end()
2. changing baudrate to 17400 8n1
2. sending 0x00
3. changing baudrate to 19200 8n1
4. sending data packet
But esphome uart haven't methods begin(), end(), break()...

How can I change baudrate?

### Which version of ESPHome has the issue?

2021.12.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

core-2021.12.7

### What platform are you using?

ESP8266

### Board

d1-mini

### Component causing the issue

uart

jesserockz, 2021-12-31:
Yes you are right, ESPHome UART does not support these methods. If there is a good reason to, then support could be added for `UARTDevice`s to be able to do this, but it would probably be quite a bit of work because of the way it is set up now.

pruwait, 2021-12-31:
This will make it easy to integrate many  uart with break devices into Hassio, such as Nice gates, Chinese air conditioners, and electricity meters.

pruwait, 2021-12-31:
2 years ago uart contained end() and begin() methods.
https://github.com/esphome/esphome/pull/1024

jesserockz, 2021-12-31:
Yeah I can see that now. Not sure when they were removed though

pruwait, 2022-01-09:
I send break + cmd this code: 
```
...
void setup (){
...
_uart =  uart_init(_UART_NO, baud_work, SERIAL_8N1, SERIAL_FULL, TX_P, 256, false);
...
}

void sendbreak_cmd() {

char br_ch = 0x00;                                               // for break
  uart_flush(_uart);                                               
  uart_set_baudrate(_uart, baud_break);               //baud = 9600
  uart_write(_uart, &br_ch, 1);                               //looong ziro = break
  //uart_write(_uart, (char *)&dummy, 1);
  uart_wait_tx_empty(_uart);                                  //waiting 
  delayMicroseconds(90);                                      //waiting more 
  uart_set_baudrate(_uart, baud_work);                 //baud=19600
  uart_write(_uart, &data[0], len);                           // body command
  uart_wait_tx_empty(_uart);                                    // waiting
}

```

How can I use   **uart_set_baudrate(_uart, baud_break)** with uart component?

idstein, 2022-12-09:
@jesserockz this feature is really required to integrate some smart meter protocols. For example IEC62056-21 mandates a baud rate changeover. Protocol initialisation is happening with a fixed baud rate, but afterwards the baud rate is negotiated between the HHU and meter device.

FYI
https://www.ungelesen.net/protagWork/media/downloads/solar-steuerung/iec62056-21%7bed1.0%7den_.pdf