esphome/issues #2890: Sonoff Pow R2 meter issue

githubcdr, 2021-12-30:
### The problem

Flashing esphome to a Sonoff Pow relay always shows 0 for the energy consumption.

### Which version of ESPHome has the issue?

2021.12.30

### What type of installation are you using?

Docker

### What platform are you using?

ESP8266

### Board

Sonoff POW

### Example YAML snippet

```yaml
esphome:
  name: $devicename
  platform: ESP8266
  board: esp01_1m
  board_flash_mode: dout

substitutions:
  devicename: sonoff-meterkast

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  fast_connect: true
  domain: .eigenhuis.lan

# Enable logging
logger:
  baud_rate: 0
  level: INFO

# Enable Home Assistant API
api:

ota:
  password: !secret ota_password
  safe_mode: True

web_server:

#prometheus:

mqtt:
  broker: mqtt.eigenhuis.lan
  discovery: False

#deep_sleep:
#  run_duration: 5s
#  sleep_duration: 60min

uart:
  rx_pin: RX
  baud_rate: 4800

time:
  - platform: sntp
    timezone: Europe/Amsterdam

binary_sensor:
  - platform: status
    name: $devicename status
  - platform: gpio
    pin:
      number: GPIO0
      mode: INPUT_PULLUP
      inverted: True
    name: $devicename Button
    on_press:
      - switch.toggle: fakebutton
  - platform: template
    name: $devicename Running
    filters:
      - delayed_off: 15s
    lambda: |-
      if (isnan(id(power).state)) {
        return {};
      } else if (id(power).state > 80) {
        // Running
        return true;
      } else {
        // Not running
        return false;
      }

sensor:
  - platform: wifi_signal
    name: $devicename WiFi Signal
    update_interval: 900s
  - platform: uptime
    name: ${devicename} Uptime
  - platform: cse7766
    update_interval: 60s
    current:
      name: ${devicename} Current
    voltage:
      name: ${devicename} Voltage
    power:
      name: ${devicename} Power
      id: power
      on_value_range:
        - above: 80.0
          then:
            - light.turn_on: led
        - below: 80.0
          then:
            - light.turn_off: led
  - platform: total_daily_energy
    name: ${devicename} Total Daily Energy
    power_id: power

switch:
  - platform: restart
    name:  ${devicename} Restart
  - platform: template
    name: ${devicename} Relay
    optimistic: true
    id: fakebutton
    turn_on_action:
    - switch.turn_on: relay
    - light.turn_on: led
    turn_off_action:
    - switch.turn_off: relay
    - light.turn_off: led
  - platform: gpio
    id: relay
    pin: GPIO12

text_sensor:
  - platform: version
    name: ${devicename} Version
  - platform: wifi_info
    ip_address:
      name: $devicename IP Address
    ssid:
      name: $devicename Connected SSID
    bssid:
      name: $devicename Connected BSSID
    mac_address:
      name: $devicename Mac Wifi Address

output:
  - platform: esp8266_pwm
    id: pow_blue_led
    pin:
      number: GPIO15
      inverted: True

light:
  - platform: monochromatic
    name: ${devicename} Blue LED
    output: pow_blue_led
    id: led
    internal: true
```

### Anything in the logs that might be useful for us?

```txt
Flashing the device to Tasmota shows the meter working again
```

arunderwood, 2022-01-01:
This config works as I'd expect on my Sonoff if that helps

```
substitutions:
# Change the disp_name to something you want  
  disp_name: livingroom
# Interval of how often the power is updated
  update_time: 60s

esphome:
  name: ${disp_name}
  platform: ESP8266
  board: esp01_1m

wifi:
  #redacted

# Enable logging
logger:
  baud_rate: 0

# Enable Home Assistant API
api:
  password: !secret api_password

ota:
  password: !secret ota_password

uart:
  rx_pin: RX
  baud_rate: 4800

binary_sensor:
  - platform: gpio
    pin:
      number: GPIO0
      mode: INPUT_PULLUP
      inverted: True
    name: ${disp_name} Button
    id: button
    on_press:
      - light.toggle: ${disp_name}_light

status_led:
  pin:
    number: GPIO13
    inverted: yes

output:
  # Note: do *not* make the relay a dimmable (PWM) signal, relays cannot handle that
  - platform: gpio
    pin: GPIO12
    id: ${disp_name}_relay
    
light:
  - platform: binary
    output: ${disp_name}_relay
    id: ${disp_name}_light
    name: ${disp_name} Light

sensor:
  - platform: cse7766
    current:
      name: ${disp_name} Current
      id: current
    voltage:
      name: ${disp_name} Voltage
      id: voltage
    power:
      name: ${disp_name} Power
      id: power
  - platform: total_daily_energy
    name: ${disp_name} Total Daily Energy
    power_id: power

time:
  - platform: sntp
    id: sntp_time 
```

githubcdr, 2022-01-05:
Thanks for the reply and the pwm comment, I'll check the results and post them here.

githubcdr, 2022-01-10:
I noticed this issue only seems to appear on some Sonoff POW R1 devices. Replacing them is easier than diving in the code for me.

closed 2022-01-10 by githubcdr