esphome/issues #2891: DSMR: Slimmelezer sends invalid data

cserhalmi-git, 2021-12-31:
### The problem

Hi, my Slimmelezer device programmed with ESPHome firmware transmits strange data as attached. Tried many ways to fix this without success. Could you please take a look whether it is a configuration- or sender- or maybe ESPHome/DSMR issue?

[uartlog.txt](https://github.com/esphome/issues/files/7795438/uartlog.txt)
[validation.txt](https://github.com/esphome/issues/files/7795440/validation.txt)

### Which version of ESPHome has the issue?

2021.12.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2021.12.6

### What platform are you using?

ESP8266

### Board

esp01_1m

### Component causing the issue

DSMR

### Example YAML snippet

```yaml
substitutions:
  device_name: slimmelezer
  device_description: "DIY P1 module to read your smart meter"
  
esphome:
  name: slimmelezer

esp8266:
  board: esp01_1m

# Enable logging
logger:
  level: VERY_VERBOSE
  baud_rate: 115200

# Enable Home Assistant API
api:

ota:
  password: "672c8b4ba6abc4b015f0349216f2d94f"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Slimmelezer Fallback Hotspot"
    password: "4C8x8eXmH1YZ"

external_components:
  - source:
      type: git
      url: https://github.com/esphome/esphome
      ref: dev
    components: [ "dsmr" ]
    refresh: 60s

uart:
  rx_pin: 13
  baud_rate: 115200
  debug:
    direction: RX
    after:
      delimiter: "\r\n"
    sequence:
      - lambda: UARTDebug::log_string(direction, bytes);

dsmr:
  max_telegram_length: 6000
  crc_check: false
  receive_timeout: 200ms

captive_portal:
```

cserhalmi-git, 2022-01-02:
I've adjusted debug output some and joined the broken lines. Seems that data fits to ELMŰ/ÉMÁSZ Hungarian energy provider specification but can not be parsed because of mismatched OBIS codes and missing bytes:

[image]

mmakaay, 2022-01-02:
You are using software serial. That might cause the missing bytes. It does not help with not implemented codes of course, but please start with the information for m:

https://esphome.io/components/sensor/dsmr.html?highlight=dsmr#improving-reader-results

cserhalmi-git, 2022-01-03:
Turned out to be a configuration problem. 

The following modifications fixed it:
- rx_buffer_size is set to 3000 to avoid overflow,
- CRC check is restored to avoid CRC bytes to be evaluated as normal telegram bytes,
- logger is switched off by setting baud rate to 0 to enable hardware UART,
- sensors definitions added to yaml to make them available in HomeAssistant.

In addition
- external dsmr component reference removed because it is no longer necessary to define max_telegram_length.

Thanks for your time and for this fast help!

closed 2022-01-03 by cserhalmi-git

theodh, 2022-01-16:
Great that the problem has been resolved! @cserhalmi-git is it possible to share your slimmelezer.yaml file?

cserhalmi-git, 2022-01-16:
Hi Theod, sure, here you are (sensors are aligned with Hungaria ELMŰ data):

```
substitutions:
  device_name: slimmelezer
  device_description: "smart meter P1 port wi-fi transmitter"
  
esphome:
  name: slimmelezer

esp8266:
  board: esp01_1m

# Enable logging
logger:
  level: VERY_VERBOSE
  baud_rate: 0

# Enable Home Assistant API
api:

ota:
  password: "672c8b4ba6abc4b015f0349216f2d94f"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Slimmelezer Fallback Hotspot"
    password: "4C8x8eXmH1YZ"

uart:
  rx_pin: GPIO13
  baud_rate: 115200
  rx_buffer_size: 3000
#  debug:
#    direction: RX
#    after:
#      delimiter: "\r\n"
#    sequence:
#     - lambda: UARTDebug::log_string(direction, bytes);

dsmr:
  max_telegram_length: 3000
  crc_check: true
  receive_timeout: 200ms

captive_portal:

sensor:
  - platform: dsmr
    energy_delivered_lux:
      name: "Hatásos import energia (+A)"
    energy_delivered_tariff1:
      name: "Hatásos import energia (+A) – tarifa 1"
    energy_delivered_tariff2:
      name: "Hatásos import energia (+A) – tarifa 2"
    energy_returned_lux:
      name: "Hatásos export energia (-A)"
    energy_returned_tariff1:
      name: "Hatásos export energia (-A) – tarifa 1"
    energy_returned_tariff2:
      name: "Hatásos export energia (-A) – tarifa 2"
    power_delivered:
      name: "Pillanatnyi import teljesítmény (+A)"
      accuracy_decimals: 3
    power_returned:
      name: "Pillanatnyi export teljesítmény (-A)"
      accuracy_decimals: 3
    electricity_failures:
      name: "Electricity Failures"
      icon: mdi:alert
    electricity_long_failures:
      name: "Long Electricity Failures"
      icon: mdi:alert
    voltage_l1:
      name: "Pillanatnyi fázis feszültség L1"
    voltage_l2:
      name: "Pillanatnyi fázis feszültség L2"
    voltage_l3:
      name: "Pillanatnyi fázis feszültség L3"
    current_l1:
      name: "Pillanatnyi áram L1"
    current_l2:
      name: "Pillanatnyi áram L2"
    current_l3:
      name: "Pillanatnyi áram L3"
  - platform: uptime
    name: "SlimmeLezer Üzemidő"
  - platform: wifi_signal
    name: "SlimmeLezer Wi-Fi jel"
    update_interval: 60s

text_sensor:
  - platform: dsmr
    identification:
      name: "COSEM logikai készüléknév"
  - platform: wifi_info
    ip_address:
      name: "SlimmeLezer IP"
    ssid:
      name: "SlimmeLezer Wi-Fi SSID"
    bssid:
      name: "SlimmeLezer Wi-Fi BSSID"
  - platform: version
    name: "ESPHome verzió"
    hide_timestamp: true

```