esphome/issues #2892: Remote receiver binary sensor updates are lost in Home Assistant

ImSorryButWho, 2021-12-31:
### The problem

Home Assistant does not seem to reliably record events from the remote receiver binary sensor type.  In my testing, approximately 50% of the time no on/off events show up in the Logbook when a code is triggered. I have the ESPHome device set up to also toggle a switch on the board, and the state of that switch is reliably reported, so communication with the device is fine.

I suspect that the extremely short time between the on and off events means that by the time the system sees the event for the change to on and checks the state, it's already been set back to off.

Obviously there are other workarounds, such as having ESPHome directly start a script in Home Assistant, but this is a method that's endorsed by the ESPHome documentation.

### Which version of ESPHome has the issue?

2022.1.0-dev

### What type of installation are you using?

pip

### Which version of Home Assistant has the issue?

2021.12.7

### What platform are you using?

ESP32

### Board

HiLetGo ESP-WROOM-32 (as generic esp32)

### Component causing the issue

remote_receiver

### Example YAML snippet

```yaml
remote_receiver:
  pin:
    number: GPIO32
    inverted: true
    mode:
      input: true
  dump: lg

switch:
  - platform: gpio
    pin: GPIO2
    name: "Blue LED"
    id: blue_led

binary_sensor:
  - platform: remote_receiver
    name: "Recent detected"
    lg:
      data: 0x20DF3EC1
      nbits: 32
    on_release:
      then:
        - switch.toggle: blue_led
```

### Anything in the logs that might be useful for us?

```txt
Logs look completely normal, but many of the changes to 'Recent detected' are not reflected in Home Assistant.  All the switch updates are.

[07:59:51][D][binary_sensor:036]: 'Recent detected': Sending state ON
[07:59:51][D][binary_sensor:036]: 'Recent detected': Sending state OFF
[07:59:51][D][switch:021]: 'Blue LED' Toggling ON.
[07:59:51][D][switch:037]: 'Blue LED': Sending state ON
[08:00:15][D][esp32.preferences:114]: Saving preferences to flash...
[08:00:16][D][binary_sensor:036]: 'Recent detected': Sending state ON
[08:00:16][D][binary_sensor:036]: 'Recent detected': Sending state OFF
[08:00:16][D][switch:021]: 'Blue LED' Toggling OFF.
[08:00:16][D][switch:037]: 'Blue LED': Sending state OFF
[08:00:17][D][binary_sensor:036]: 'Recent detected': Sending state ON
[08:00:17][D][binary_sensor:036]: 'Recent detected': Sending state OFF
[08:00:17][D][switch:021]: 'Blue LED' Toggling ON.
[08:00:17][D][switch:037]: 'Blue LED': Sending state ON
[08:00:18][D][binary_sensor:036]: 'Recent detected': Sending state ON
[08:00:18][D][binary_sensor:036]: 'Recent detected': Sending state OFF
[08:00:18][D][switch:021]: 'Blue LED' Toggling OFF.
[08:00:18][D][switch:037]: 'Blue LED': Sending state OFF
[08:00:20][D][binary_sensor:036]: 'Recent detected': Sending state ON
[08:00:20][D][binary_sensor:036]: 'Recent detected': Sending state OFF
[08:00:20][D][switch:021]: 'Blue LED' Toggling ON.
[08:00:20][D][switch:037]: 'Blue LED': Sending state ON
[08:00:21][D][binary_sensor:036]: 'Recent detected': Sending state ON
[08:00:21][D][binary_sensor:036]: 'Recent detected': Sending state OFF
[08:00:21][D][switch:021]: 'Blue LED' Toggling OFF.
[08:00:21][D][switch:037]: 'Blue LED': Sending state OFF
[08:00:22][D][binary_sensor:036]: 'Recent detected': Sending state ON
[08:00:22][D][binary_sensor:036]: 'Recent detected': Sending state OFF
[08:00:22][D][switch:021]: 'Blue LED' Toggling ON.
[08:00:22][D][switch:037]: 'Blue LED': Sending state ON
[08:00:23][D][binary_sensor:036]: 'Recent detected': Sending state ON
[08:00:23][D][binary_sensor:036]: 'Recent detected': Sending state OFF
[08:00:23][D][switch:021]: 'Blue LED' Toggling OFF.
[08:00:23][D][switch:037]: 'Blue LED': Sending state OFF
[08:00:24][D][binary_sensor:036]: 'Recent detected': Sending state ON
[08:00:24][D][binary_sensor:036]: 'Recent detected': Sending state OFF
[08:00:24][D][switch:021]: 'Blue LED' Toggling ON.
[08:00:24][D][switch:037]: 'Blue LED': Sending state ON
[08:00:25][D][binary_sensor:036]: 'Recent detected': Sending state ON
[08:00:25][D][binary_sensor:036]: 'Recent detected': Sending state OFF
[08:00:25][D][switch:021]: 'Blue LED' Toggling OFF.
[08:00:25][D][switch:037]: 'Blue LED': Sending state OFF
[08:00:25][D][binary_sensor:036]: 'Recent detected': Sending state ON
[08:00:25][D][binary_sensor:036]: 'Recent detected': Sending state OFF
[08:00:25][D][switch:021]: 'Blue LED' Toggling ON.
[08:00:25][D][switch:037]: 'Blue LED': Sending state ON
[08:00:26][D][binary_sensor:036]: 'Recent detected': Sending state ON
[08:00:26][D][binary_sensor:036]: 'Recent detected': Sending state OFF
[08:00:26][D][switch:021]: 'Blue LED' Toggling OFF.
[08:00:26][D][switch:037]: 'Blue LED': Sending state OFF
[08:00:27][D][binary_sensor:036]: 'Recent detected': Sending state ON
[08:00:27][D][binary_sensor:036]: 'Recent detected': Sending state OFF
[08:00:27][D][switch:021]: 'Blue LED' Toggling ON.
[08:00:27][D][switch:037]: 'Blue LED': Sending state ON
[08:00:29][D][binary_sensor:036]: 'Recent detected': Sending state ON
[08:00:29][D][binary_sensor:036]: 'Recent detected': Sending state OFF
[08:00:29][D][switch:021]: 'Blue LED' Toggling OFF.
[08:00:29][D][switch:037]: 'Blue LED': Sending state OFF
[08:00:30][D][binary_sensor:036]: 'Recent detected': Sending state ON
[08:00:30][D][binary_sensor:036]: 'Recent detected': Sending state OFF
[08:00:30][D][switch:021]: 'Blue LED' Toggling ON.
[08:00:30][D][switch:037]: 'Blue LED': Sending state ON
[08:00:30][D][binary_sensor:036]: 'Recent detected': Sending state ON
[08:00:30][D][binary_sensor:036]: 'Recent detected': Sending state OFF
[08:00:30][D][switch:021]: 'Blue LED' Toggling OFF.
[08:00:30][D][switch:037]: 'Blue LED': Sending state OFF
[08:00:31][D][binary_sensor:036]: 'Recent detected': Sending state ON
[08:00:31][D][binary_sensor:036]: 'Recent detected': Sending state OFF
[08:00:31][D][switch:021]: 'Blue LED' Toggling ON.
[08:00:31][D][switch:037]: 'Blue LED': Sending state ON
[08:00:32][D][binary_sensor:036]: 'Recent detected': Sending state ON
[08:00:32][D][binary_sensor:036]: 'Recent detected': Sending state OFF
[08:00:32][D][switch:021]: 'Blue LED' Toggling OFF.
[08:00:32][D][switch:037]: 'Blue LED': Sending state OFF
[08:00:33][D][binary_sensor:036]: 'Recent detected': Sending state ON
[08:00:33][D][binary_sensor:036]: 'Recent detected': Sending state OFF
[08:00:33][D][switch:021]: 'Blue LED' Toggling ON.
[08:00:33][D][switch:037]: 'Blue LED': Sending state ON
[08:00:33][D][binary_sensor:036]: 'Recent detected': Sending state ON
[08:00:33][D][binary_sensor:036]: 'Recent detected': Sending state OFF
[08:00:33][D][switch:021]: 'Blue LED' Toggling OFF.
[08:00:33][D][switch:037]: 'Blue LED': Sending state OFF
[08:00:34][D][binary_sensor:036]: 'Recent detected': Sending state ON
[08:00:34][D][binary_sensor:036]: 'Recent detected': Sending state OFF
[08:00:34][D][switch:021]: 'Blue LED' Toggling ON.
[08:00:34][D][switch:037]: 'Blue LED': Sending state ON
```

### Additional information

Moved from https://github.com/home-assistant/core/issues/63056.

I suspect the problem is a race in Home Assistant between receiving the off update and handling the state updated event.  

Pragmatically, the solution may be to delay a few ms between sending the on and off events.  I naively tried changing the `yield()` on https://github.com/esphome/esphome/blob/dev/esphome/components/remote_base/remote_base.h:283 to `delay(1000)` as a test, but the delay doesn't seem to actually happen.  I'm not familiar enough with the ESPHome environment to know why.

icarome, 2022-01-01:
You can ( and should) use a filter and add a delayed_off: 100ms. that should fix your problem. works perfectly to me

jesserockz, 2022-01-03:
While that is a valid solution, it does not solve the problem. The remote_receiver probably needs to be modified to keep it on for longer by default.