esphome/issues #2893: Faulty modbus sensor readouts

anatoly-savchenkov, 2021-12-31:
### The problem

Starting from ESPHome 2021.12 I see sporadic wrong readouts from one of my Modbus sensors (ABB E31 single phase 4 tariffs power meter). I was using Modbus Controller component since the 2021.10 release and saw no issues like this for 2 months. Before that I used custom python script to query the same things for more than a year with no garbled readouts. So, I conclude this is something in 2021.12. I use other Modbus devices but those are ok even with the December release. What's unique with ABB E31 is CRC errors coming every few seconds.  But wrong readouts happen way less frequently, something like few times a day worst case. Seems like the problem is somehow related to handling of sequential CRC errors.

### Which version of ESPHome has the issue?

2021.12.3

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2021.12.7

### What platform are you using?

ESP8266

### Board

Sonoff RE5V1C Relay

### Component causing the issue

Modbus Controller Sensor

### Example YAML snippet

```yaml
esphome:
  name: ${devicename}

  on_boot:
    priority: -100
    then:
    - lambda: |-
        id(boot_counter) += 1;

esp8266:
  board: esp01_1m
  restore_from_flash: true
  framework:
    version: latest

wifi:
  ssid: "XXXXXX"
  password: "XXXXX"
  fast_connect: true
  reboot_timeout: 15min

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: ${friendly_devicename} Hotspot
    password: "XXXXXX"

#captive_portal:

# Enable logging
logger:
  level: VERBOSE #makes uart stream available in esphome logstream
  baud_rate: 0 #disable logging over uart
  esp8266_store_log_strings_in_flash: false

# Enable boot up messages
debug:
  
# Enable Home Assistant API
api:
[... 557 lines cut ...]
[23:16:01][V][sensor:062]: 'Current': Received new state 1326.000000
[23:16:01][D][sensor:112]: 'Current': Sending state 13.26000 A with 2 decimals of accuracy
[23:16:01][D][modbus_controller.sensor:025]: Sensor new state: 3078.00
[23:16:01][V][sensor:062]: 'Power': Received new state 3078.000000
[23:16:01][D][sensor:112]: 'Power': Sending state 3078.00000 W with 0 decimals of accuracy
[23:16:01][D][modbus_controller.sensor:025]: Sensor new state: 4194.00
[23:16:01][V][sensor:062]: 'Power factor': Received new state 4194.000000
[23:16:01][D][sensor:112]: 'Power factor': Sending state 98.00000 % with 0 decimals of accuracy
[23:16:01][V][modbus_controller:027]: Sending next modbus command to device 247 register 0x502D count 1
[23:16:01][V][modbus:184]: Modbus write: F7.03.50.2D.00.01.11.95 (8)
[23:16:01][V][modbus_controller:447]: Command sent 3 0x502D 1
[23:16:01][V][modbus:058]: Modbus received Byte  247 (0Xf7)
[23:16:01][V][modbus:058]: Modbus received Byte  3 (0X3)
[23:16:01][V][modbus:058]: Modbus received Byte  2 (0X2)
[23:16:01][V][modbus:058]: Modbus received Byte  0 (0X0)
[23:16:01][V][modbus:058]: Modbus received Byte  29 (0X1d)
[23:16:01][V][modbus:058]: Modbus received Byte  176 (0Xb0)
[23:16:01][V][modbus:058]: Modbus received Byte  88 (0X58)
[23:16:01][V][modbus_controller:048]: Modbus response queued
[23:16:01][V][modbus_controller:055]: Process modbus response for address 0x502D size: 2
[23:16:01][V][modbus_controller:096]: data for register address : 0x502D : 
[23:16:01][D][modbus_controller.sensor:025]: Sensor new state: 29.00
[23:16:01][V][sensor:062]: 'ABB Temperature': Received new state 29.000000
[23:16:01][D][sensor:112]: 'ABB Temperature': Sending state 29.00000 °C with 0 decimals of accuracy
```

martgras, 2022-01-01:
Possible reasons for CRC errors 
are GND from modbus is not connected or missing terminator resistors esp. with long cables see https://electrical-engineering-portal.com/correct-cabling-modbus-rs485#termination-resistance

Another reason is uart logging interfering with modbus also communicating over the Uart

anatoly-savchenkov, 2022-01-01:
This is understood, terminator resistors are in place on both end of the line. Cable length is ~7 meters. Meter (ABB E31 412-200) has only A and B terminals, so no ground pin. Serial logging is disabled (baud_rate: 0) and it would equally affect other Modbus devices as well.

Actually, let's not fight with the root cause of CRC mismatches, they were always there for this device. Maybe device is faulty or has FW issues.

I quickly looked at the code of modbus controller. This portion of the log is especially interesting:

````
[23:16:00][V][modbus_controller:027]: Sending next modbus command to device 247 register 0x500A count 5
[23:16:00][V][modbus:184]: Modbus write: F7.03.50.0A.00.05.A0.5D (8)
[23:16:00][V][modbus_controller:447]: Command sent 3 0x500A 5
[23:16:00][D][modbus_controller:033]: Modbus command to device=247 register=0x500A countdown=0 removed from queue after send
[23:16:00][V][modbus:058]: Modbus received Byte  247 (0Xf7)
[23:16:00][V][modbus:058]: Modbus received Byte  3 (0X3)
[23:16:00][V][modbus:058]: Modbus received Byte  10 (0Xa)
[23:16:00][V][modbus:058]: Modbus received Byte  92 (0X5c)
[23:16:00][V][modbus:058]: Modbus received Byte  116 (0X74)
[23:16:00][V][modbus:058]: Modbus received Byte  5 (0X5)
[23:16:00][V][modbus:058]: Modbus received Byte  46 (0X2e)
[23:16:00][V][modbus:058]: Modbus received Byte  12 (0Xc)
[23:16:00][V][modbus:058]: Modbus received Byte  6 (0X6)
[23:16:00][V][modbus:058]: Modbus received Byte  16 (0X10)
[23:16:00][V][modbus:058]: Modbus received Byte  98 (0X62)
[23:16:00][V][modbus:058]: Modbus received Byte  19 (0X13)
[23:16:00][V][modbus:058]: Modbus received Byte  136 (0X88)
[23:16:00][V][modbus:058]: Modbus received Byte  215 (0Xd7)
[23:16:00][V][modbus:058]: Modbus received Byte  68 (0X44)
[23:16:00][V][modbus_controller:048]: Modbus response queued
[23:16:00][V][modbus_controller:055]: Process modbus response for address 0x502D size: 10
[23:16:00][V][modbus_controller:096]: data for register address : 0x502D : 
````

In send_next_command_() send_countdown reaches zero, so command_queue_.pop_front() is called and the command is dropped. At the same time, this send iteration finally succeeds without CRC errors and on_modbus_data() is called fetching the next command command_queue_.front() and attaching the response to the wrong command.

martgras, 2022-01-01:
good catch that's a possibility.  I changed the code to drop the command before sending to prevent the scenario you described.  Can't really test on my end  so can you please try 
```

external_components:
    - source:
        type: git
        url: https://github.com:/martgras/esphome
        ref: modbus-sendfix
     components: [modbus_controller]
```

anatoly-savchenkov, 2022-01-01:
Thanks, I've deployed your branch for testing.

anatoly-savchenkov, 2022-01-02:
I didn't catch the same "5 CRC errors in a row" condition yet. However, I see one anomaly which I cannot explain. What's the reason of ignoring the first response for 0x500A register and repeating a command to read 0x500A register in the log below. I see such sequences before and after the patch, sometimes I see such repetitions as frequently as CRC errors. It would be nice to extend the logging of data validation / control logic to better diagnose why response is rejected.

````
[06:34:25][V][modbus_controller:159]: Updating modbus component
[06:34:25][V][modbus_controller:126]: Range : 5002 Size: 4 (3) skip: 0
[06:34:25][V][modbus_controller:126]: Range : 500A Size: 5 (3) skip: 0
[06:34:25][V][modbus_controller:126]: Range : 502D Size: 1 (3) skip: 0
[06:34:25][V][modbus_controller:033]: Sending next modbus command to device 247 register 0x5002 count 4
[06:34:25][V][modbus:184]: Modbus write: F7.03.50.02.00.04.E0.5F (8)
[06:34:25][V][modbus_controller:448]: Command sent 3 0x5002 4
[06:34:26][V][modbus:058]: Modbus received Byte  247 (0Xf7)
[06:34:26][V][modbus:058]: Modbus received Byte  3 (0X3)
[06:34:26][V][modbus:058]: Modbus received Byte  8 (0X8)
[06:34:26][V][modbus:058]: Modbus received Byte  0 (0X0)
[06:34:26][V][modbus:058]: Modbus received Byte  21 (0X15)
[06:34:26][V][modbus:058]: Modbus received Byte  74 (0X4a)
[06:34:26][V][modbus:058]: Modbus received Byte  57 (0X39)
[06:34:26][V][modbus:058]: Modbus received Byte  0 (0X0)
[06:34:26][V][modbus:058]: Modbus received Byte  8 (0X8)
[06:34:26][V][modbus:058]: Modbus received Byte  226 (0Xe2)
[06:34:26][V][modbus:058]: Modbus received Byte  84 (0X54)
[06:34:26][V][modbus:058]: Modbus received Byte  150 (0X96)
[06:34:26][V][modbus:058]: Modbus received Byte  111 (0X6f)
[06:34:26][V][modbus_controller:049]: Modbus response queued
[06:34:26][V][modbus_controller:056]: Process modbus response for address 0x5002 size: 8
[06:34:26][V][modbus_controller:097]: data for register address : 0x5002 : 
[06:34:26][D][modbus_controller.sensor:025]: Sensor new state: 1395257.00
[06:34:26][V][sensor:062]: 'T1 Energy': Received new state 1395257.000000
[06:34:26][D][sensor:112]: 'T1 Energy': Sending state 13952.56934 kWh with 2 decimals of accuracy
[06:34:26][D][modbus_controller.sensor:025]: Sensor new state: 582228.00
[06:34:26][V][sensor:062]: 'T2 Energy': Received new state 582228.000000
[06:34:26][D][sensor:112]: 'T2 Energy': Sending state 5822.27979 kWh with 2 decimals of accuracy
[06:34:26][V][modbus_controller:033]: Sending next modbus command to device 247 register 0x500A count 5
[06:34:26][V][modbus:184]: Modbus write: F7.03.50.0A.00.05.A0.5D (8)
[06:34:26][V][modbus_controller:448]: Command sent 3 0x500A 5
[06:34:26][V][modbus:058]: Modbus received Byte  247 (0Xf7)
[06:34:26][V][modbus:058]: Modbus received Byte  3 (0X3)
[06:34:26][V][modbus:058]: Modbus received Byte  74 (0X4a)
[06:34:26][V][modbus:058]: Modbus received Byte  93 (0X5d)
[06:34:26][V][modbus:058]: Modbus received Byte  218 (0Xda)
[06:34:26][V][modbus:058]: Modbus received Byte  2 (0X2)
[06:34:26][V][modbus:058]: Modbus received Byte  175 (0Xaf)
[06:34:26][V][modbus:058]: Modbus received Byte  6 (0X6)
[06:34:26][V][modbus:058]: Modbus received Byte  41 (0X29)
[06:34:26][V][modbus:058]: Modbus received Byte  16 (0X10)
[06:34:26][V][modbus:058]: Modbus received Byte  96 (0X60)
[06:34:26][V][modbus:058]: Modbus received Byte  19 (0X13)
[06:34:26][V][modbus:058]: Modbus received Byte  136 (0X88)
[06:34:26][V][modbus:058]: Modbus received Byte  150 (0X96)
[06:34:26][V][modbus:058]: Modbus received Byte  164 (0Xa4)

>>Response ^^^above^^^ is silently ignored.

[06:34:26][V][modbus_controller:033]: Sending next modbus command to device 247 register 0x500A count 5
[06:34:26][V][modbus:184]: Modbus write: F7.03.50.0A.00.05.A0.5D (8)
[06:34:26][V][modbus_controller:448]: Command sent 3 0x500A 5
[06:34:26][V][modbus:058]: Modbus received Byte  247 (0Xf7)
[06:34:26][V][modbus:058]: Modbus received Byte  3 (0X3)
[06:34:26][V][modbus:058]: Modbus received Byte  10 (0Xa)
[06:34:26][V][modbus:058]: Modbus received Byte  93 (0X5d)
[06:34:26][V][modbus:058]: Modbus received Byte  218 (0Xda)
[06:34:26][V][modbus:058]: Modbus received Byte  2 (0X2)
[06:34:26][V][modbus:058]: Modbus received Byte  175 (0Xaf)
[06:34:26][V][modbus:058]: Modbus received Byte  6 (0X6)
[06:34:26][V][modbus:058]: Modbus received Byte  41 (0X29)
[06:34:26][V][modbus:058]: Modbus received Byte  16 (0X10)
[06:34:26][V][modbus:058]: Modbus received Byte  96 (0X60)
[06:34:26][V][modbus:058]: Modbus received Byte  19 (0X13)
[06:34:26][V][modbus:058]: Modbus received Byte  168 (0Xa8)
[06:34:26][V][modbus:058]: Modbus received Byte  150 (0X96)
[06:34:26][V][modbus:058]: Modbus received Byte  164 (0Xa4)
[06:34:26][W][modbus:096]: Modbus CRC Check failed! 7C97!=A496
````

anatoly-savchenkov, 2022-01-02:
One more interesting log where command is silently dropped:

````
[18:58:00][V][modbus_controller:167]: Updating modbus component
[18:58:00][V][modbus_controller:134]: Range : 5002 Size: 4 (3) skip: 0
[18:58:00][V][modbus_controller:134]: Range : 500A Size: 5 (3) skip: 0
[18:58:00][V][modbus_controller:134]: Range : 502D Size: 1 (3) skip: 0
[18:58:00][V][modbus_controller:035]: Sending next modbus command to device 247 register 0x5002 count 4
[18:58:00][V][modbus:184]: Modbus write: F7.03.50.02.00.04.E0.5F (8)
[18:58:00][V][modbus_controller:456]: Command sent 3 0x5002 4
[18:58:01][V][modbus:058]: Modbus received Byte  247 (0Xf7)
[18:58:01][V][modbus:058]: Modbus received Byte  3 (0X3)
[18:58:01][V][modbus:058]: Modbus received Byte  8 (0X8)
[18:58:01][V][modbus:058]: Modbus received Byte  0 (0X0)
[18:58:01][V][modbus:058]: Modbus received Byte  21 (0X15)
[18:58:01][V][modbus:058]: Modbus received Byte  87 (0X57)
[18:58:01][V][modbus:058]: Modbus received Byte  42 (0X2a)
[18:58:01][V][modbus:058]: Modbus received Byte  0 (0X0)
[18:58:01][V][modbus:058]: Modbus received Byte  8 (0X8)
[18:58:01][V][modbus:058]: Modbus received Byte  226 (0Xe2)
[18:58:01][V][modbus:058]: Modbus received Byte  137 (0X89)
[18:58:01][V][modbus:058]: Modbus received Byte  208 (0Xd0)
[18:58:01][V][modbus:058]: Modbus received Byte  120 (0X78)
[18:58:01][V][modbus_controller:055]: Modbus response queued
[18:58:01][V][modbus_controller:064]: Process modbus response for address 0x5002 size: 8
[18:58:01][V][modbus_controller:105]: data for register address : 0x5002 : 
[18:58:01][D][modbus_controller.sensor:025]: Sensor new state: 1398570.00
[18:58:01][V][sensor:062]: 'T1 Energy': Received new state 1398570.000000
[18:58:01][D][sensor:112]: 'T1 Energy': Sending state 13985.69922 kWh with 2 decimals of accuracy
[18:58:01][D][modbus_controller.sensor:025]: Sensor new state: 582281.00
[18:58:01][V][sensor:062]: 'T2 Energy': Received new state 582281.000000
[18:58:01][D][sensor:112]: 'T2 Energy': Sending state 5822.81006 kWh with 2 decimals of accuracy
[18:58:01][V][modbus_controller:035]: Sending next modbus command to device 247 register 0x500A count 5
[18:58:01][V][modbus:184]: Modbus write: F7.03.50.0A.00.05.A0.5D (8)
[18:58:01][V][modbus_controller:456]: Command sent 3 0x500A 5
[18:58:01][V][modbus:058]: Modbus received Byte  247 (0Xf7)
[18:58:01][V][modbus:058]: Modbus received Byte  3 (0X3)
[18:58:01][V][modbus:058]: Modbus received Byte  10 (0Xa)
[18:58:01][V][modbus:058]: Modbus received Byte  92 (0X5c)
[18:58:01][V][modbus:058]: Modbus received Byte  141 (0X8d)
[18:58:01][V][modbus:058]: Modbus received Byte  5 (0X5)
[18:58:01][V][modbus:058]: Modbus received Byte  15 (0Xf)

>> Why does processing of register 0x500A count 5 command stop here?

[18:58:01][V][modbus_controller:035]: Sending next modbus command to device 247 register 0x502D count 1
[18:58:01][V][modbus:184]: Modbus write: F7.03.50.2D.00.01.11.95 (8)
[18:58:01][V][modbus_controller:456]: Command sent 3 0x502D 1
[18:58:01][V][modbus:058]: Modbus received Byte  247 (0Xf7)
[18:58:01][V][modbus:058]: Modbus received Byte  3 (0X3)
[18:58:01][V][modbus:058]: Modbus received Byte  2 (0X2)
[18:58:01][V][modbus:058]: Modbus received Byte  0 (0X0)
[18:58:01][V][modbus:058]: Modbus received Byte  30 (0X1e)
[18:58:01][V][modbus:058]: Modbus received Byte  240 (0Xf0)
[18:58:01][V][modbus:058]: Modbus received Byte  89 (0X59)
[18:58:01][V][modbus_controller:055]: Modbus response queued
[18:58:01][V][modbus_controller:064]: Process modbus response for address 0x502D size: 2
[18:58:01][V][modbus_controller:105]: data for register address : 0x502D : 
[18:58:01][D][modbus_controller.sensor:025]: Sensor new state: 30.00
[18:58:01][V][sensor:062]: 'ABB Temperature': Received new state 30.000000
[18:58:01][D][sensor:112]: 'ABB Temperature': Sending state 30.00000 °C with 0 decimals of accuracy
````

martgras, 2022-01-02:
Can't explain why the response is ignored. Pushed a small logging change to add a log entry when no handler is found

anatoly-savchenkov, 2022-01-02:
I saw that change, this log was produced by the FW with than change included. Another even more interesting (unexplainable) log is below. Is it possible that log entries are simply missed? I'll try to add some local instrumentation to the code to see how execution goes.

````
[21:29:58][V][modbus_controller:167]: Updating modbus component
[21:29:58][V][modbus_controller:134]: Range : 5002 Size: 4 (3) skip: 0
[21:29:58][V][modbus_controller:134]: Range : 500A Size: 5 (3) skip: 0
[21:29:58][V][modbus_controller:134]: Range : 502D Size: 1 (3) skip: 0
[21:29:58][V][modbus_controller:035]: Sending next modbus command to device 247 register 0x5002 count 4
[21:29:58][V][modbus:184]: Modbus write: F7.03.50.02.00.04.E0.5F (8)
[21:29:58][V][modbus_controller:456]: Command sent 3 0x5002 4
[21:29:58][V][modbus:058]: Modbus received Byte  247 (0Xf7)
[21:29:58][V][modbus:058]: Modbus received Byte  3 (0X3)
[21:29:58][V][modbus:058]: Modbus received Byte  8 (0X8)
[21:29:58][V][modbus:058]: Modbus received Byte  0 (0X0)
[21:29:58][V][modbus:058]: Modbus received Byte  21 (0X15)
[21:29:58][V][modbus:058]: Modbus received Byte  90 (0X5a)
[21:29:58][V][modbus:058]: Modbus received Byte  58 (0X3a)
[21:29:58][V][modbus:058]: Modbus received Byte  0 (0X0)
[21:29:58][V][modbus:058]: Modbus received Byte  8 (0X8)
[21:29:58][V][modbus:058]: Modbus received Byte  226 (0Xe2)
[21:29:58][V][modbus:058]: Modbus received Byte  137 (0X89)
[21:29:58][V][modbus:058]: Modbus received Byte  16 (0X10)
[21:29:58][V][modbus:058]: Modbus received Byte  166 (0Xa6)
[21:29:58][V][modbus_controller:055]: Modbus response queued
[21:29:58][V][modbus_controller:064]: Process modbus response for address 0x5002 size: 8
[21:29:58][V][modbus_controller:105]: data for register address : 0x5002 : 
[21:29:58][D][modbus_controller.sensor:025]: Sensor new state: 1399354.00
[21:29:58][V][sensor:062]: 'T1 Energy': Received new state 1399354.000000
[21:29:58][D][sensor:112]: 'T1 Energy': Sending state 13993.54004 kWh with 2 decimals of accuracy
[21:29:58][D][modbus_controller.sensor:025]: Sensor new state: 582281.00
[21:29:58][V][sensor:062]: 'T2 Energy': Received new state 582281.000000
[21:29:58][D][sensor:112]: 'T2 Energy': Sending state 5822.81006 kWh with 2 decimals of accuracy
[21:29:58][V][modbus_controller:035]: Sending next modbus command to device 247 register 0x500A count 5
[21:29:58][V][modbus:184]: Modbus write: F7.03.50.0A.00.05.A0.5D (8)
[21:29:58][V][modbus_controller:456]: Command sent 3 0x500A 5
[21:29:58][V][modbus:058]: Modbus received Byte  247 (0Xf7)
[21:29:58][V][modbus:058]: Modbus received Byte  3 (0X3)
[21:29:58][V][modbus:058]: Modbus received Byte  10 (0Xa)
[21:29:58][V][modbus:058]: Modbus received Byte  92 (0X5c)
[21:29:58][V][modbus:058]: Modbus received Byte  241 (0Xf1)
[21:29:58][V][modbus:058]: Modbus received Byte  5 (0X5)
[21:29:58][V][modbus:058]: Modbus received Byte  49 (0X31)
[21:29:58][V][modbus:058]: Modbus received Byte  12 (0Xc)
[21:29:58][V][modbus:058]: Modbus received Byte  35 (0X23)
[21:29:58][V][modbus:058]: Modbus received Byte  16 (0X10)
[21:29:58][V][modbus:058]: Modbus received Byte  98 (0X62)
[21:29:58][V][modbus:058]: Modbus received Byte  19 (0X13)
[21:29:58][V][modbus:058]: Modbus received Byte  136 (0X88)
[21:29:58][V][modbus:058]: Modbus received Byte  42 (0X2a)
[21:29:58][V][modbus:058]: Modbus received Byte  212 (0Xd4)
[21:29:58][V][modbus_controller:055]: Modbus response queued
[21:29:58][V][modbus_controller:064]: Process modbus response for address 0x500A size: 10

>> Response was registered but new sensor state was not generated

[21:29:58][V][modbus_controller:035]: Sending next modbus command to device 247 register 0x502D count 1
[21:29:58][V][modbus:184]: Modbus write: F7.03.50.2D.00.01.11.95 (8)
[21:29:58][V][modbus_controller:456]: Command sent 3 0x502D 1
[21:29:58][V][modbus:058]: Modbus received Byte  247 (0Xf7)
[21:29:58][V][modbus:058]: Modbus received Byte  3 (0X3)
[21:29:58][V][modbus:058]: Modbus received Byte  130 (0X82)
[21:29:58][V][modbus:058]: Modbus received Byte  0 (0X0)
[21:29:58][V][modbus:058]: Modbus received Byte  29 (0X1d)
[21:29:58][V][modbus:058]: Modbus received Byte  176 (0Xb0)
[21:29:58][V][modbus:058]: Modbus received Byte  88 (0X58)

>> Response was ignored just like in the example from the previous comment

[21:29:58][V][modbus_controller:035]: Sending next modbus command to device 247 register 0x502D count 1
[21:29:58][V][modbus:184]: Modbus write: F7.03.50.2D.00.01.11.95 (8)
[21:29:58][V][modbus_controller:456]: Command sent 3 0x502D 1
[21:29:58][V][modbus:058]: Modbus received Byte  247 (0Xf7)
[21:29:58][V][modbus:058]: Modbus received Byte  3 (0X3)
[21:29:58][V][modbus:058]: Modbus received Byte  2 (0X2)
[21:29:58][V][modbus:058]: Modbus received Byte  0 (0X0)
[21:29:58][V][modbus:058]: Modbus received Byte  29 (0X1d)
[21:29:58][V][modbus:058]: Modbus received Byte  176 (0Xb0)
[21:29:58][V][modbus:058]: Modbus received Byte  88 (0X58)
[21:29:58][V][modbus_controller:055]: Modbus response queued
[21:29:58][V][modbus_controller:064]: Process modbus response for address 0x502D size: 2
[21:29:58][V][modbus_controller:105]: data for register address : 0x502D : 
[21:29:58][D][modbus_controller.sensor:025]: Sensor new state: 29.00
[21:29:58][V][sensor:062]: 'ABB Temperature': Received new state 29.000000
[21:29:58][D][sensor:112]: 'ABB Temperature': Sending state 29.00000 °C with 0 decimals of accuracy
[21:29:59][V][modbus_controller:167]: Updating modbus component
````

martgras, 2022-01-03:
There are some really odd responses

Command
`Modbus write: F7.03.50.0A.00.05.A0.5D `

Response:

```
[06:34:26][V][modbus_controller:448]: Command sent 3 0x500A 5
[06:34:26][V][modbus:058]: Modbus received Byte  247 (0Xf7)   == Device Address 
[06:34:26][V][modbus:058]: Modbus received Byte  3 (0X3)        == Function code
[06:34:26][V][modbus:058]: Modbus received Byte  74 (0X4a)    == Number of bytes to follow
[06:34:26][V][modbus:058]: Modbus received Byte  93 (0X5d)    == data
[06:34:26][V][modbus:058]: Modbus received Byte  218 (0Xda)  == data
```
The number of bytes can't be correct. It should be 10 (0xA) .
But oddly enough the request passed the CRC check so that must be an incorrect response from the device. 
The next response in these logs has the correct response length but fails CRC check.  The CRC code transmitted by the device is the same for both requests.  
Not sure what to conclude from that :)

anatoly-savchenkov, 2022-01-03:
How much harm such erroneous "Number of bytes to follow" field can cause? Do we use this number to read specified number of bytes from the interface?

Is it feasible to validate responses against expected size? I believe when we request certain number of registers we know exact response size, right?

martgras, 2022-01-03:
I had that validation in the code but removed it some time ago.
Some implementations return 32 bit values in 1 register
And yes number of bytes is required to know where the CRC bytes are located.
Basically modbus respone parsing is a state engine that fails if bytes are not at the correct location

anatoly-savchenkov, 2022-01-08:
I was traveling for a week and now I have gigabytes of modbus logs to look at :). ABB meter generated several 5-CRC-mismatch-in-a-row cases and I see that your fix for modbus_controller from the modbus-sendfix branch works well:
````
[12:34:06][V][modbus_controller:035]: Sending next modbus command to device 247 register 0x500A count 5
[12:34:06][V][modbus:184]: Modbus write: F7.03.50.0A.00.05.A0.5D (8)
[12:34:06][V][modbus_controller:456]: Command sent 3 0x500A 5
[12:34:06][V][modbus:058]: Modbus received Byte  247 (0Xf7)
[12:34:06][V][modbus:058]: Modbus received Byte  3 (0X3)
[12:34:06][V][modbus:058]: Modbus received Byte  10 (0Xa)
[12:34:06][V][modbus:058]: Modbus received Byte  92 (0X5c)
[12:34:06][V][modbus:058]: Modbus received Byte  13 (0Xd)
[12:34:06][V][modbus:058]: Modbus received Byte  3 (0X3)
[12:34:06][V][modbus:058]: Modbus received Byte  84 (0X54)
[12:34:06][V][modbus:058]: Modbus received Byte  7 (0X7)
[12:34:06][V][modbus:058]: Modbus received Byte  166 (0Xa6)
[12:34:06][V][modbus:058]: Modbus received Byte  16 (0X10)
[12:34:06][V][modbus:058]: Modbus received Byte  98 (0X62)
[12:34:06][V][modbus:058]: Modbus received Byte  19 (0X13)
[12:34:06][V][modbus:058]: Modbus received Byte  136 (0X88)
[12:34:06][V][modbus:058]: Modbus received Byte  2 (0X2)
[12:34:06][V][modbus:058]: Modbus received Byte  91 (0X5b)
[12:34:06][W][modbus:096]: Modbus CRC Check failed! 5902!=5B02
[12:34:07][D][modbus_controller:029]: Modbus command to device=247 register=0x500A countdown=0 no response received - removed from send queue
[12:34:07][V][modbus_controller:035]: Sending next modbus command to device 247 register 0x502D count 1
````
Just to be clear, I was testing the version without changes to modbus.cpp. So, I propose pushing modbus_controller.cpp changes to the main repo and closing of this issue.

Meanwhile I'm switching to modbus-queue branch and will test it with all my devices.

martgras, 2022-01-09:
the sendfix is included in https://github.com/esphome/esphome/pull/2981 that was  just merged into dev.

closed 2022-01-09 by martgras