esphome/issues #2896: Regular device reboots

varac, 2022-01-02:
### The problem

I have 2 esphome devices ([Gosund P1](https://community.home-assistant.io/t/gosund-p1-power-strip-with-power-monitoring/184820/33?u=varac) and an ESP-01S+DHT22 sensor) which keep on rebooting on a regular base every 10-15 minutes, as seen on uptime prometheus graph below:

[image]

Both are not (yet) connected to Home-Assistant

For simplicity I only describe the ESP-01S+DHT22 config heree.

### Which version of ESPHome has the issue?

2021.12.2

### What type of installation are you using?

pip

### Which version of Home Assistant has the issue?

-

### What platform are you using?

ESP8266

### Board

esp01_1m

### Component causing the issue

I don't know

### Example YAML snippet

```yaml
ESP-01S+DHT22 config: 

substitutions:
  devicename: temphum-l1
  location: l1
esphome:
  name: temphum-l1
  build_path: .esphome/build/temphum-l1
  platformio_options: {}
  includes: []
  libraries: []
  name_add_mac_suffix: false
esp8266:
  board: esp01_1m
  framework:
    version: 2.7.4
    source: ~3.20704.0
    platform_version: platformio/espressif8266 @ 2.6.3
  restore_from_flash: false
  board_flash_mode: dout
sensor:
- platform: dht
  model: DHT22
  pin:
    number: 2
    mode:
      input: true
      analog: false
      output: false
      open_drain: false
      pullup: false
      pulldown: false
    inverted: false
  temperature:
    name: L1 Temperature
    disabled_by_default: false
    force_update: false
    unit_of_measurement: °C
    accuracy_decimals: 1
    device_class: temperature
    state_class: measurement
  humidity:
    name: L1 Humidity
    disabled_by_default: false
    force_update: false
    unit_of_measurement: '%'
    accuracy_decimals: 0
    device_class: humidity
    state_class: measurement
  update_interval: 60s
- platform: uptime
  name: Uptime
  disabled_by_default: false
  force_update: false
  unit_of_measurement: s
  icon: mdi:timer-outline
  accuracy_decimals: 0
  state_class: total_increasing
  entity_category: diagnostic
  update_interval: 60s
- platform: wifi_signal
  name: Signal Strength
  disabled_by_default: false
  force_update: false
  unit_of_measurement: dBm
  accuracy_decimals: 0
  device_class: signal_strength
  state_class: measurement
  entity_category: diagnostic
  update_interval: 60s
logger:
  baud_rate: 115200
  tx_buffer_size: 512
  deassert_rts_dtr: false
  hardware_uart: UART0
  level: DEBUG
  logs: {}
  esp8266_store_log_strings_in_flash: true
api:
  password: ''
  port: 6053
  reboot_timeout: 15min
ota:
  password: ''
  safe_mode: true
  port: 8266
  reboot_timeout: 5min
  num_attempts: 10
web_server:
  port: 80
  css_url: https://esphome.io/_static/webserver-v1.min.css
  js_url: https://esphome.io/_static/webserver-v1.min.js
  include_internal: false
  ota: true
prometheus: {}
text_sensor:
- platform: wifi_info
  ip_address:
    name: IP Address
    disabled_by_default: false
    entity_category: diagnostic
wifi:
  ap:
    ssid: Esphome Fallback Hotspot
    password: !secret 'wifi_fallback_pass'
    ap_timeout: 1min
  domain: .local
  reboot_timeout: 15min
  power_save_mode: NONE
  fast_connect: false
  output_power: 20.0
  networks:
  - ssid: !secret 'wifi_l1_ssid'
    password: !secret 'wifi_l1_pass'
    priority: 0.0
  use_address: temphum-l1.local
```
```

### Anything in the logs that might be useful for us?

```txt
Will add later
```

varac, 2022-01-02:
Fun, when I was looking at the configs I realized that I had both devices the `Gosund P1 Fallback Hotspot` ap ssid, which I changed to `Esphome Fallback Hotspot`, updated both firmwares and now both devices keep beeing up at least for the last 30min: 

[image]

I'll have a look later if they stay up.

varac, 2022-01-03:
I was too hopeful, the devices are rebooting again in a similar interval as desribed above.
Here's the log of the ESP-01S+DHT22 one:

```
[13:25:41][I][app:102]: ESPHome version 2021.12.2 compiled on Jan  2 2022, 11:33:48
[13:25:41][C][wifi:488]: WiFi:
[13:25:41][C][wifi:350]:   Local MAC: 98:CD:AC:23:79:C9
[13:25:41][C][wifi:351]:   SSID: '90er'
[13:25:41][C][wifi:352]:   IP Address: 192.168.8.183
[13:25:41][C][wifi:354]:   BSSID: 94:83:C4:0E:F0:19
[13:25:41][C][wifi:355]:   Hostname: 'temphum-l1'
[13:25:41][C][wifi:357]:   Signal strength: -39 dB ▂▄▆█
[13:25:41][C][wifi:361]:   Channel: 1
[13:25:41][C][wifi:362]:   Subnet: 255.255.255.0
[13:25:41][C][wifi:363]:   Gateway: 192.168.8.1
[13:25:41][C][wifi:364]:   DNS1: 192.168.8.1
[13:25:41][C][wifi:365]:   DNS2: 0.0.0.0
[13:25:41][C][logger:233]: Logger:
[13:25:41][C][logger:234]:   Level: DEBUG
[13:25:41][C][logger:235]:   Log Baud Rate: 115200
[13:25:41][C][logger:236]:   Hardware UART: UART0
[13:25:41][C][uptime.sensor:031]: Uptime Sensor 'Uptime'
[13:25:41][C][uptime.sensor:031]:   State Class: 'total_increasing'
[13:25:41][C][uptime.sensor:031]:   Unit of Measurement: 's'
[13:25:41][C][uptime.sensor:031]:   Accuracy Decimals: 0
[13:25:41][C][uptime.sensor:031]:   Icon: 'mdi:timer-outline'
[13:25:41][C][dht:017]: DHT:
[13:25:41][C][dht:018]:   Pin: GPIO2
[13:25:41][C][dht:024]:   Model: DHT22 (or equivalent)
[13:25:41][C][dht:027]:   Update Interval: 60.0s
[13:25:41][C][dht:029]:   Temperature 'L1 Temperature'
[13:25:41][C][dht:029]:     Device Class: 'temperature'
[13:25:41][C][dht:029]:     State Class: 'measurement'
[13:25:41][C][dht:029]:     Unit of Measurement: '°C'
[13:25:41][C][dht:029]:     Accuracy Decimals: 1
[13:25:41][C][dht:030]:   Humidity 'L1 Humidity'
[13:25:41][C][dht:030]:     Device Class: 'humidity'
[13:25:41][C][dht:030]:     State Class: 'measurement'
[13:25:41][C][dht:030]:     Unit of Measurement: '%'
[13:25:41][C][dht:030]:     Accuracy Decimals: 0
[13:25:41][C][web_server:162]: Web Server:
[13:25:41][C][web_server:163]:   Address: temphum-l1.local:80
[13:25:41][C][ota:082]: Over-The-Air Updates:
[13:25:41][C][ota:083]:   Address: temphum-l1.local:8266
[13:25:41][C][api:134]: API Server:
[13:25:41][C][api:135]:   Address: temphum-l1.local:6053
[13:25:41][C][api:139]:   Using noise encryption: NO
[13:25:41][C][wifi_signal.sensor:009]: WiFi Signal 'Signal Strength'
[13:25:41][C][wifi_signal.sensor:009]:   Device Class: 'signal_strength'
[13:25:41][C][wifi_signal.sensor:009]:   State Class: 'measurement'
[13:25:41][C][wifi_signal.sensor:009]:   Unit of Measurement: 'dBm'
[13:25:41][C][wifi_signal.sensor:009]:   Accuracy Decimals: 0
[13:25:41][C][wifi_info:009]: WifiInfo IPAddress 'IP Address'
[13:25:41][C][mdns:084]: mDNS:
[13:25:41][C][mdns:085]:   Hostname: temphum-l1
[13:26:01][D][dht:048]: Got Temperature=22.5°C Humidity=61.9%
[13:26:01][D][sensor:113]: 'L1 Temperature': Sending state 22.50000 °C with 1 decimals of accuracy
[13:26:01][D][sensor:113]: 'L1 Humidity': Sending state 61.90000 % with 0 decimals of accuracy
[13:26:07][D][sensor:113]: 'Uptime': Sending state 879.93799 s with 0 decimals of accuracy
[13:26:14][D][sensor:113]: 'Signal Strength': Sending state -33.00000 dBm with 0 decimals of accuracy
[13:27:01][D][dht:048]: Got Temperature=22.6°C Humidity=61.5%
[13:27:01][D][sensor:113]: 'L1 Temperature': Sending state 22.60000 °C with 1 decimals of accuracy
[13:27:01][D][sensor:113]: 'L1 Humidity': Sending state 61.50000 % with 0 decimals of accuracy
[13:27:07][D][sensor:113]: 'Uptime': Sending state 939.94299 s with 0 decimals of accuracy
[13:27:14][D][sensor:113]: 'Signal Strength': Sending state -36.00000 dBm with 0 decimals of accuracy
[13:28:01][D][dht:048]: Got Temperature=22.5°C Humidity=61.2%
[13:28:01][D][sensor:113]: 'L1 Temperature': Sending state 22.50000 °C with 1 decimals of accuracy
[13:28:01][D][sensor:113]: 'L1 Humidity': Sending state 61.20000 % with 0 decimals of accuracy
[13:28:07][D][sensor:113]: 'Uptime': Sending state 999.93799 s with 0 decimals of accuracy
[13:28:14][D][sensor:113]: 'Signal Strength': Sending state -38.00000 dBm with 0 decimals of accuracy
[13:29:01][D][dht:048]: Got Temperature=22.6°C Humidity=61.0%
[13:29:01][D][sensor:113]: 'L1 Temperature': Sending state 22.60000 °C with 1 decimals of accuracy
[13:29:01][D][sensor:113]: 'L1 Humidity': Sending state 61.00000 % with 0 decimals of accuracy
[13:29:07][D][sensor:113]: 'Uptime': Sending state 1059.94202 s with 0 decimals of accuracy
[13:29:14][D][sensor:113]: 'Signal Strength': Sending state -39.00000 dBm with 0 decimals of accuracy
...
```

I didn't catch a reboot and am away from the device, but from looking at earlier logs I can tell that no relevant information can be seen during a reboot.

gonzalop, 2022-01-04:
I had a similar problem and resolved it by disabling the HomeAssistant API. There was a message in the debug log from the device saying that it was rebooting because there was no connection to the API port within some time interval. It seems you have that API enabled:
```
api:
  password: ''
  port: 6053
  reboot_timeout: 15min
```
If your HomeAssistant instance is not connecting to the device, setting the reboot_timeout to 0s, commenting out, or removing those lines will make the regular reboots go away. The [docs](https://esphome.io/components/api.html#configuration-variables) explain why.

varac, 2022-01-06:
@gonzalop Ahh, now I get it, thanks! That makes sense.

closed 2022-01-06 by varac