esphome/issues #2897: Wifi problems on esp32

hneel, 2022-01-02:
### The problem

I have a couple of [url=https://www.dfrobot.com/product-1798.html]these DFRobot Beetle ESP32[/url] boards and I noticed Wifi is slow/unstable with them. 

In the past I have used regular Arduino sketches and there were no Wifi problems then. I have tried one board that I hade been using earlier and also a new one straight outof the box, and both had the same problems.

Sometimes Wifi doesn't start at all, sometimes it does, but then it's slow/unstable. Firmware updates over Wifi always fail, either at start or halfway.
I’ve also tried many different settings inside the sketch. manual_ip, power_save_mode: none, fast_connect: true/false, various board types. So far, nothing helps.

BTW, I have no problems using Esphome using other (Esp8266) boards.

### Which version of ESPHome has the issue?

2021.12.1 

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

core-2021.12.7

### What platform are you using?

ESP32

### Board

DFRobotot Beetle ESP32

### Example YAML snippet

```yaml
esphome:
  name: ir-remote

esp32:
  board: esp32dev
  framework:
    type: arduino

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:
  password: "dde33969d5956d7fe1d05f981ac4fa20"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  power_save_mode: none

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Ir-Remote Fallback Hotspot"
    password: "rSZIZoN1fFzz"

captive_portal:

# Enable Web
web_server:
  port: 80
  
  
light:
  - platform: binary
[... 253 lines cut ...]
[C][wifi:363]:   Gateway: 10.0.0.98
[C][wifi:364]:   DNS1: 10.0.0.98
[C][wifi:365]:   DNS2: 0.0.0.0
[C][logger:233]: Logger:
[C][logger:234]:   Level: DEBUG
[C][logger:235]:   Log Baud Rate: 115200
[C][logger:236]:   Hardware UART: UART0
[C][captive_portal:144]: Captive Portal:
[C][web_server:162]: Web Server:
[C][web_server:163]:   Address: ir-remote.local:80
[C][ota:082]: Over-The-Air Updates:
[C][ota:083]:   Address: ir-remote.local:3232
[C][ota:086]:   Using Password.
[C][api:134]: API Server:
[C][api:135]:   Address: ir-remote.local:6053
[C][api:139]:   Using noise encryption: NO
[C][mdns:084]: mDNS:
[C][mdns:085]:   Hostname: ir-remote
```

### Additional information

[image]

(GPIO2 is the LED. If I remove the light and output sections from the above yaml sketch, the warning goes away, but the problem remains)

hneel, 2022-01-02:
I've put debug to very_verbose:

[D][esp32.preferences:114]: Saving preferences to flash...
[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 1 - SCAN_DONE
[V][wifi_esp32:454]: Event: WiFi Scan Done status=0 number=7 scan_id=128
[D][wifi:385]: Found networks:
[I][wifi:429]: - 'MySSID' [redacted]▂▄▆█
[D][wifi:430]:     Channel: 6
[D][wifi:431]:     RSSI: -53 dB
[I][wifi:429]: - 'MySSID' [redacted]▂▄▆█
[D][wifi:430]:     Channel: 1
[D][wifi:431]:     RSSI: -91 dB
[I][wifi:429]: - 'MySSID' [redacted]▂▄▆█
[D][wifi:430]:     Channel: 1
[D][wifi:431]:     RSSI: -92 dB
[I][wifi:429]: - 'MySSID' [redacted]▂▄▆█
[D][wifi:430]:     Channel: 11
[D][wifi:431]:     RSSI: -94 dB
[D][wifi:434]: - [redacted] [redacted]▂▄▆█
[D][wifi:434]: - [redacted] [redacted]▂▄▆█
[D][wifi:434]: - [redacted] [redacted]▂▄▆█
[I][wifi:245]: WiFi Connecting to 'MySSID'...
[V][wifi:247]: Connection Params:
[V][wifi:248]:   SSID: 'MySSID'
[V][wifi:251]:   BSSID: [redacted]
[V][wifi:271]:   Password: [redacted]
[V][wifi:276]:   Channel: 6
[V][wifi:285]:   Using DHCP IP
[V][wifi:287]:   Hidden: NO
[VV][esp-idf:000]: I (5287) wifi:
[VV][esp-idf:000]: new:<6,0>, old:<1,0>, ap:<255,255>, sta:<6,0>, prof:1
[VV][esp-idf:000]: 

[VV][esp-idf:000]: I (6033) wifi:
[VV][esp-idf:000]: state: init -> auth (b0)
[VV][esp-idf:000]: 

[VV][esp-idf:000]: I (7035) wifi:
[VV][esp-idf:000]: state: auth -> init (200)
[VV][esp-idf:000]: 

[VV][esp-idf:000]: I (7036) wifi:
[VV][esp-idf:000]: new:<6,0>, old:<6,0>, ap:<255,255>, sta:<6,0>, prof:1
[VV][esp-idf:000]: 

[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:391] _eventCallback(): Reason: 2 - AUTH_EXPIRE
[W][wifi_esp32:495]: Event: Disconnected ssid='MySSID' bssid=[redacted] reason='Auth Expired'
[W][wifi:533]: Error while connecting to network.
[W][wifi:568]: Restarting WiFi adapter...
[V][wifi_esp32:041]: Disabling STA.
[VV][esp-idf:000]: I (7089) wifi:
[VV][esp-idf:000]: flush txq
[VV][esp-idf:000]:

Sangoku, 2022-02-18:
Can Confirm this. My board does the same.  

i have a  WeMos TTgo ESP32 Uno D1 R32 and it spits the same error when i enter the board type as 

esp32:
  board: esp32doit-devkit-v1 

BUT
when i do it with this board it works...

esp32:
  board: nodemcu-32s

Maybe that is your problem. Wrong board type?

fuhrysteve, 2022-03-07:
I had this issue with a TTGO board. What worked for me was adding `power_save_mode: none`

```yaml
wifi:    
  ssid: "myssid"
  password: "mypass"    
  manual_ip:    
    static_ip: x.x.x.x    
    gateway: x.x.x.x
    subnet: x.x.x.x    
    dns1: x.x.x.x    
  fast_connect: true       
  power_save_mode: none
```

Sangoku, 2022-03-07:
> I had this issue with a TTGO board. What worked for me was adding `power_save_mode: none`
>
> ```yaml
> wifi:
>   ssid: "myssid"
>   password: "mypass"
>   manual_ip:
>     static_ip: x.x.x.x [...]

This solved my problem with reconnecting after update.