esphome/issues #2899: Exclude only one switch from logging

Protoncek, 2022-01-02:
### The problem

I have a problem/question: 
i have a switch defined which makes LED flashing if wifi is connected. Because of that i have pretty spammed log by sending this switch on and off position twice per second into logger.

I'd like to exclude only thie switch component from logging (like "switch.led_wifi"). So far i only managed to exclude ALL switch actions with this:

logger:
  logs:
    switch: ERROR

But that's not what i'd want, and nothing else works. If i state only my entity, say "switch.led_wifi" , or "led_wifi" ... it doesn't filter it out. 
Is there any possibility to eliminate only one entry from logging or is the only option to rule out all switches?
Thanks!

### Which version of ESPHome has the issue?

2021.12.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2021.12.7

### What platform are you using?

ESP32

### Board

WROOM32U

### Component causing the issue

Switch component

OttoWinter, 2022-01-18:
> Is there any possibility to eliminate only one entry from logging or is the only option to rule out all switches?

No that is not possible, and I don't think we should support that either. The logging system is intentionally simple, because it's a critical component that should never break.

closed 2022-01-18 by OttoWinter

Protoncek, 2022-01-18:
OK, thanks. In that case my last question would be: is it any harm if i leave logging on thus having an event every second? Will affect module's efficiency/speed? 
I mean, beside the fact that if i'll want to debug then whole log will be a mess...