esphome/issues #2900: Could not use ADC GPIO on ESP32-DOIT-DEVKIT-V1(30pin)

huishizhao, 2022-01-03:
### The problem

I have a ESP-WROOM-32 board with 30 GPIO pins, I checked the Pin out was same as ESP32-DOIT-DEVKIT-V1(30Pin), I would like to use GPIO13 and 15 to collect soil moisture, the config error  to ESP32 doesn't support ADC on this pin.
[image]

 I check all board configuration in ESPHOME, as document description, ADC GPIO only allows GPIO 32-39. 

### Which version of ESPHome has the issue?

2021.11.4

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2021.12.4

### What platform are you using?

ESP32

### Board

esp32doit-devkit-v1(30Pin)

### Component causing the issue

adc

### Example YAML snippet

```yaml
esphome:
  name: soil-moisture
  platform: ESP32
  board: esp32doit-devkit-v1

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:
  password: "cb6d103aa9453606479d64d9291f5c16"

wifi:
  ssid: "TP-LINK_JESSICA"
  password: "zhaoyufei060720"

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Soil-Moisture Fallback Hotspot"
    password: "LWFqQEmUl2bv"

captive_portal:
sensor:
  - platform: adc
    pin: GPIO13
    name: "soilmoisturesensor1"
    update_interval: 10s
```

### Anything in the logs that might be useful for us?

```txt
INFO Reading configuration /config/esphome/soil-moisture.yaml...
Failed config

sensor.adc: [source /config/esphome/soil-moisture.yaml:26]
  platform: adc
  
  ESP32 doesn't support ADC on this pin.
  pin: GPIO13 [source /config/esphome/soil-moisture.yaml:27]
  name: soilmoisturesensor1
  update_interval: 10s
```

martgras, 2022-01-03:
Connect them to PIN 12  (GPIO34) and and PIN 11 (GPIO35) is an option 
PIN 9 - 15 provides access to GPIO 32 - 39.

OttoWinter, 2022-01-18:
GPIO13 and GPIO15 are connected to a ADC channels, but they're on ADC2 which is shared by the wifi system, so when wifi is enabled you won't get any meaningful data from it. That's why ESPHome prevents access on those pins.

closed 2022-01-18 by OttoWinter