esphome/issues #2901: esp-idf issue

vautieri, 2022-01-03:
### The problem

Compiling /data/armour/.pioenvs/armour/app_trace/gcov/gcov_rtio.o
/config/esphome/armour.yaml:803:3: error: multi-line comment [-Werror=comment]
/config/esphome/armour.yaml:813:3: error: multi-line comment [-Werror=comment]
/config/esphome/armour.yaml:967:3: error: multi-line comment [-Werror=comment]
/config/esphome/armour.yaml:995:3: error: multi-line comment [-Werror=comment]
Compiling /data/armour/.pioenvs/armour/app_update/esp_ota_ops.o

### Which version of ESPHome has the issue?

v2021.12.3

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP32-IDF

### Board

heltec_wifi_kit_32

### Example YAML snippet

```yaml
The line number with the error does not even come close to lining up with anything in my yaml.  Note, I did notice if I compile with arduino, these are warnings but warnings are not set to fail.  However, it appears you have warnings set to be errors by default for esp32-idf?  I don't know how to disable the warning to be an error via the yaml.. (I didn't look real hard either).  Anyways, I saw this was an issue on  v2021.12.1 which I thought it was suppose to be fixed in the "next" version, which I thought this would be.  Its not a show stopper as I can still use Arduino, but I really want to change over...
```

martgras, 2022-01-03:
This was fixed in https://github.com/esphome/esphome/pull/2891.   
Right now it's only in the dev branch but should be in the next release

OttoWinter, 2022-01-18:
Closing per https://github.com/esphome/esphome/pull/2891

closed 2022-01-18 by OttoWinter