esphome/issues #2902: Tuya Fan Component DP3 for speed appears to have different types between makes

jeggleston1981, 2022-01-03:
### The problem

In a previous issue data point, 3 for fan speed control was changed from int to enum. But the fan I have uses int.  It would seem there is variation between makes/brands(annoying), is it possible to make it an option whether it is an int or enum type. The fan is a Ventair SkyDC fan and is quite widely available.
I noted elsewhere that DP4 for speed should be an int and DP3 should be a enum. I not sure to ensure future compatibility if it is viable  to have an option for enum speed or int speed type to which either DP3 or DP4 can be assigned.

Previous Fix - Break the Tuya set_datapoint_value method into separate methods per datapoint type #2059

### Which version of ESPHome has the issue?

2021.11.4

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP8266

### Board

ESP8285

### Component causing the issue

tuya/fan

### Example YAML snippet

```yaml
tuya:
fan:
  - platform: "tuya"
    name: "Main Bedroom Fan"
    switch_datapoint: 1
    speed_datapoint: 3
    direction_datapoint: 8
    speed_count: 5
```

### Anything in the logs that might be useful for us?

```txt
[12:08:57][C][tuya.fan:055]: Tuya Fan:
[12:08:57][C][tuya.fan:056]:   Speed count 5
[12:08:57][C][tuya.fan:058]:   Speed has datapoint ID 3
[12:08:57][C][tuya.fan:060]:   Switch has datapoint ID 1
[12:08:57][C][tuya.fan:064]:   Direction has datapoint ID 8
[12:08:57][C][ota:082]: Over-The-Air Updates:
[12:08:57][C][ota:083]:   Address: master-fan-tuya.local:8266
[12:08:57][C][api:134]: API Server:
[12:08:57][C][api:135]:   Address: master-fan-tuya.local:6053
[12:08:57][C][api:139]:   Using noise encryption: NO
[12:08:57][C][mdns:084]: mDNS:
[12:08:57][C][mdns:085]:   Hostname: master-fan-tuya
[12:08:57][C][tuya:028]: Tuya:
[12:08:57][C][tuya:039]:   Datapoint 1: switch (value: ON)
[12:08:57][C][tuya:045]:   Datapoint 8: enum (value: 0)
[12:08:57][C][tuya:045]:   Datapoint 2: enum (value: 0)
[12:08:57][C][tuya:041]:   Datapoint 3: int value (value: 4)
[12:08:57][C][tuya:045]:   Datapoint 22: enum (value: 0)
[12:08:57][C][tuya:055]:   Product: '{"p":"vzj97d3m05yjhchn","v":"1.0.0","m":2}'
[12:09:17][D][tuya:502]: Setting datapoint 1 to 1
[12:09:17][D][tuya:502]: Setting datapoint 8 to 0
[12:09:17][D][tuya:502]: Setting datapoint 3 to 1
[12:09:17][E][tuya:507]: Attempt to set datapoint 3 with incorrect type
```

### Additional information

In the Tuya Developers debugging it lists datapoint 3 for speed as an int
{
        "code": "fan_speed_percent",
        "dp_id": 3,
        "type": "Integer",
        "values": "{\"min\":\"1\",\"max\":\"5\",\"scale\":\"0\",\"step\":\"1\"}"
      },

brywithawhy, 2022-02-11:
@jeggleston1981 did you find a fix for this? I'm experiencing the same issue.
My speed is on datapoint 2 which is listed as an int.

My logs:

> [15:49:22][C][tuya.fan:055]: Tuya Fan:
> [15:49:22][C][tuya.fan:056]:   Speed count 24
> [15:49:22][C][tuya.fan:058]:   Speed has datapoint ID 2
> [15:49:22][C][tuya.fan:060]:   Switch has datapoint ID 1
> [15:49:22][C][ota:082]: Over-The-Air Updates:
> [15:49:22][C][ota:083]:   Address: fan1.local:8266
> [15:49:22][C][ota:086]:   Using Password.
> [15:49:22][C][api:134]: API Server:
> [15:49:22][C][api:135]:   Address: fan1.local:6053
> [15:49:22][C][api:139]:   Using noise encryption: NO
> [15:49:22][C][mdns:084]: mDNS:
> [15:49:22][C][mdns:085]:   Hostname: fan1
> [15:49:22][V][mdns:086]:   Services:
> [15:49:22][V][mdns:088]:   - _esphomelib, _tcp, 6053
> [15:49:22][V][mdns:090]:     TXT: version = 2022.1.3
> [15:49:22][V][mdns:090]:     TXT: mac = e09806b688e7
> [15:49:22][V][mdns:090]:     TXT: platform = ESP8266
> [15:49:22][V][mdns:090]:     TXT: board = esp01_1m
> [15:49:22][C][tuya:028]: Tuya:
> [15:49:22][C][tuya:039]:   Datapoint 1: switch (value: ON)
> [15:49:22][C][tuya:041]:   Datapoint 2: int value (value: 10)
> [15:49:22][C][tuya:041]:   Datapoint 3: int value (value: 0)
> [15:49:22][C][tuya:041]:   Datapoint 4: int value (value: 0)
> [15:49:22][C][tuya:041]:   Datapoint 5: int value (value: 0)
> [15:49:22][C][tuya:041]:   Datapoint 6: int value (value: 2355)
> [15:49:22][C][tuya:041]:   Datapoint 7: int value (value: 0)
> [15:49:22][C][tuya:055]:   Product: '{"p":"pUZjqSipclQ0ACyY","v":"1.0.0","m":0}'
> [15:49:27][V][tuya:351]: Sending Tuya: CMD=0x00 VERSION=0 DATA=[] INIT_STATE=5
> [15:49:27][V][tuya:107]: Received Tuya: CMD=0x00 VERSION=3 DATA=[01] INIT_STATE=5
> [15:49:27][V][tuya:132]: MCU Heartbeat (0x01)
> [15:49:29][V][tuya:107]: Received Tuya: CMD=0x07 VERSION=3 DATA=[01.01.00.01.01 (5)] INIT_STATE=5
> [15:49:29][D][tuya:264]: Datapoint 1 update to ON
> [15:49:29][V][tuya.fan:028]: MCU reported switch is: ON
> [15:49:29][V][tuya.fan:069]: Setting switch: ON
> [15:49:29][D][tuya:499]: Setting datapoint 1 to 1
> [15:49:29][V][tuya:507]: Not sending unchanged value
> [15:49:29][V][tuya.fan:082]: Setting speed: 11
> [15:49:29][D][tuya:499]: Setting datapoint 2 to 10
> [15:49:29][E][tuya:504]: Attempt to set datapoint 2 with incorrect type
> [15:49:29][V][tuya:107]: Received Tuya: CMD=0x07 VERSION=3 DATA=[02.02.00.04.00.00.00.0A (8)] INIT_STATE=5
> [15:49:29][D][tuya:272]: Datapoint 2 update to 10
> [15:49:29][V][tuya.fan:017]: MCU reported speed of: 10
> [15:49:29][V][tuya.fan:069]: Setting switch: ON
> [15:49:29][D][tuya:499]: Setting datapoint 1 to 1
> [15:49:29][V][tuya:507]: Not sending unchanged value
> [15:49:29][V][tuya.fan:082]: Setting speed: 11
> [15:49:29][D][tuya:499]: Setting datapoint 2 to 10
> [15:49:29][E][tuya:504]: Attempt to set datapoint 2 with incorrect type
> [15:49:29][V][tuya:107]: Received Tuya: CMD=0x07 VERSION=3 DATA=[03.02.00.04.00.00.00.00 (8)] INIT_STATE=5
> [15:49:29][D][tuya:272]: Datapoint 3 update to 0
> [15:49:29][V][tuya:107]: Received Tuya: CMD=0x07 VERSION=3 DATA=[04.02.00.04.00.00.00.00 (8)] INIT_STATE=5
> [15:49:29][D][tuya:272]: Datapoint 4 update to 0
> [15:49:29][V][tuya:107]: Received Tuya: CMD=0x07 VERSION=3 DATA=[05.02.00.04.00.00.00.00 (8)] INIT_STATE=5
> [15:49:29][D][tuya:272]: Datapoint 5 update to 0
> [15:49:29][V][tuya:107]: Received Tuya: CMD=0x07 VERSION=3 DATA=[06.02.00.04.00.00.09.34 (8)] INIT_STATE=5
> [15:49:29][D][tuya:272]: Datapoint 6 update to 2356
> [15:49:29][V][tuya:107]: Received Tuya: CMD=0x07 VERSION=3 DATA=[07.02.00.04.00.00.00.00 (8)] INIT_STATE=5
> [15:49:29][D][tuya:272]: Datapoint 7 update to 0
>

sumur1au, 2022-02-16:
@brywithawhy . James forked the project. He has a youtube video on it h[ttps://www.egglec.com.au/post/skyfan-dc-guide-for-home-assistant ](https://www.egglec.com.au/post/skyfan-dc-guide-for-home-assistant)

n8man, 2023-09-20:
I'm having this issue with a Prichor Smart Ceiling Fan controller (Amazon B0C4X7S3RR).

```
[19:45:00][C][tuya:041]: Tuya:
[19:45:00][C][tuya:056]:   Datapoint 20: switch (value: OFF)
[19:45:00][C][tuya:058]:   Datapoint 22: int value (value: 70)
[19:45:00][C][tuya:056]:   Datapoint 60: switch (value: OFF)
[19:45:00][C][tuya:058]:   Datapoint 62: int value (value: 0)
[19:45:00][C][tuya:062]:   Datapoint 63: enum (value: 0)
```

dpID 62 is the fan speed

When I try to turn on the fan and set the speed:
```
[19:48:04][D][fan:021]: 'Fan' - Setting:
[19:48:04][D][fan:024]:   State: ON
[19:48:04][D][fan:030]:   Speed: 2
[19:48:04][D][tuya:597]: Setting datapoint 60 to 1
[19:48:04][D][tuya:597]: Setting datapoint 62 to 1
[19:48:04][E][tuya:602]: Attempt to set datapoint 62 with incorrect type
```

SirMrDexter, 2023-10-31:
I managed to make it work using Tuya Number component instead of using the fan data points.
https://esphome.io/components/number/tuya

@jeggleston1981 this seems to give better control both for fan speed and light dimmer on SkyFan DC.
Here's a sample of what I got so far,
https://github.com/SirMrDexter/easy_skyfanDC/blob/d24ac43b7eac8f1df499bb0771ec6f073ea4e5c0/skyfandc-esp8266.yaml#L62

closed 2023-12-17 by jesserockz