esphome/issues #2903: http_request.get: get_string() always returns the same (first) value

roitagar, 2022-01-03:
### The problem

`http_request.cpp` / `HttpRequestComponent::get_string()` - the local variable used for converting the `String` to `char* is defined `static const` - its value is set exactly once on first usage, and never changed again.
This was introduced in PR esphome/esphome#1041 (handles #1175)

As a result, get_string() always returns the same (first) value.

Since `HttpClient::getString()` constructs a new `String` and fills it every time, I think it makes sense to return that directly instead of `char*`, but I'd hate to break the API 😞 

Alternative solution can be to cache the string as a class member when requested, and invalidate it every time a new request is sent.

### Which version of ESPHome has the issue?

2021.12.3

### What type of installation are you using?

pip

### What platform are you using?

ESP32

### Board

esp32dev (Lilygo TTGO T5 v2.3.1)

### Component causing the issue

http_request

### Example YAML snippet

```yaml
http_request:
  id: my_req

interval:
  - interval: 5s
    then:
      - http_request.get:
          url: http://worldtimeapi.org/api/timezone/Asia/Jerusalem
          on_response:
            then:
              - logger.log:
                  format: "Got %s"
                  args: ['id(my_req).get_string()']
```

### Anything in the logs that might be useful for us?

```txt
[11:11:29][D][main:117]: Got {"abbreviation":"IST","client_ip":"109.186.23.167","datetime":"2022-01-03T10:50:50.565038+02:00","day_of_week":1,"day_of_year":3,"dst":false,"dst_from":null,"dst_offset":0,"dst_until":null,"raw_offset":7200,"timezone":"Asia/Jerusalem","unixtime":1641199850,"utc_datetime":"2022-01-03T08:50:50.565038+00:00","utc_offset":"+02:00","week_number":1}
[11:11:29][D][http_request:089]: HTTP Request completed; URL: http://worldtimeapi.org/api/timezone/Asia/Jerusalem; Code: 200
[11:11:34][D][main:117]: Got {"abbreviation":"IST","client_ip":"109.186.23.167","datetime":"2022-01-03T10:50:50.565038+02:00","day_of_week":1,"day_of_year":3,"dst":false,"dst_from":null,"dst_offset":0,"dst_until":null,"raw_offset":7200,"timezone":"Asia/Jerusalem","unixtime":1641199850,"utc_datetime":"2022-01-03T08:50:50.565038+00:00","utc_offset":"+02:00","week_number":1}
[11:11:34][D][http_request:089]: HTTP Request completed; URL: http://worldtimeapi.org/api/timezone/Asia/Jerusalem; Code: 200
```

martgras, 2022-01-03:
I've submitted a PR to fix this issue. 
If you want to test add this to your yaml 
```
external_components:
  - source: github://pr#2987
    components: ["http_request"]
```

closed 2022-01-03 by oxan

roitagar, 2022-01-03:
Wow, that was fast, and elegant.
The fix works, thank you.

To test it I also needed to include the `"json"` component, some internal compatibility I guess.