esphome/issues #2904: Speed fan incorrectly sets their state after power on

a005, 2022-01-03:
### The problem

In some cases action "fan.toggle"  incorrect sets the fan speed.

- Just after flashing firmware or reset the device the action "fan.toggle" sets fan state to "on" and fan speed to  >0 (correct). 
- After power on/off cycle the action "fan.toggle" sets fan state to "on" and fan speed to  0 (incorrect).

### Which version of ESPHome has the issue?

2021.12.1

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

2021.12.3

### What platform are you using?

ESP8266

### Board

nodemcuv2

### Component causing the issue

Speed Fan

### Example YAML snippet

```yaml
esphome:
  name: test_fan
  platform: ESP8266
  board: nodemcuv2

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
    
### Enable logging ###
logger:
### Enable Home Assistant API ###
api:

fan:
  - platform: speed
    id: speed_fan
    output: fan_output_value
    name: "2 Speed Fan"
    speed_count: 2
    on_turn_on:
    - logger.log: "Fan Turned On"
    on_speed_set:
    - logger.log: "Fan Speed was changed"
    on_turn_off:
    - logger.log: "Fan Turned Off"

binary_sensor:
  - platform: gpio
    name: "Power toggle button"
    internal: true
    pin:
      number: GPIO10
      mode: INPUT_PULLUP
      inverted: True
    on_press:
      - fan.toggle: speed_fan

#relay outputs for 2-speed fan motor 
switch:
  - platform: gpio
    id: relay1
    pin: D6
    restore_mode: ALWAYS_OFF
    #only one should be on
    interlock: &interlock_group [relay1, relay2]
  - platform: gpio
    id: relay2
    pin: D8
    restore_mode: ALWAYS_OFF
    interlock: *interlock_group
    
output:
# 2-speed fan output_value    
  - platform: template
    id: fan_output_value
    type: float
    write_action:
      - if:
          condition:
            lambda: return ((state == 0));
          then:
            - switch.turn_off: relay1
            - switch.turn_off: relay2
            - logger.log: "fan_output_value: 0"
      - if:
          condition:
            lambda: return ((state > 0) && (state <= .5));
          then:
            - switch.turn_on: relay1
            - switch.turn_off: relay2
            - logger.log: "fan_output_value: <=.5"
      - if:
          condition:
            lambda: return ((state > .5));
          then:
            - switch.turn_off: relay1
            - switch.turn_on: relay2
            - logger.log: "fan_output_value: >.50"
```

### Anything in the logs that might be useful for us?

```txt
Toggle button pressed after power on :

[D][binary_sensor:036]: 'Power toggle button': Sending state ON
[D][main:187]: Fan Turned On
[D][main:199]: Fan Speed was changed
[D][speed.fan:035]: Setting speed: 0.00
[D][switch:017]: 'relay1' Turning OFF.
[D][switch:017]: 'relay2' Turning OFF.
[D][main:151]: fan_output_value: 0

Toggle button pressed after reset device :

[D][binary_sensor:036]: 'Power toggle button': Sending state ON
[D][main:187]: Fan Turned On
[D][main:199]: Fan Speed was changed
[D][speed.fan:035]: Setting speed: 0.50
[D][switch:013]: 'relay1' Turning ON.
[D][switch:037]: 'relay1': Sending state ON
[D][switch:017]: 'relay2' Turning OFF.
[D][main:158]: fan_output_value: <=.5
```

### Additional information

[image]

In HA the fan state is also incorrect

nagyrobi, 2022-01-03:
Observed similar behavior.
I think it's because of how restore_state works.

OttoWinter, 2022-01-18:
`fan.toggle` is handled inside Home Assistant (we only receive `fan.turn_on`/`fan.turn_off` requests). So the issue would be there then.

kr0ner, 2024-07-21:
might be related to this https://github.com/esphome/issues/issues/6062