esphome/issues #2906: [Modbus] Garbage byte received before request is sent affects handling of the response

anatoly-savchenkov, 2022-01-03:
### The problem

While debugging issue #2893 I found one interesting thing (this is another Modbus device in my system). In the log below there are 3 transactions. The first one is good. The second and the third show identical request/response packets but CRC check failed for the second and succeeded for the third transaction, see last 2 transactions in the log below.

I believe the reason is the single erroneous byte received even before the second request was sent:
````
[20:23:18][V][modbus:058]: Modbus received Byte  255 (0Xff)
````
No idea why this byte appeared but if my hypothesis is true, handling of a request / response is not cleanly initialized somewhere and previous erroneous transaction(s) may affect the subsequent transaction probably causing more CRC errors than it should be.

I didn't look at the code yet. Probably @martgras will have immediate ideas about this. If no, I'll continue debugging.

### Which version of ESPHome has the issue?

2021.12.3

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2021.12.7

### What platform are you using?

ESP8266

### Board

Wemos D1 mini

### Component causing the issue

Modbus Controller

### Anything in the logs that might be useful for us?

```txt
[20:23:18][V][modbus_controller:027]: Sending next modbus command to device 1 register 0x03 count 1
[20:23:18][V][modbus:184]: Modbus write: 01.03.00.03.00.01.74.0A (8)
[20:23:18][V][modbus_controller:447]: Command sent 3 0x3 1
[20:23:18][V][modbus:058]: Modbus received Byte  1 (0X1)
[20:23:18][V][modbus:058]: Modbus received Byte  3 (0X3)
[20:23:18][V][modbus:058]: Modbus received Byte  2 (0X2)
[20:23:18][V][modbus:058]: Modbus received Byte  0 (0X0)
[20:23:18][V][modbus:058]: Modbus received Byte  1 (0X1)
[20:23:18][V][modbus:058]: Modbus received Byte  121 (0X79)
[20:23:18][V][modbus:058]: Modbus received Byte  132 (0X84)
[20:23:18][V][modbus_controller:048]: Modbus response queued
[20:23:18][V][modbus_controller:055]: Process modbus response for address 0x3 size: 2
[20:23:18][V][modbus_controller:096]: data for register address : 0x3 : 
[20:23:18][V][modbus_controller.switch:040]: Publish 'Ventilation': new value = ON type = 3 address = 3 offset = 0
[20:23:18][V][modbus:058]: Modbus received Byte  255 (0Xff)
[20:23:18][V][modbus_controller:027]: Sending next modbus command to device 1 register 0x4666 count 1
[20:23:18][V][modbus:184]: Modbus write: 01.03.46.66.00.01.71.5D (8)
[20:23:18][V][modbus_controller:447]: Command sent 3 0x4666 1
[20:23:19][V][modbus:058]: Modbus received Byte  1 (0X1)
[20:23:19][V][modbus:058]: Modbus received Byte  3 (0X3)
[20:23:19][V][modbus:058]: Modbus received Byte  2 (0X2)
[20:23:19][V][modbus:058]: Modbus received Byte  15 (0Xf)
[20:23:19][V][modbus:058]: Modbus received Byte  160 (0Xa0)
[20:23:19][V][modbus:058]: Modbus received Byte  189 (0Xbd)
[20:23:19][V][modbus:058]: Modbus received Byte  204 (0Xcc)
[20:23:19][W][modbus:096]: Modbus CRC Check failed! D88D!=CCBD
[20:23:19][V][modbus_controller:027]: Sending next modbus command to device 1 register 0x4666 count 1
[20:23:19][V][modbus:184]: Modbus write: 01.03.46.66.00.01.71.5D (8)
[20:23:19][V][modbus_controller:447]: Command sent 3 0x4666 1
[20:23:19][V][modbus:058]: Modbus received Byte  1 (0X1)
[20:23:19][V][modbus:058]: Modbus received Byte  3 (0X3)
[20:23:19][V][modbus:058]: Modbus received Byte  2 (0X2)
[20:23:19][V][modbus:058]: Modbus received Byte  15 (0Xf)
[20:23:19][V][modbus:058]: Modbus received Byte  160 (0Xa0)
[20:23:19][V][modbus:058]: Modbus received Byte  189 (0Xbd)
[20:23:19][V][modbus:058]: Modbus received Byte  204 (0Xcc)
[20:23:19][V][modbus_controller:048]: Modbus response queued
[20:23:19][V][modbus_controller:055]: Process modbus response for address 0x4666 size: 2
```

martgras, 2022-01-04:
changed modbus component to ignore bytes when not expecting a response
`Modbus received unexpected Byte %d (0X%x) discarding it` is logged when it happens. 
But I'm not sure if just ignoring the byte is a good idea for other components using modbus.   In theory it should be ok because modbus devices don't send data unless a command is sent to them this change is at "heart" of modbus so I want to be sure not breaking other code. 

Anyways for testing use:

```
external_components:
    - source:
        type: git
        url: https://github.com:/martgras/esphome
        ref: modbus-sendfix
     components: [modbus_controller, modbus]
```

martgras, 2022-01-05:
I have refactored modbus esp. if more than one modbus device is used.
Can you please try using these versions 

```yaml 
external_components:
    - source:
        type: git
        url: https://github.com:/martgras/esphome
        ref: modbus-queue
     components: [ modbus]
    - source:
        type: git
        url: https://github.com:/martgras/esphome
        ref: modbus-sendfix
     components: [modbus_controller]
```

nagyrobi, 2022-01-06:
> No idea why this byte appeared

Do you use termination resistors on the line?
See https://know.innon.com/bias-termination-rs485-network
Missing such terminators can easily cause echoed bytes.

exciton, 2025-01-10:
@anatoly-savchenkov - you could try this PR to see if it helps: https://github.com/esphome/esphome/pull/8032