esphome/issues #2908: "An error occurred saving" message in Safari due to bad ingress proxying for 304 responses

masto, 2022-01-04:
### The problem

I apologize for this not being the best bug report -- I tracked down the problem and found a "fix", but I'm afraid I'm out of time for a while, so I'll try to brain dump and hope someone else can pick up the pieces to get an acceptable fix merged.

I've noticed for quite some time that when I click "Save" in the ESPHome YAML editor, I get a popup saying "An error occurred saving _filename.yaml_". I just click it again (and sometimes again and again) until it saves correctly. I'm guilty of assuming everyone must know about this. I recently demonstrated this in a video (https://youtu.be/Kdf6W_Ied4o?t=1202) and a bunch of people told me they've never seen the error, so I got nerd-sniped into investigating it, which took some digging with Wireshark.

I can only reproduce this in Safari on macOS (11.6 but I doubt it matters). The error does not appear in Chrome **but HTTP issues are still happening**, it just doesn't manifest with the save error message.

In Safari, all the console says is `Failed to load resource: The network connection was lost.` After staring at a lot of packet traces, I realized what's going on here. If I open the ESPHome Dashboard directly (not through Home Assistant but e.g. running it on port 6052), I can see a series of HTTP requests all made over a single keep-alive connection:

```
GET /ping HTTP/1.1
Host: home-assistant-demo.local:6052
Cookie: ...
Connection: keep-alive
If-None-Match: "3da846ee084e5988542f1fcdc62cd589e23fcdee"
Accept: */*
User-Agent: Mozilla/5.0 (Macintosh; Intel Mac OS X 10_15_7) AppleWebKit/605.1.15 (KHTML, like Gecko) Version/14.1.2 Safari/605.1.15
Accept-Language: en-us
Referer: http://home-assistant-demo.local:6052/
Accept-Encoding: gzip, deflate

HTTP/1.1 304 Not Modified
Server: nginx
Date: Tue, 04 Jan 2022 02:37:36 GMT
Connection: keep-alive
Etag: "3da846ee084e5988542f1fcdc62cd589e23fcdee"
X-Content-Type-Options: nosniff
X-XSS-Protection: 1; mode=block
X-Robots-Tag: none

GET /ping HTTP/1.1
Host: home-assistant-demo.local:6052
Cookie: ...
Connection: keep-alive
If-None-Match: "3da846ee084e5988542f1fcdc62cd589e23fcdee"
Accept: */*
User-Agent: Mozilla/5.0 (Macintosh; Intel Mac OS X 10_15_7) AppleWebKit/605.1.15 (KHTML, like Gecko) Version/14.1.2 Safari/605.1.15
Accept-Language: en-us
Referer: http://home-assistant-demo.local:6052/
Accept-Encoding: gzip, deflate

HTTP/1.1 304 Not Modified
Server: nginx
Date: Tue, 04 Jan 2022 02:37:38 GMT
Connection: keep-alive
Etag: "3da846ee084e5988542f1fcdc62cd589e23fcdee"
X-Content-Type-Options: nosniff
X-XSS-Protection: 1; mode=block
X-Robots-Tag: none
```

But when I access the same thing through Home Assistant ingress, I see the following:

```
GET /api/hassio_ingress/1OOdgh0-iY4ZuJP76CzMohijGzUTv_-XpCtbY5EcO5E/ping HTTP/1.1
Host: home-assistant-demo.local:8123
Cookie: ...
Connection: keep-alive
If-None-Match: "3da846ee084e5988542f1fcdc62cd589e23fcdee"
Accept: */*
User-Agent: Mozilla/5.0 (Macintosh; Intel Mac OS X 10_15_7) AppleWebKit/605.1.15 (KHTML, like Gecko) Version/14.1.2 Safari/605.1.15
Accept-Language: en-us
Referer: http://home-assistant-demo.local:8123/api/hassio_ingress/1OOdgh0-iY4ZuJP76CzMohijGzUTv_-XpCtbY5EcO5E/
Accept-Encoding: gzip, deflate

HTTP/1.1 304 Not Modified
Server: nginx
Date: Tue, 04 Jan 2022 02:42:32 GMT
Connection: keep-alive
Etag: "3da846ee084e5988542f1fcdc62cd589e23fcdee"
X-Content-Type-Options: nosniff
X-XSS-Protection: 1; mode=block
X-Robots-Tag: none
Content-Type: application/octet-stream
Transfer-Encoding: chunked

0

GET /api/hassio_ingress/1OOdgh0-iY4ZuJP76CzMohijGzUTv_-XpCtbY5EcO5E/ping HTTP/1.1
Host: home-assistant-demo.local:8123
Cookie: ...
Connection: keep-alive
If-None-Match: "3da846ee084e5988542f1fcdc62cd589e23fcdee"
Accept: */*
User-Agent: Mozilla/5.0 (Macintosh; Intel Mac OS X 10_15_7) AppleWebKit/605.1.15 (KHTML, like Gecko) Version/14.1.2 Safari/605.1.15
Accept-Language: en-us
Referer: http://home-assistant-demo.local:8123/api/hassio_ingress/1OOdgh0-iY4ZuJP76CzMohijGzUTv_-XpCtbY5EcO5E/
Accept-Encoding: gzip, deflate
```

Notice that the `304 Not Modified` response is being rewritten to use Transfer-Encoding: chunked. 304 responses have no content, so the chunk size is empty. I don't know enough to quote chapter and verse of the HTTP spec, but I can tell you for sure that this causes the strange behavior that follows, which is that the browser sends the next request header and then _immediately closes the connection_. You can see it in a network trace:

[image]

Safari sends a GET and then a FIN packet, cutting off the connection. This seems to result in a race condition that ultimately results in the failed to save errors. What's interesting is that _Chrome behaves similarly_ when it gets this chunked 304, immediately resetting the connection:

[image]

It doesn't get confused about the state like Safari does, but it's clearly triggering unexpected behavior.

All that said, I found the bit of code in Home Assistant's ingress.py that decides whether the request is "simple" or needs chunked handling, and added a one-line workaround:

https://github.com/masto/home-assistant-core/commit/421dd8f2b414c1fb5d15a3ba2ebdea652dbff26e

This stops the connection from being closed, and both Safari and Chrome will happily sit there all day:

[image]

Furthermore, I no longer get the failure to save in Safari.

I'm assuming this commit, as-is, will be not accepted into Home Assistant. It would be too easy to nit-pick whether there are other types of requests that should be handled the same way, and whether matching on status == 304 is correct, and there ought to be unit tests, etc. I spent the last day of my winter vacation on this, and when work starts tomorrow I won't have time to follow up.

So I'm leaving all the information here. This is exactly what's happening, and a way to stop it happening, but it might need someone who has time to take it to a landing. Hope it's still helpful!

### Which version of ESPHome has the issue?

2021.12.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2021.12.7

### What platform are you using?

ESP32

balloob, 2022-01-04:
Good catch! We were literally discussing this after we saw your video as no one had experienced that before either. 

btw for your next video, instead of suggesting ESP Flasher, use https://web.esphome.io/ 👍

masto, 2022-01-04:
> https://web.esphome.io/

Yes, that is glorious, thanks for it! I just found out about it this morning.

balloob, 2022-01-04:
btw, I don't know what kind of videos you usually make, but we would love to see a video about web.esphome.io that we can embed in the next ESPHome release notes. The goal of the web version is that many people don't have easy access to their ESPHome dashboards over HTTPS, so this allows them to install things.

closed 2022-01-04 by balloob