esphome/issues #2911: fatal error: SPI.h: No such file or directory

GrzesiuS, 2022-01-04:
### The problem

After ESPHome update (can't remember previous version) I can't compile my custom component based on nRF24L01.

`INFO Reading configuration /config/testowy.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing odbiornik (board: nodemcuv2; framework: arduino; platform: platformio/espressif8266 @ 2.6.3)
--------------------------------------------------------------------------------
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
Dependency Graph
|-- <ESPAsyncTCP-esphome> 1.2.3
|-- <RF24> 1.4.2
|-- <ESP8266TrueRandom> 1.0.0
|-- <RokkitHash> 0.0.0-alpha+sha.20151212
|-- <ESPAsyncWebServer-esphome> 2.1.0
|   |-- <ESPAsyncTCP-esphome> 1.2.3
|   |-- <Hash> 1.0
|   |-- <ESP8266WiFi> 1.0
|-- <DNSServer> 1.1.1
|-- <ESP8266WiFi> 1.0
|-- <ArduinoJson-esphomelib> 5.13.3
|-- <ESP8266mDNS> 1.2
Compiling .pioenvs/odbiornik/src/main.cpp.o
Compiling .pioenvs/odbiornik/lib400/RF24/RF24.cpp.o
Archiving .pioenvs/odbiornik/lib190/libRokkitHash.a
In file included from src/main.cpp:26:0:
src/nrf24l01_receiver.h:3:17: fatal error: spi.h: No such file or directory

*************************************************************
* Looking for spi.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:spi.h"
* Web  > https://platformio.org/lib/search?query=header:spi.h
*
*************************************************************

 #include "spi.h"
                 ^
In file included from .piolibdeps/odbiornik/RF24/RF24.cpp:10:0:
.piolibdeps/odbiornik/RF24/RF24_config.h:106:29: fatal error: SPI.h: No such file or directory

*************************************************************
* Looking for SPI.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:SPI.h"
* Web  > https://platformio.org/lib/search?query=header:SPI.h
*
*************************************************************

             #include <SPI.h>
                             ^
compilation terminated.
compilation terminated.
Compiling .pioenvs/odbiornik/lib181/Hash/Hash.cpp.o
*** [.pioenvs/odbiornik/lib400/RF24/RF24.cpp.o] Error 1
Compiling .pioenvs/odbiornik/libfb1/ESP8266WiFi/BearSSLHelpers.cpp.o
*** [.pioenvs/odbiornik/src/main.cpp.o] Error 1
========================== [FAILED] Took 2.54 seconds ==========================`

I have tried using 'SPI.h' (this worked) or 'spi.h' and still can't find the header file.

Is there any other log I can provide to help resolve this issue?

### Which version of ESPHome has the issue?

the latest docker image

### What type of installation are you using?

Docker

### What platform are you using?

ESP8266

### Board

nodemcu

### Component causing the issue

custom

### Example YAML snippet

```yaml
esphome:
  name: odbiornik
  platform: ESP8266
  board: nodemcuv2

  
  includes:
    - nrf24l01_receiver.h
    - nrf24l01_helpers.h

  libraries:
    - "nrf24/RF24@^1.3.12"
    - "marvinroger/ESP8266TrueRandom"
    - "sukkopera/RokkitHash"

web_server:
  port: 80
```

martgras, 2022-01-04:
please try adding SPI to the libraries section
```
   libraries:
    - SPI
```

GrzesiuS, 2022-01-04:
doesn't work.

I tried to find SPI on planformio but I didn't find it.

jenscski, 2022-01-05:
Have you tried adding spi to your configuration?
https://esphome.io/components/spi.html

oxan, 2022-01-05:
> doesn't work.

Add it before the other libraries you include.

GrzesiuS, 2022-01-09:
Adding SPI configuration doesn't work.

Changing yaml order and inserting libraries before include section doesn't work either

jenscski, 2022-01-10:
I'm trying to replicate your error, and was able to get spi.h not found using this simplified version

```
esphome:
  name: odbiornik
  platform: ESP8266
  board: nodemcuv2

  libraries:
    - "nrf24/RF24@^1.3.12"
    - "marvinroger/ESP8266TrueRandom"
    - "sukkopera/RokkitHash"
```

But if I changed the yaml file to this, then It compiled

```
esphome:
  name: odbiornik
  platform: ESP8266
  board: nodemcuv2

  libraries:
    - SPI
    - "nrf24/RF24@^1.3.12"
    - "marvinroger/ESP8266TrueRandom"
    - "sukkopera/RokkitHash"
```

Did you try to include SPI i library list?

Which version of ESPHome are you using?
`docker run --rm esphome/esphome version`

GrzesiuS, 2022-01-10:
it works!

Putting SPI at the top of the libraries list fixed the issue.

Thank you!

closed 2022-01-10 by GrzesiuS