esphome/issues #2912: Sonoff Basic not connecting to WiFi anymore

ajfriesen, 2022-01-04:
### The problem

I have updated my sonoff basic today.

After the update the sonoff was not available anymore in home assistant.

I bit the bullet and crammed it out the corner in order to reflash the device.

Flashing with my FTDI USB adapter works fine.
After flashing I disconnect and reconnect the FTDI USB cable and check the logs.
Everything works fine. Connecting to wifi and the switch works.

Putting the sonoff basic back into the case and main power however does not work anymore.
I can not ping the device and can not see it connecting to wifi on my unfi AP.

Anything I did wrong or missed in the changelog which would indicate this error?

### Which version of ESPHome has the issue?

2021.12.3

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

core-2021.12.7

### What platform are you using?

ESP8266

### Board

sonoff basic

### Example YAML snippet

```yaml
---
# Device: sonoff basic
# Light for globus lamp
#

substitutions:
  devicename: sonoff-basic-03
  upper_devicename: Globus

esphome:
  name: ${devicename}
  platform: ESP8266
  board: esp01_1m

wifi:
  networks:
  - ssid: !secret wifi_ssid
    password: !secret wifi_password
  reboot_timeout: 15min
  
  ap:
    ssid: ${devicename}
    password: !secret esphome_ap_password

# Enable logging
logger:

# Enable Home Assistant API
api:
  reboot_timeout: 15min

ota:

sensor:
  - platform: wifi_signal
    name: ${devicename}_wifi_signal
    update_interval: 10s
  - platform: uptime
    name: ${devicename}_uptime

text_sensor:
  - platform: version
    name: ${devicename}_esphome_version

# GPIO13 Green LED
status_led:
  pin:
    number: GPIO13
    inverted: yes

# GPIO0 Button inverted
binary_sensor:
  - platform: gpio
    id: button
    pin:
      number: GPIO0
      mode: INPUT_PULLUP
      inverted: True
    name: ${devicename}_button
    on_press:
      - light.toggle: lamp

# GPIO12 Relay
output:
  - platform: gpio
    id: relay
    pin: GPIO12  

light:
  - platform: binary
    id: lamp
    name: ${upper_devicename}
    output: relay
```

### Additional information

I have a different sonoff device which updates just fine:

```
---
# Device: sonoff basic
# Light switch for the star
#

substitutions:
  devicename: sonoff_basic_02
  upper_devicename: Stern

esphome:
  name: ${devicename}
  platform: ESP8266
  board: esp01_1m

wifi:
  networks:
  - ssid: !secret wifi_ssid
    password: !secret wifi_password
  reboot_timeout: 15min

# Enable logging
logger:

# Enable Home Assistant API
api:
  reboot_timeout: 15min

ota:

sensor:
  - platform: wifi_signal
    name: ${devicename}_wifi_signal
    update_interval: 10s
  - platform: uptime
    name: ${devicename}_uptime

text_sensor:
  - platform: version
    name: ${devicename}_esphome_version

# GPIO13 Green LED
status_led:
  pin:
    number: GPIO13
    inverted: yes

# GPIO0 Button inverted
binary_sensor:
  - platform: gpio
    id: button
    pin:
      number: GPIO0
      mode: INPUT_PULLUP
      inverted: True
    name: ${devicename}_button"
    on_press:
      - light.toggle: lamp

# GPIO12 Relay
output:
  - platform: gpio
    id: relay
    pin: GPIO12  

light:
  - platform: binary
    id: lamp
    name: ${upper_devicename}
    output: relay
```

randybb, 2022-01-04:
I would suspect it is a HW issue with PSU on your device as you have mentioned in #2826 that another device works just fine. Anyway, there is no difference between sonoff basic and any other esp8266 devices that are just working fine, so it must be related to your HW.

gubagu, 2022-01-05:
Similar problem on my side. All of my ESP devices went offline. All devices are reflashed Sonoffs. I did not have the problem with the previous version. 
```ESPHome API for sonoff*******.local: Error resolving IP address: [Errno -2] Name or service not known```
Now the whole system is offline. The previous version did not have any problems with the name resolutions.
Any ideas how could get them online again?

ajfriesen, 2022-01-05:
Updated 2 other s20. Everything is fine.

Another 2 sonoff basics, which one had a fallback ap.
Both not working anymore.
-  physical switch works
- wifi connection does not
- fallback up does not work as well

ajfriesen, 2022-01-06:
@randybb 
I think for 2 of my devices you might be right with hardware issues. They do not do anything when plugged in to main.

However, I have here 2 sonoffs basics which just blink every 2 seconds and are not reachable via wifi.
I even did remove the underscores in their names because of the warning of it.

I can not see them getting a IP. That is weird since the config for other sonoffs is the same.

randybb, 2022-01-06:
Many sonoff devices have problems with faulty PSU over time (dried capacitors?). It can be dead completely or it is just providing not enough current that is needed for wifi - so it can boot and it is always crashing while enabling wifi. This is the reason why logs are so important. Ideally some investigations are needed like to power it from usb-uart completely and if it works, then disconnect 3v3 (keep gnd, rx and tx) and connect it to power.

ajfriesen, 2022-01-07:
Here are the logs from a device where I have powered it by main and connected tx, rx and gnd as suggested:

```
[18:59:52][D][sensor:113]: 'sonoff-basic-04_uptime': Sending state 282.41599 s with 0 decimals of accuracy
[18:59:57][D][wifi:370]: Starting scan...
[19:00:03][D][wifi:385]: Found networks:
[19:00:03][D][wifi:387]:   No network found!
[19:00:08][D][wifi:370]: Starting scan...
[19:00:13][D][wifi:385]: Found networks:
[19:00:13][D][wifi:434]: - [redacted] [redacted]▂▄▆█
[19:00:13][W][wifi:439]: No matching network found!
[19:00:18][D][wifi:370]: Starting scan...
[19:00:24][D][wifi:385]: Found networks:
[19:00:24][D][wifi:387]:   No network found!
[19:00:29][D][wifi:370]: Starting scan...
[19:00:35][D][wifi:385]: Found networks:
[19:00:35][D][wifi:434]: - [redacted] [redacted]▂▄▆█
[19:00:35][W][wifi:439]: No matching network found!
[19:00:40][D][wifi:370]: Starting scan...
[19:00:46][D][wifi:385]: Found networks:
[19:00:46][D][wifi:434]: - [redacted] [redacted]▂▄▆█
[19:00:46][W][wifi:439]: No matching network found!
[19:00:51][D][wifi:370]: Starting scan...
[19:00:52][D][sensor:113]: 'sonoff-basic-04_uptime': Sending state 342.41599 s with 0 decimals of accuracy
[19:00:57][D][wifi:385]: Found networks:
[19:00:57][I][wifi:429]: - 'Jochen und Jutta' [redacted]▂▄▆█
[19:00:57][D][wifi:430]:     Channel: 2
[19:00:57][D][wifi:431]:     RSSI: -69 dB
[19:00:57][D][wifi:434]: - [redacted] [redacted]▂▄▆█
[19:00:57][I][wifi:245]: WiFi Connecting to 'Jochen und Jutta'...
[19:01:04][W][wifi_esp8266:483]: Event: Disconnected ssid='Jochen und Jutta' bssid=[redacted] reason='Auth Expired'
[19:01:04][W][wifi:533]: Error while connecting to network.
[19:01:04][W][wifi:568]: Restarting WiFi adapter...
[19:01:04][I][wifi:245]: WiFi Connecting to 'Jochen und Jutta'...
[19:01:08][W][wifi_esp8266:483]: Event: Disconnected ssid='Jochen und Jutta' bssid=[redacted] reason='Auth Expired'
[19:01:08][W][wifi:533]: Error while connecting to network.
[19:01:08][W][wifi:568]: Restarting WiFi adapter...
[19:01:13][D][wifi:370]: Starting scan...
[19:01:19][D][wifi:385]: Found networks:
[19:01:19][I][wifi:429]: - 'Jochen und Jutta' [redacted]▂▄▆█
[19:01:19][D][wifi:430]:     Channel: 2
[19:01:19][D][wifi:431]:     RSSI: -63 dB
[19:01:19][D][wifi:434]: - [redacted] [redacted]▂▄▆█
[19:01:19][I][wifi:245]: WiFi Connecting to 'Jochen und Jutta'...
[19:01:25][W][wifi_esp8266:483]: Event: Disconnected ssid='Jochen und Jutta' bssid=[redacted] reason='Auth Expired'
[19:01:25][W][wifi:533]: Error while connecting to network.
[19:01:25][W][wifi:568]: Restarting WiFi adapter...
[19:01:25][I][wifi:245]: WiFi Connecting to 'Jochen und Jutta'...
[19:01:27][W][wifi_esp8266:483]: Event: Disconnected ssid='Jochen und Jutta' bssid=[redacted] reason='Association Expired'
[19:01:27][W][wifi:533]: Error while connecting to network.
[19:01:27][W][wifi:568]: Restarting WiFi adapter...
```

Sometimes it is not able to find the wifi, sometimes when getting the wifi there is some auth error which is weird.

---

When I connect with the FTDI only (3v3, gnd, rx and tx) we have this:

```
INFO Reading configuration /config/esphome/sonoff_basic_04.yaml...
INFO Starting log output from /dev/ttyUSB0 with baud rate 115200
[19:05:19][I][wifi:502]: WiFi Connected!
[19:05:19][C][wifi:350]:   Local MAC: EC:FA:BC:06:9E:D8
[19:05:19][C][wifi:351]:   SSID: [redacted]
[19:05:19][C][wifi:352]:   IP Address: 192.168.10.116
[19:05:19][C][wifi:354]:   BSSID: [redacted]
[19:05:19][C][wifi:355]:   Hostname: 'sonoff-basic-04'
[19:05:19][C][wifi:357]:   Signal strength: -76 dB ▂▄▆█
[19:05:19][C][wifi:361]:   Channel: 2
[... 38 lines cut ...]
[19:05:19][C][version.text_sensor:021]: Version Text Sensor 'sonoff-basic-04_esphome_version'
[19:05:19][C][version.text_sensor:021]:   Icon: 'mdi:new-box'
[19:05:19][C][ota:082]: Over-The-Air Updates:
[19:05:19][C][ota:083]:   Address: sonoff-basic-04.local:8266
[19:05:19][C][api:134]: API Server:
[19:05:19][C][api:135]:   Address: sonoff-basic-04.local:6053
[19:05:19][C][api:139]:   Using noise encryption: NO
[19:05:19][C][wifi_signal.sensor:009]: WiFi Signal 'sonoff-basic-04_wifi_signal'
[19:05:19][C][wifi_signal.sensor:009]:   Device Class: 'signal_strength'
[19:05:19][C][wifi_signal.sensor:009]:   State Class: 'measurement'
[19:05:19][C][wifi_signal.sensor:009]:   Unit of Measurement: 'dBm'
[19:05:19][C][wifi_signal.sensor:009]:   Accuracy Decimals: 0
[19:05:19][C][mdns:084]: mDNS:
[19:05:19][C][mdns:085]:   Hostname: sonoff-basic-04
[19:05:24][D][sensor:113]: 'sonoff-basic-04_wifi_signal': Sending state -74.00000 dBm with 0 decimals of accuracy
[19:05:27][D][binary_sensor:036]: 'sonoff-basic-04_button': Sending state ON
[19:05:27][D][light:035]: 'Lichterkette Bett' Setting:
[19:05:27][D][light:046]:   State: ON
[19:05:27][D][binary_sensor:036]: 'sonoff-basic-04_button': Sending state OFF
```

Instant connection. No auth error.

NOTE:
WIFI AP is in the same room within 2 metes distance without any obstacles. Should not be the problem.

ajfriesen, 2022-01-07:
When I only use this minimal config WiFi works descent with main.
That is super odd.

```
---
# Device: sonoff basic
# Light chain on the bed
#

substitutions:
  devicename: sonoff-basic-04
  upper_devicename: Lichterkette Bett

esphome:
  name: ${devicename}
  platform: ESP8266
  board: esp01_1m

wifi:
  networks:
  - ssid: !secret wifi_ssid
    password: !secret wifi_password
  reboot_timeout: 15min

# Enable logging
logger:

# Enable Home Assistant API
api:
  reboot_timeout: 15min

ota:
```

Did I do something wrong on my initial config?

randybb, 2022-01-07:
Strange, I expected continuous crashing. The only issue that I see is your week signal strength -77 dBm is pretty bad.

I did a small test on a crusty old nodemcu v1 (it was mine first ESP!), flashed it with the same config, connected to my bench PSU and it is able boot and connect to WiFi without issues from 2.5 V with OCP set to 151 mA.

I only have Sonoff Basic R2, 2x Sonoff Dual R2, 4x Sonoff mini R2, Sonoff TH16, Sonoff 433, Sonoff T1, Sonoff 4Ch R2, a few other ESP8266 devices (and then bunch of ESP32 devices, which is irrelevant) and everything is working fine. The only difference is that I am running latest DEV, but there have been no changes to WiFi stack as far as I know and much better wifi reception -36 - -64 dBm. Even the weakest one with -74 dBm is "irrigation system" sonoff 4Ch sitting in well shaft with a patch antenna on metal manhole cover 10 m from closest AP and is working just fine.

ajfriesen, 2022-01-07:
Any idea how I can help figuring this out?

I would also be okay to sending this unit to someone with more knowledge than me in that topic.
Have at least 2 which behave the same and can not use them anyway.

nagyrobi, 2022-01-12:
Sonoff Basics are known for their weak internal power supplies. It fails immediately as soon as the MCU needs a bit more power due to longer periods of activity.

Tasmota even had a Lite binary version for a while, with 70% of their software modules stripped out just to assure stability only for Sonoff Basics.

I also have 4 of them. They won't migrate to ESPHome.

ajfriesen, 2022-01-12:
I have no problem with that statement.

But isn't it odd that it worked for years just fine? I mean I bought these devices around 2017.
Had tasmota and esphome running on them perfectly fine.

And some of my basics still work with the new firmware.

spike886, 2022-01-14:
I saw the same issue with all my sonoff basic when I updated to the last version. But I was able to make all of them work again. I had to disable the `wifi_signal` and `safe_mode` features.
Have 8 sonoff and I can reproduce the same issue with all of them. The devices won't connect to wifi if I enable any of these features but it connects again as soon I flash it with the same code without any of those features.

spike886, 2022-01-15:
I did some tests to add information here. I found something funny. If I change the logger config to `VERBOSE` or VERY_VERBOSE` it starts working even with `wifi_signal` and `safe_mode` enabled

smoke007, 2022-01-22:
@spike886 and @ajfriesen,
This is a known issue that goes back to Jun 2019.  See the common fixes in https://github.com/esphome/issues/issues/455.  You're right, changing logging level sometimes fixes the corruption during compile.  

Here's more details I wrote up here https://github.com/esphome/issues/issues/2964#issuecomment-1019006987

martinjuhasz, 2022-01-24:
I'm also running into this issue. I'm running 4 sonoffs with esphome. while flashing two of them THE OTHER TWO became unresponsive.

ajfriesen, 2022-01-30:
By removing the status_led I was able to flash the sonoff basics to a working state again.

Super weird!

ajfriesen, 2022-03-18:
Just updated to the latest esphome.

I took a less important so off basic and updated.

Did not come back. I need to get it out again and see what's wrong.

arko36, 2022-07-11:
Only when connecting to the main 220/110 the wifi does not work anymore, I also sourced it with a lab PS on 5v and same issue. However with the FTDI works fine.
I came across this link https://schinckel.net/2019/11/09/sonoff-basic-unable-to-connect-to-wifi-on-ac-power/ and follow it, **shorting RX/TX with a jumper** and plug it to the mains works OK **leaving all yaml code as before**, power 20db, captive portal, GPIO 0 etc no mod in the code just shortcut RX and TX with a cable.
I hope this gives some light on the C code bug fix on the RX/TX on boot ? I have no idea.

ajfriesen, 2022-07-11:
I have switched all my sonoff basics (about 8) to tasmota now.
All of them work just about fine.

Do not want to wait for this any longer (no offense).

arko36, 2022-07-11:
non taken, I got [E][api:084]: No client connected to API. Rebooting... randomly now. I have time to wait will not upgrade the rest of the sonoffs, may be @OttoWinter can follow up my discovery ? Thanks.

ssieb, 2022-07-11:
> non taken, I got [E][api:084]: No client connected to API. Rebooting... randomly now. I have time to wait will not upgrade the rest of the sonoffs, may be @OttoWinter can follow up my discovery ? Thanks.

It's not likely to be random.  There's a reboot timeout for the api which has a default of 15 minutes.

arko36, 2022-07-11:
Yes, you a re right, every time it reboots the lights turned on!, I set it up to 3h. But still the issue of sometimes losing connection and rebooting while it is in front of the router.

aneisch, 2022-11-16:
> By removing the status_led I was able to flash the sonoff basics to a working state again.
> 
> Super weird!

Same experience for me on latest v2022.10.2.

Mikrofarad, 2022-12-21:
Still valid for v2022.12.3 on Home Assistant with an [ESP-WROOM-32 WiFi Bluetooth OLED Display 18650 Battery Charger Board](https://artofcircuits.com/product/esp-wroom-32-wifi-bluetooth-oled-display-18650-battery-charger-board)

spike886, 2023-02-09:
This issue is the reason why I couldn't update any of the devices that I have installed on my house for the last year. 
So I had to look for a workaround to this issue to be able to continue working on the automatization of my devices.
To do it, I added to all my devices the following config:

```
esphome:
  name: $devicename
  platform: ESP8266
  board: esp01_1m
  on_boot:
    - then:
        - script.execute: safe_mode_wifi

script:
  - id: safe_mode_wifi
    then:
      - delay: 10 min
      - if:
          condition:
            not:
              wifi.connected:
          then:
            - logger.log:  
                format: "starting safe mode"
                level: INFO
            - switch.turn_on: safe_mode_1
          else:
            - logger.log: "wifi connected, safe mode not started"
  

switch: 
  -  platform: safe_mode
     id: safe_mode_1
```

This will make my device go in safe mode if it can't connect to WiFi 10 mins after booting.

From my tests, the error connecting to WiFi doesn't happen on safe mode. So this will allow me to update the firmware using OTA again. Then the trick is to try updating the configuration of the device with a different level of the logger, in most of the cases it works with VERY_VERBOSE

```
logger:
  level: VERY_VERBOSE
```

You should be able to do that, testing different logger levels until you find one that allow the device connecting to WIFI.
Using this workaround, I was able to update all my devices and enable all configurations I needed.

It would be very nice is a behavior like this in implemented on the default firmware when it fails connecting to WiFi, because it's known that with this error the access point doesn't work either

SAOPP, 2023-04-24:
> By removing the status_led I was able to flash the sonoff basics to a working state again.
> 
> Super weird!
Thanks buddy, works for me too! I will remove all these `status_led`s from my stuff 'coz it's not needed at all for my case. Yes, and to avoid similar situations with these boards in the future. :+1:

ojaksch, 2023-08-28:
Interesting thread. I fell into the same trap the last days when changing some of my SonOff's from Tasmota (latest Version = 13.1.0) to esphome. All S20s are working fine so far (haven't converted all of them yet), but _some_ Basics (bought an original box of 10) are failing - BUT they are and were unstable with newer Tasmota versions too. Today I've tried esphome again (current version), no WiFi. Changed some of the yaml according to the suggestions here, still no luck. Reflashed Tasmota, still no dice. As a last resort I've flashed them with an old Tasmota 9.4 and guess what: stable and working devices. Reproduceable.

ajfriesen, 2023-08-28:
In the end, I experienced instability with Tasmota as well and got rid of my sonoff basics.
They are now stored in a box with a bit of pain 😉 

These are not worth my precious time and got replaced with shelly devices.

ojaksch, 2023-08-29:
A little bit of success. My box of 10 Basics is having mixed versions of R1 (labeled "Sonoff TH_V1.1 2017-5-5") and R2 (labeled "Sonoff RF R2 V1.0", but aren't having RF). The R1's are the problematic ones which can't connect to WiFi using esphome (current version) nor Tasmota > v12.2.0. Looks like there are no problems with my R2's.

I'll have to take a closer look to my already working Basics to see which PCB version is used.
If useful to someone I'll provide photos of all Bascis I have.

steve-prentice, 2024-03-19:
> Removing the `status_led` does result in a working device. I haven't really looked much into this (was going to switch the platform from arduino to esp-idf, but it seems that's only an ESP32 thing), but I did re-add the LED as a switch, and it works fine like that. Here's my working YAML on ESPHome version 2023.8.3:

@jdrowell - Thanks! Solved my problem, status_led for me too. I would have given up if I hadn't have spotted this!

taxist08, 2024-06-23:
I also encountered the same problem, 2 years later, although the native firmware, as well as the tasmota firmware, work without problems, but I want to transfer all the equipment to esphome. Maybe there is another solution, with a working LED?

steve-prentice, 2024-06-23:
Honestly in the end I gave up with those old Basics and got a new Sonoff Basic R4, under £5. That’s working perfectly on ESPHome!

smoke007, 2024-06-23:
Same. One by one, I've been slowly switching over all my Basics to Mini r2.
Zero issues with those on esp and flashing OTA updates.

netmms, 2024-07-14:
The problem is the AC/DC circuit and I fixed it.

If there is interest, I can post the solution, but it is a hack.

spike886, 2024-07-15:
Yes, please, share it if you can.

mateuszdrab, 2024-08-10:
My basics are running tasmota and have been reasonably reliable for years however lately after a power cut, none would connect. 2 units are just sitting here unable to connect to WiFi when powered by mains, connect fine when powered through the IO pins.

mateuszdrab, 2024-08-10:
> My basics are running tasmota and have been reasonably reliable for years however lately after a power cut, none would connect. 2 units are just sitting here unable to connect to WiFi when powered by mains, connect fine when powered through the IO pins.

It happened suddenly, there was no firmware upgrade during the time.

I do notice a bit of what seems to be black smoke on the cover the PCB sits on; however, don't see any damage on the PCB components itself. Sounds like something is on its way out with those units.

I'll get them replaced.

fragtion, 2024-09-15:
I have several sonoff basic devices which have been giving me problems on Tasmota for years. Wi-fi didn't connect or was very unreliable. Tried all sorts of hardware hacks like adding ceramic capacitor, removing LED etc, but none of it worked

In the end I fixed ALL of them by flashing them with ESPhome instead. I don't know why Tasmota is unreliable on them and ESPhome works, but it just does. So for these devices I will just use ESPhome instead. Anyone else struggling with unreliable wi-fi on the basic, should try ESPhome. Good chance it will solve your problems like it did for me