esphome/issues #2914: underscore no longer allowed in the name...?

grahambrown11, 2022-01-05:
### The problem

I can no longer create new device with underscores (`_`) in the name!
The input box in HA changes `_` to `-` and the help text has `Only use lowercase letters (a-z), numbers (0-9) or dash (-)`
Please allow an underscore too I've already got devices with underscores in the name and would like to continue using that naming convention.

### Which version of ESPHome has the issue?

2021.12.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2021.12.5

### What platform are you using?

ESP8266

martgras, 2022-01-05:
Yes. Actually hostnames with underscores always violated the RFCs. 
See https://github.com/esphome/esphome/pull/2647

grahambrown11, 2022-01-05:
Thanks, I found this too https://esphome.io/guides/faq.html#why-shouldn-t-i-use-underscores-in-my-device-name
I'm cloding and I guess renaming my old devices to use a dash instead of the underscore...

closed 2022-01-05 by grahambrown11