esphome/issues #2915: KEYESTUDIO CO2 Sensor Module CCS811 NTC

matteos1, 2022-01-05:
### The problem

this product KEYESTUDIO CO2 Sensor Module CCS811 NTC https://wiki.keyestudio.com/KS0457_keyestudio_CCS811_Carbon_Dioxide_Air_Quality_Sensor 
have problem with esphome.
the error is
[17:38:58][I][i2c.arduino:048]: Scanning i2c bus for active devices...
[17:38:59][I][i2c.arduino:060]: Found no i2c devices!

### Which version of ESPHome has the issue?

 2021.12.3

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

Home Assistant 2021.12.8

### What platform are you using?

ESP8266

### Board

nodemcu

### Component causing the issue

ic2

### Example YAML snippet

```yaml
esphome:
  name: votair
substitutions:
  hostname: 'VOTAIR:'
esp8266:
  board: nodemcuv2

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:
  password: "626e3ee26f586edd01aca7253ec33cfa"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Votair Fallback Hotspot"
    password: "KmqpoQJOWO6B"

captive_portal:

web_server:
  port: 80

sensor:

  - platform: wifi_signal
    name: $hostname WiFi Signal
    id: rssi_value
    update_interval: 60s
  - platform: uptime
    name: $hostname Uptime
    id: uptime_s
    update_interval: 15s
  - platform: template
    name: $hostname WiFi
    lambda: |-
      return id(rssi_value).state * 2 + 200.0;
    icon: mdi:wifi
    unit_of_measurement: '%'
    update_interval: 60s

  - platform: ccs811
    eco2:
      name: "CCS811 eCO2 Value"
    tvoc:
      name: "CCS811 Total Volatile Organic Compound"
    address: 0x5A
    update_interval: 60s

  
switch:
  - platform: restart
    name: EspHome Node Restart
binary_sensor:
  - platform: status
    name: $hostname Status       
text_sensor:

  - platform: wifi_info
    ssid:
      name: $hostname WiFi SSID   
    ip_address:
      name: $hostname  IP Address
    bssid:
      name: $hostname Connected BSSID
    mac_address:
      name: $hostname Mac Wifi Address
    scan_results:
      name: $hostname Latest Scan Results
      
  - platform: version
    name: $hostname ESPHome Version
  - platform: wifi_info

  - platform: template
    name: $hostname Uptime (formatted)
    lambda: |-
      uint32_t dur = id(uptime_s).state;
      int dys = 0;
      int hrs = 0;
      int mnts = 0;
      if (dur > 86399) {
        dys = trunc(dur / 86400);
        dur = dur - (dys * 86400);
      }
      if (dur > 3599) {
        hrs = trunc(dur / 3600);
        dur = dur - (hrs * 3600);
      }
      if (dur > 59) {
        mnts = trunc(dur / 60);
        dur = dur - (mnts * 60);
      }
      char buffer[17];
      sprintf(buffer, "%ud %02uh %02um %02us", dys, hrs, mnts, dur);
      return {buffer};
    icon: mdi:clock-start
    update_interval: 15s
    
i2c:
  sda: D1
  scl: D2
```

### Anything in the logs that might be useful for us?

```txt
[17:38:58][I][i2c.arduino:048]: Scanning i2c bus for active devices...
[17:38:59][I][i2c.arduino:060]: Found no i2c devices!
```

matteos1, 2022-01-05:
[ccs811:165]: Communication failed! Is the sensor connected?

matteos1, 2022-01-05:
[17:49:40][C][i2c.arduino:032]: I2C Bus:
[17:49:40][C][i2c.arduino:033]:   SDA Pin: GPIO5
[17:49:40][C][i2c.arduino:034]:   SCL Pin: GPIO4
[17:49:40][C][i2c.arduino:035]:   Frequency: 50000 Hz
[17:49:40][C][i2c.arduino:041]:   Recovery: failed, SCL is held low on the bus
[17:49:40][I][i2c.arduino:048]: Scanning i2c bus for active devices...
[17:49:40][I][i2c.arduino:060]: Found no i2c devices!

matteos1, 2022-01-05:
try with wemos d1 and nodemcu but nothing

Habbie, 2022-01-18:
How did you connect it?