esphome/issues #2917: ESP8266HTTPClient.h: No such file or directory

comet424, 2022-01-07:
### The problem

i trying to upload a nextion tft screen through esphome through Home Assitant and when it complies i get 

INFO Reading configuration /config/esphome/moms-car-temp-2.yaml...
INFO Detected timezone 'America/Toronto'
INFO Generating C++ source...
INFO Compiling app...
Processing moms-car-temp-2 (board: d1_mini; framework: arduino; platform: platformio/espressif8266 @ 2.6.3)
--------------------------------------------------------------------------------
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
Dependency Graph
|-- <ESP8266WiFi> 1.0
|-- <ESP8266mDNS> 1.2
Compiling /data/moms-car-temp-2/.pioenvs/moms-car-temp-2/src/esphome/components/nextion/nextion.cpp.o
Compiling /data/moms-car-temp-2/.pioenvs/moms-car-temp-2/src/esphome/components/nextion/nextion_commands.cpp.o
In file included from src/esphome/components/nextion/nextion.cpp:1:0:
src/esphome/components/nextion/nextion.h:15:31: fatal error: ESP8266HTTPClient.h: No such file or directory

***************************************************************************
* Looking for ESP8266HTTPClient.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:ESP8266HTTPClient.h"
* Web  > https://platformio.org/lib/search?query=header:ESP8266HTTPClient.h
*
***************************************************************************

 #include <ESP8266HTTPClient.h>
                               ^
compilation terminated.
*** [/data/moms-car-temp-2/.pioenvs/moms-car-temp-2/src/esphome/components/nextion/nextion.cpp.o] Error 1
In file included from src/esphome/components/nextion/nextion_commands.cpp:1:0:
src/esphome/components/nextion/nextion.h:15:31: fatal error: ESP8266HTTPClient.h: No such file or directory

***************************************************************************
* Looking for ESP8266HTTPClient.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:ESP8266HTTPClient.h"
* Web  > https://platformio.org/lib/search?query=header:ESP8266HTTPClient.h
*
***************************************************************************

 #include <ESP8266HTTPClient.h>
                               ^
compilation terminated.
*** [/data/moms-car-temp-2/.pioenvs/moms-car-temp-2/src/esphome/components/nextion/nextion_commands.cpp.o] Error 1
========================== [FAILED] Took 0.61 seconds ==========================

### Which version of ESPHome has the issue?

2021.12.1 

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2021.12.8

### What platform are you using?

ESP8266

### Board

wemos mini d1

### Component causing the issue

nothing i think esphome is broken

### Example YAML snippet

```yaml
api:
  password: !secret api_password
  services:
    - service: update_nextion
      then:
        - lambda: 'id(nextion1)->upload_tft();'

display:
  - platform: nextion
    id: nextion1
    tft_url: http://192.168.0.30/car.tft
#    uart_id: uart_2
    update_interval: 10s
    
    # ...
    lambda: |-
```

### Anything in the logs that might be useful for us?

```txt
INFO Reading configuration /config/esphome/moms-car-temp-2.yaml...
INFO Detected timezone 'America/Toronto'
INFO Generating C++ source...
INFO Compiling app...
Processing moms-car-temp-2 (board: d1_mini; framework: arduino; platform: platformio/espressif8266 @ 2.6.3)
--------------------------------------------------------------------------------
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
Dependency Graph
|-- <ESP8266WiFi> 1.0
|-- <ESP8266mDNS> 1.2
Compiling /data/moms-car-temp-2/.pioenvs/moms-car-temp-2/src/esphome/components/nextion/nextion.cpp.o
Compiling /data/moms-car-temp-2/.pioenvs/moms-car-temp-2/src/esphome/components/nextion/nextion_commands.cpp.o
In file included from src/esphome/components/nextion/nextion.cpp:1:0:
src/esphome/components/nextion/nextion.h:15:31: fatal error: ESP8266HTTPClient.h: No such file or directory

***************************************************************************
* Looking for ESP8266HTTPClient.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:ESP8266HTTPClient.h"
* Web  > https://platformio.org/lib/search?query=header:ESP8266HTTPClient.h
*
***************************************************************************

 #include <ESP8266HTTPClient.h>
                               ^
compilation terminated.
*** [/data/moms-car-temp-2/.pioenvs/moms-car-temp-2/src/esphome/components/nextion/nextion.cpp.o] Error 1
In file included from src/esphome/components/nextion/nextion_commands.cpp:1:0:
src/esphome/components/nextion/nextion.h:15:31: fatal error: ESP8266HTTPClient.h: No such file or directory

***************************************************************************
* Looking for ESP8266HTTPClient.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:ESP8266HTTPClient.h"
* Web  > https://platformio.org/lib/search?query=header:ESP8266HTTPClient.h
*
***************************************************************************

 #include <ESP8266HTTPClient.h>
                               ^
compilation terminated.
*** [/data/moms-car-temp-2/.pioenvs/moms-car-temp-2/src/esphome/components/nextion/nextion_commands.cpp.o] Error 1
========================== [FAILED] Took 0.63 seconds ==========================
```

### Additional information

i know i downgraded to the 12.1  due to the .2 i think it was had broken things  and i think it was the programer told us how to install the previous version as esphome was broken in home assistant

i actually dont know how  to  see if the newest version is fixed  the bugs how to connect to the latest    i think its .4  when i was on esphome website trying to learn how to upload nexition screens  instead me me having to take about my device to change it..

so i not sure if its been fixed..  or how to get to the latest ....

jsuanet, 2022-01-11:
This is a duplicate of https://github.com/esphome/issues/issues/2721  As a workaround you can add  `http_request:` to the YAML configuration file.

comet424, 2022-01-12:
@jsuanet   didnt help  i added  http_resquest:
it will complie but it wont upload the nextion screen  file to the nextion
ive tried from home assistant  
and a website i host  and it will not upload

jsuanet, 2022-01-16:
@comet424  What error message do you get (ESPHome-log)  if you activate the service via HA?  If I look at your `tft_url` I'm missing the port number as described in: https://esphome.io/components/display/nextion.html?highlight=nextion#uploading-a-tft-file. I followed these instructions and my `tft_url:` looks like: `tft_url: http://xx.xx.xx.xx:8123/local/tft/my_design.tft` and it is working fine (however i switched from using an esp8266 to an esp32, so not able to test it again with an esp8266).

comet424, 2022-01-16:
@jsuanet im not getting an error when it complies and then uploads  if that mean error log..
i tried the 

tft_url: http://192.168.0.12:8123/local/nextion/car.tft

it will complile it will upload the to the mini d1.. but it wont upload the new screen file to the nexition screeen

is there another error log to look at?

comet424, 2022-01-16:
i also posted the code i also tried in the other link u posted the duplicated one you mentioned..

it just wont upload  tft

jsuanet, 2022-01-17:
@comet424 

Based on your `tft_url:` some questions:
- Is 192.168.0.12 the IP-address off your HA instance
- Is the physical location off the tft file:  `/config/www/nextion` on machine with the above IP-address

I see you are using the add-on; so I'm interested in the logs form the ESPHome dashboard. To get this logs:  first push the button `LOGS` in the dashboard by your nextion device, let this run and then perform (in another browser tab) the `tft update service` from the developer tools. When the tft-file is not found at the `tft_url` location I get a logging like this:

```
[09:48:29][D][nextion_upload:169]: Connected
[09:48:29][D][nextion_upload:175]: Requesting URL: http://192.168.178.54:8123/local/tft/design-v2.tft
[09:48:30][W][nextion_upload:186]: HTTP Request failed; URL: http://192.168.178.54:8123/local/tft/design-v2.tft; Error: , retrying (1/5)
[09:48:30][W][nextion_upload:186]: HTTP Request failed; URL: http://192.168.178.54:8123/local/tft/design-v2.tft; Error: , retrying (2/5)
[09:48:30][W][nextion_upload:186]: HTTP Request failed; URL: http://192.168.178.54:8123/local/tft/design-v2.tft; Error: , retrying (3/5)
[09:48:30][W][nextion_upload:186]: HTTP Request failed; URL: http://192.168.178.54:8123/local/tft/design-v2.tft; Error: , retrying (4/5)
[09:48:31][W][nextion_upload:186]: HTTP Request failed; URL: http://192.168.178.54:8123/local/tft/design-v2.tft; Error: , retrying (5/5)
[09:48:31][D][nextion_upload:310]: Restarting Nextion
[09:48:32][D][nextion_upload:313]: Restarting esphome
```
If I start the update with the tft-file at the correct location, the logs look like:
```
[09:52:24][D][nextion_upload:169]: Connected
[09:52:24][D][nextion_upload:175]: Requesting URL: http://192.168.178.54:8123/local/tft/design-v2.tft
[09:52:25][D][nextion_upload:209]: Updating Nextion NX8048P050_011C...
[09:52:25][D][nextion_upload:235]: Waiting for upgrade response
[09:52:25][E][uart:015]: Reading from UART timed out at byte 0!
[09:52:25][E][uart:015]: Reading from UART timed out at byte 0!
[09:52:25][D][nextion_upload:239]: Upgrade response is  3
[09:52:25][D][nextion_upload:242]: Available 0 : 0x00
[09:52:25][D][nextion_upload:242]: Available 1 : 0x00
[09:52:25][D][nextion_upload:242]: Available 2 : 0x05
[09:52:25][D][nextion_upload:246]: preparation for tft update done
[09:52:25][D][nextion_upload:274]: Allocating buffer size 65536, Heap size is 223276
[09:52:25][D][nextion_upload:291]: Updating tft from "http://192.168.178.54:8123/local/tft/design-v2.tft" with a file size of 1024576 using 65536 chunksize, Heap Size 156220
[09:52:25][D][nextion_upload:048]: Requesting range: bytes=0-16383
[09:52:26][E][uart:015]: Reading from UART timed out at byte 0!
[09:52:26][E][uart:015]: Reading from UART timed out at byte 0!
[09:52:26][E][uart:015]: Reading from UART timed out at byte 0!
[09:52:26][E][uart:015]: Reading from UART timed out at byte 0!
[09:52:27][E][uart:015]: Reading from UART timed out at byte 0!
[09:52:27][E][uart:015]: Reading from UART timed out at byte 0!
[09:52:27][E][uart:015]: Reading from UART timed out at byte 0!
[09:52:27][E][uart:015]: Reading from UART timed out at byte 0!
[09:52:27][E][uart:015]: Reading from UART timed out at byte 0!
[09:52:27][E][uart:015]: Reading from UART timed out at byte 0!
[09:52:27][E][uart:015]: Reading from UART timed out at byte 0!
[09:52:27][E][uart:015]: Reading from UART timed out at byte 0!
[09:52:27][E][uart:015]: Reading from UART timed out at byte 0!
[09:52:28][E][uart:015]: Reading from UART timed out at byte 0!
[09:52:28][E][uart:015]: Reading from UART timed out at byte 0!
[09:52:28][E][uart:015]: Reading from UART timed out at byte 0!
[09:52:28][E][uart:015]: Reading from UART timed out at byte 0!
[09:52:28][E][uart:015]: Reading from UART timed out at byte 0!
[09:52:28][E][uart:015]: Reading from UART timed out at byte 0!
[09:52:28][D][nextion_upload:121]: Nextion reported new range 655360
[09:52:28][D][nextion_upload:302]: Heap Size 152324, Bytes left 369216
[09:52:28][D][nextion_upload:048]: Requesting range: bytes=655360-720895
[09:52:34][D][nextion_upload:302]: Heap Size 149900, Bytes left 303680
[09:52:34][D][nextion_upload:048]: Requesting range: bytes=720896-786431
```

comet424, 2022-01-20:
@jsuanet   im not getting any of that 
so yes   the 192.168.0.12   is the home assistant 
and in the www folder i made a nextion folder and placed the tfts in there
so i did the logs i dont get any of that  above 
so i been playing with the code..  as i was making 3rd display and used some of your code for screen brightness
but here is the code

`esphome:
  name: moms-car-temp-3
  platform: ESP8266
  board: d1_mini  

  on_boot:
    then:
        lambda: |-

            //auto time_text = id(display_time).now().strftime("%I:%M%P");
            //auto time_hour = id(display_time).now().strftime("%I");
            //    auto time_min = id(display_time).now().strftime("%M");
            //  auto time_sec = id(display_time).now().strftime("%S");
             // it.set_component_text("min",time_min.c_str());
            //  it.set_component_text("hour",time_hour.c_str());
            //  it.set_component_text("sec",time_sec.c_str());

# Enable logging
logger:
  baud_rate: 0
# Enable Home Assistant API
api:
  password: !secret api_password
  services:
    - service: update_nextion
      then:
        - lambda: 'id(nextion1)->upload_tft();'

ota:
  password: !secret ota_password

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
#  manual_ip:
#    static_ip: 192.168.0.234
#    gateway: 192.168.0.1
#    subnet: 255.255.255.0  
  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Car-Temp Fallback Hotspot"
    password: "QQUIre8IsCRy"

captive_portal:

# Temperature Settings    
sensor:
  - platform: dht
    model: am2302
    pin: D7
    temperature:
      name: "Mom's Car Outside Temperature"
      id: outtemp
#      filters:
#        offset: 2
    humidity:
      name: "Mom's Car Outside Humidity"
      id: outside_humidity
    update_interval: 6s

uart:
#  uart_id: uart_2
  rx_pin: D1
[... 151 lines cut ...]
|   |-- <Hash> 1.0
|   |-- <ESP8266WiFi> 1.0
|-- <DNSServer> 1.1.1
|-- <ESP8266WiFi> 1.0
|-- <ArduinoJson> 6.18.5
|-- <ESP8266HTTPClient> 1.2
|-- <ESP8266mDNS> 1.2
Compiling /data/moms-car-temp-3/.pioenvs/moms-car-temp-3/src/main.cpp.o
Linking /data/moms-car-temp-3/.pioenvs/moms-car-temp-3/firmware.elf
Building /data/moms-car-temp-3/.pioenvs/moms-car-temp-3/firmware.bin
RAM:   [====      ]  42.5% (used 34848 bytes from 81920 bytes)
Flash: [=====     ]  54.4% (used 568124 bytes from 1044464 bytes)
esp8266_copy_factory_bin(["/data/moms-car-temp-3/.pioenvs/moms-car-temp-3/firmware.bin"], ["/data/moms-car-temp-3/.pioenvs/moms-car-temp-3/firmware.elf"])
========================= [SUCCESS] Took 4.48 seconds =========================
INFO Successfully compiled program.
INFO Resolving IP address of moms-car-temp-3.local
INFO  -> 192.168.0.115
INFO Uploading /data/moms-car-temp-3/.pioenvs/moms-car-temp-3/firmware.bin (572272 bytes)
INFO Compressed to 399474 bytes
Uploading: [============================================================] 100% Done...

INFO Waiting for result...
INFO OTA successful
INFO Successfully uploaded program.
INFO Starting log output from moms-car-temp-3.local using esphome API`

comet424, 2022-01-20:
ugh why wont pasting the code work right when i paste it inbetween the ` `

jsuanet, 2022-01-21:
Please use 3 backtics for posting a multiple line code block

The second log you added, is from the update of the ESP-firmware, that will NOT update the TFT-file used by the nextion display itself.

To update the TFT-file in the nextion display via WIFI/serial you have to perform the following steps:

1. Activate a LOG session for the device and let it run. (only necessary for getting the logs).
2. In a second browser tab select `Developer Tools` in the sidebar from your Home Assistant screen.
3. Select the `Services` Tab.
4. Select the `ESPHome:{your node name}_update_nextion` service (see my screenshot).
5. Press button `CALL SERVICE` and look in the LOG browser tab what happens. 

Then you should get some of the log lines I posted before.

[image]

For the workaround to avoid the compile error, you only have to add the `http_request:` tag (This will trigger the code generator to add the necessary library). The other 2 lines in your config are not necessary. I think it is better to remove the lines:
```
  useragent: esphome/device
  timeout: 10s
```

jsuanet, 2022-02-09:
@comet424 No, it says `preparation for tft update failed 0` That's another problem and not related to this issue (the compile error and proposed workaround to solve the compile error).

Also, this is an issue tracker, not the place to ask for support. It is better to post your questions on the ESPHome community site (https://community.home-assistant.io/c/esphome/) or the ESPHome discord channel.

ssieb, 2022-07-21:
The original issue here is still valid.  For some reason, it works fine on ESP32, but on ESP8266, you get this error because the request library isn't automatically brought in.

ssieb, 2022-11-20:
Closing as duplicate of https://github.com/esphome/issues/issues/2721

closed 2022-11-20 by ssieb