esphome/issues #2920: Thermostat Climate Controller

lucagia, 2022-01-08:
### The problem

Hi everyone,
I have a problem with the "Climate" component in particular when defining the `off_mode`:
These are the triggers set:
```
    idle_action:
      - switch.turn_on: minimum
    off_mode:
      - switch.turn_off: minimum
```
  
Sometimes, when I turn off the thermostat, the switch of `minimum` remains on contrary to what is requested by `off_mode`
Thank you
Luca

### Which version of ESPHome has the issue?

2021.12.3

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2021.12.8

### What platform are you using?

ESP8266

### Board

Sonoff 4CH

### Component causing the issue

Thermostat Climate Controller

### Example YAML snippet

```yaml
climate:
  - platform: thermostat
    id: stufa
    name: "Stufa"
    sensor: temp
    default_target_temperature_low: 20 °C
    min_heating_off_time: 300s
    min_heating_run_time: 300s
    min_idle_time: 30s
    default_mode: "off"
    heat_mode:
      - switch.turn_on: accensione
    heat_action:
      - switch.turn_off: minimo
    idle_action:
      - switch.turn_on: minimo
    off_mode:
      - switch.turn_off: minimo
      - switch.turn_off: accensione
```

oxan, 2022-01-10:
See the note in the [documentation](https://esphome.io/components/climate/thermostat.html#heating-and-cooling-modes), `off_mode` shouldn't be used to turn on/off heating/cooling devices, use `idle_action` for that instead.

lucagia, 2022-01-10:
Thanks for your reply, this is what i would like to do:
[image]

do you have any suggestions?

kbx81, 2022-01-21:
@lucagia you say this "sometimes" happens...could you elaborate? For example, about how often does this happen? One in ten? One in one hundred? Are there any other components in your configuration that might also be controlling the switch? Is it possible that something else in HA is causing the switch to be turned on? Perhaps something with another automation? What appears in the logs when the switch state (doesn't) change(s)? I'm just trying to throw out ideas -- I haven't experienced this behavior myself.

lucagia, 2022-01-21:
I correct myself, it always happens.
Prerequisites: I control the main ignition of the heating system with:
```
heat_mode:
      - switch.turn_on: accensione

off_mode:
      - switch.turn_off: accensione
```

while I adjust the modulation with:
```
heat_action:
      - switch.turn_off: minimo
idle_action:
      - switch.turn_on: minimo
```
I think that the thermostat component remembers the last action given, for example, in `idle_action:` when I turn it off.
If another automation changes the  the switch state associated with the `idle_action:` (when the thermostat is OFF) the thermostat component does not notice it.
The result, in my case, is that if the thermostat is turned on `heat_mode:` and the set temperature is lower than the room temperature, the system starts to generate heat anyway

kbx81, 2022-01-23:
I had to remind myself how this works "under the hood"...so I took another look at the code and realized this has an [obvious answer](https://github.com/esphome/esphome/blob/a6a9ebfde250bea94717123bd15f63a3c21740eb/esphome/components/thermostat/thermostat_climate.cpp#L50-L51); it's specifically related to the comment in the docs as @oxan called out above.

Whenever the controller's `action` and/or `mode` changes, the respective `mode` trigger is called, then the respective `action` trigger is called. In your case, because you are sharing a switch between an `action` and a `mode`, the change made by `off_mode ` to `minimo` is immediately undone by `idle_action`. There isn't a way to fix this -- even if we transposed those two lines of code, you'd encounter this problem  in the opposite way if you were sharing a switch between, for example, `heat_mode` and `heat_action`.

I wouldn't really consider this a bug/issue. Climate modes and climate actions are two distinct, separate categories for control, as called out in the documentation @oxan linked above.

If you _must_ share this switch between both a `mode` and `action`, I think a workaround you could use to "fix" this specific issue would be to try inserting a delay into your `off_mode`:

```yaml
    off_mode:
      - delay: 1s
      - switch.turn_off: minimo
      - switch.turn_off: accensione
```

This would effectively cause the changes required by the `off` mode to take effect _after_ the climate action has completed. I hope this solves it for you! Let us know...

lucagia, 2022-01-24:
I will try to adopt your solution, thank you!

closed 2022-01-24 by lucagia