esphome/issues #2923: ESP32 w/ TSL2561 crashes at high brightness

LeonEmm, 2022-01-09:
### The problem

Hello everybody,

recently I changed from ESPEasy to ESPHome due a migration to Homeassistant. I have to ESP32 with a TSL2561 to measure my outside brightness. Since the first OTA Flash to ESPHome, my Sensor crashes at high Brigntess (>10,000lux). With the ESPEasy firmware nothing changed.
When the sensor crashes, the ESP is still reachable but is reporting 0lx. This happens on two ESP-Instances with the same configuration. Has anyone an Idea, why this happens?

### Which version of ESPHome has the issue?

2021.12.3

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2021.12.8

### What platform are you using?

ESP32

### Board

ESP32 esp-wrover-kit

### Component causing the issue

TSL2561

### Example YAML snippet

```yaml
esphome:
  name: brightsouth
  platform: ESP32
  board: esp-wrover-kit

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:
  safe_mode: true
  password: "ascan"

wifi:
  ssid: "****"
  password: "****"
  reboot_timeout: 5min
  manual_ip:
    # Set this to the IP of the ESP
    static_ip: ***.***.***.***
    # Set this to the IP address of the router. Often ends with .1
    gateway: ***.***.***.***
    # The subnet of the network. 255.255.255.0 works for most home networks.
    subnet: 255.255.255.0
  power_save_mode: none

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Brighteast Fallback Hotspot"
    password: "*****"

captive_portal:

i2c:
  sda: 21
  scl: 22
  scan: true
  id: bus_a
  frequency: 100kHz
  
sensor:
  - platform: tsl2561
    name: "Sensor South"
    address: 0x29
    update_interval: 240s
    integration_time: 14ms
    
binary_sensor:
  - platform: status
    name: "Sensor South Status"

button:
  - platform: restart
    name: "Sensor South Restart"
```

### Anything in the logs that might be useful for us?

```txt
No Error Messages reported.
```

LeonEmm, 2022-01-13:
Edit:I Changed the frequency of the i2c bus to 400kHz and the update-interval of the Sensor to 60 secondsbut still got the same Error. The Log-File says on a sunny day:

`[14:24:48][D][tsl2561:126]: Got illuminance=0.0lx
[14:24:48][D][sensor:113]: 'Sensor South': Sending state 0.00000 lx with 1 decimals of accuracy`

[image]

when the Sensor reports 0.0 lux, it is still reachable but i have to restart the ESP to get Values. From this point the sensor is working perfectly until it hits another sunny day and climbes above the 10k lux mark...

LeonEmm, 2022-03-24:
A small update:
I performed every update available and the drops ar getting less and less, but they are still threre....

B48D81EFCC, 2023-06-09:
Hi @LeonEmm,

I am experiencing a similar behavior with TSL2561.

I my case the sensor is not reporting 0, when his max value is breached, but the sensor reports nan.

`[12:08:56][W][tsl2561:068]: TSL2561 sensor is saturated.
[12:08:56][D][tsl2561:126]: Got illuminance=nanlx
[12:08:56][D][sensor:094]: 'Illuminance (TSL2561)': Sending state nan lx with 1 decimals of accuracy`

B48D81EFCC, 2025-08-12:
I have not tested it by myself, but maybe a filter can be used as workaround: https://esphome.io/components/sensor/index.html

For example a lambda filter:
```
filters:
  - lambda: |-
      if (isnan(id(SENSOR_ID).raw_state))
      {
          return 20000;
      }
```