esphome/issues #2924: HA initiated update to Modbus Number is lost

anatoly-savchenkov, 2022-01-10:
### The problem

If a new value for a Modbus number entity comes when query command for the same register is already scheduled for execution, new value is lost and is not being written to the Modbus register.

### Which version of ESPHome has the issue?

dev

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2021.12.8

### What platform are you using?

ESP8266

### Board

Wemos D1 mini

### Component causing the issue

Modbus Controller

### Example YAML snippet

```yaml
uart:
  rx_pin: RX
  tx_pin: TX
  baud_rate: 19200
  parity: EVEN

modbus:
  id: modbus1  

modbus_controller:
  - id: jl204c5m
    ## the Modbus device addr
    address: 1
    modbus_id: modbus1
    setup_priority: -10
    update_interval: 5s

sensor:
  - platform: modbus_controller
    modbus_controller_id: jl204c5m
    name: "Ventilation Power State"
    register_type: read
    address: 10
    value_type: U_WORD
    accuracy_decimals: 0

  - platform: modbus_controller
    modbus_controller_id: jl204c5m
    name: "Ventilation Status"
    register_type: read
    address: 11
    value_type: U_DWORD_R
    accuracy_decimals: 0

  - platform: modbus_controller
    modbus_controller_id: jl204c5m
    name: "Ventilation Error"
    register_type: read
    address: 19
[... 240 lines cut ...]
[22:21:15][V][modbus_controller:055]: Modbus response queued
[22:21:15][V][modbus_controller:062]: Process modbus response for address 0x3 size: 2
[22:21:15][V][modbus_controller:098]: data for register address : 0x3 : 
[22:21:15][V][modbus_controller.switch:040]: Publish 'Ventilation': new value = ON type = 3 address = 3 offset = 0
[22:21:15][V][modbus_controller:035]: Sending next modbus command to device 1 register 0x4666 count 1

>> But in reality we just read the value of 0x4666 loosing the new value.

[22:21:15][V][modbus_controller:461]: Command sent 3 0x4666 1
[22:21:15][V][modbus:225]: Modbus write raw: 01.03.46.66.00.01.13.24 (8)
[22:21:15][V][modbus:072]: Modbus received Byte  1 (0X1)
[22:21:15][V][modbus:072]: Modbus received Byte  3 (0X3)
[22:21:15][V][modbus:072]: Modbus received Byte  2 (0X2)
[22:21:15][V][modbus:072]: Modbus received Byte  17 (0X11)
[22:21:15][V][modbus:072]: Modbus received Byte  248 (0Xf8)
[22:21:15][V][modbus:072]: Modbus received Byte  181 (0Xb5)
[22:21:15][V][modbus:072]: Modbus received Byte  150 (0X96)
[22:21:15][V][modbus_controller:055]: Modbus response queued
[22:21:15][V][modbus_controller:062]: Process modbus response for address 0x4666 size: 2
[22:21:15][V][modbus_controller:098]: data for register address : 0x4666 : 
[22:21:15][D][modbus.number:023]: Number new state : 46.00
[22:21:15][D][number:036]: 'Ventilation Mix Valve': Sending state 46.000000

>> The value of 49 is lost.
```

martgras, 2022-01-10:
TLDR;

There is a bug but maybe enforced by the short update interval or the device not responding in time. 

I can't reproduce the problem 
Actually you should never see "Duplicate modbus command found"  - this is nearly in all cases a communication problem with the device.   Maybe increase  `update_interval` for testing.  5s is pretty short 

```yaml
number:
  - platform: modbus_controller
    modbus_controller_id: epever
    id: battery_capacity_number
    name: "Battery Cap Number"
    address: 0x9001
    response_size: 1
    value_type: U_WORD
```
Logs

```
[16:16:58][D][modbus.number:023]: Number new state : 115.00
[16:16:58][D][number:036]: 'Battery Cap Number': Sending state 115.000000
[16:17:01][D][number:010]: 'Battery Cap Number' - Setting
[16:17:01][D][number:029]:   Value: 100.000000
[16:17:01][D][modbus.number:055]: Updating register: connected Sensor=Battery Cap Number start address=0x9001 register count=1 new value=100.00 (val=100.00)
[16:17:01][V][modbus_controller:036]: Sending next modbus command to device 1 register 0x9001 count 1
[16:17:01][V][modbus:184]: Modbus write: 01.06.90.01.00.64.F4.E1 (8)
[16:17:01][V][modbus_controller:461]: Command sent 6 0x9001 1
[16:17:01][V][modbus:058]: Modbus received Byte  1 (0X1)
[16:17:01][V][modbus:058]: Modbus received Byte  6 (0X6)
[16:17:01][V][modbus:058]: Modbus received Byte  144 (0X90)
[16:17:01][V][modbus:058]: Modbus received Byte  1 (0X1)
[16:17:01][V][modbus:058]: Modbus received Byte  0 (0X0)
[16:17:01][V][modbus:058]: Modbus received Byte  100 (0X64)
[16:17:01][V][modbus:058]: Modbus received Byte  244 (0Xf4)
[16:17:01][V][modbus:058]: Modbus received Byte  225 (0Xe1)
[16:17:01][V][modbus_controller:055]: Modbus response queued
[16:17:01][V][modbus_controller:063]: Process modbus response for address 0x9001 size: 4
[16:17:01][V][modbus_controller:308]: Command ACK 0x9001 256
```
Note  the modbus command ` 01.06.90.01.00.64.F4.E1`. The second byte is the modbus function code. 06 means "preset single register"  followed by the Command ACK line
And from here changing the value to 101 and then 115

```
[16:17:01][D][number:036]: 'Battery Cap Number': Sending state 100.000000
[16:17:32][D][number:010]: 'Battery Cap Number' - Setting
[16:17:32][D][number:029]:   Value: 101.000000
[16:17:32][D][modbus.number:055]: Updating register: connected Sensor=Battery Cap Number start address=0x9001 register count=1 new value=101.00 (val=101.00)
[16:17:32][V][modbus_controller:036]: Sending next modbus command to device 1 register 0x9001 count 1
[16:17:32][V][modbus:184]: Modbus write: 01.06.90.01.00.65.35.21 (8)
[16:17:32][V][modbus_controller:461]: Command sent 6 0x9001 1
[16:17:33][V][modbus:058]: Modbus received Byte  1 (0X1)
[16:17:33][V][modbus:058]: Modbus received Byte  6 (0X6)
[16:17:33][V][modbus:058]: Modbus received Byte  144 (0X90)
[16:17:33][V][modbus:058]: Modbus received Byte  1 (0X1)
[16:17:33][V][modbus:058]: Modbus received Byte  0 (0X0)
[16:17:33][V][modbus:058]: Modbus received Byte  101 (0X65)
[16:17:33][V][modbus:058]: Modbus received Byte  53 (0X35)
[16:17:33][V][modbus:058]: Modbus received Byte  33 (0X21)
[16:17:33][V][modbus_controller:055]: Modbus response queued
[16:17:33][V][modbus_controller:063]: Process modbus response for address 0x9001 size: 4
[16:17:33][V][modbus_controller:308]: Command ACK 0x9001 256
[16:17:33][D][number:036]: 'Battery Cap Number': Sending state 101.000000
[16:17:36][D][number:010]: 'Battery Cap Number' - Setting
[16:17:36][D][number:029]:   Value: 115.000000
[16:17:36][D][modbus.number:055]: Updating register: connected Sensor=Battery Cap Number start address=0x9001 register count=1 new value=115.00 (val=115.00)
[16:17:36][V][modbus_controller:036]: Sending next modbus command to device 1 register 0x9001 count 1
[16:17:36][V][modbus:184]: Modbus write: 01.06.90.01.00.73.B4.EF (8)
[16:17:37][V][modbus_controller:461]: Command sent 6 0x9001 1
[16:17:37][V][modbus:058]: Modbus received Byte  1 (0X1)
[16:17:37][V][modbus:058]: Modbus received Byte  6 (0X6)
[16:17:37][V][modbus:058]: Modbus received Byte  144 (0X90)
[16:17:37][V][modbus:058]: Modbus received Byte  1 (0X1)
[16:17:37][V][modbus:058]: Modbus received Byte  0 (0X0)
[16:17:37][V][modbus:058]: Modbus received Byte  115 (0X73)
[16:17:37][V][modbus:058]: Modbus received Byte  180 (0Xb4)
[16:17:37][V][modbus:058]: Modbus received Byte  239 (0Xef)
[16:17:37][V][modbus_controller:055]: Modbus response queued
[16:17:37][V][component:199]: Component modbus took a long time for an operation (0.05 s).
[16:17:37][V][component:200]: Components should block for at most 20-30ms.
[16:17:37][V][modbus_controller:063]: Process modbus response for address 0x9001 size: 4
[16:17:37][V][modbus_controller:308]: Command ACK 0x9001 256
[16:17:37][D][number:036]: 'Battery Cap Number': Sending state 115.000000
```

I can't sport a write command (01.06.... ) nor a Command ACK so the command never went out. 

But reviewing the code shows that the modbus function code is not considered when checking for duplicate commands  
PR#3031 will fix it . For testing use: 

```yaml
external_components:
  - source: github://pr#3031
    components: ["modbus_controller"]
```

anatoly-savchenkov, 2022-01-10:
I think you are missing at least a part of my point. My setup has rather long list of registers (namely, 14 ranges). Command queue is created in the very beginning of the update cycle. After that controller starts to execute commands from the queue one by one. All 14 ranges complete within 1 second (looking at the log file). But here is the point where race happens between execution of transactions from the queue and updates coming from HA. In the fast majority of cases these two events don't interfere, and I see logs like yours. But several times per day those two events clash and I see the log like in my original report.

I see your fix but I'm not sure it's ideal. With your proposed fix write transaction (code = 6) will be added to the end of the command queue. After that queue will have read command followed by write command to the same register. When read command will be executed it will read out old value and thus, it will overwrite HA value with the value from the device. Write command will be executed as well and update the value on the device side. So, only the next iteration will read it back. So, on the HA side value will bounce back and forth.

I think ideal fix should overwrite read command in the queue with the write command. So, instead of checking for equality of item->function_code == command.function_code it would be better to overwrite function code as well inside the condition just like payload is overwritten.

martgras, 2022-01-10:
No the correct value should be published See https://github.com/esphome/esphome/blob/073828235f67318df933fbe7350a3e0dc17bb773/esphome/components/modbus_controller/number/modbus_number.cpp#L70
You can see that in the logs as well if you use a longer update interval `Command ACK 0x9001` is followed by publish new state 
From what I see if you are unlucky - a read is sent => HA old value  followed by write => HA new value.  So maybe HA gets updated in a short interval and the read is not really necessary but values should be correct

anatoly-savchenkov, 2022-01-10:
I've increased modbus polling rate to 1s and manually generated changes from HA. Fix works well. Thanks!

I'm close to my ultimate setup with 7 modbus devices on the same bus, more testing will come soon. One problem I already expect is heap size limitation on ESP8266. Modbus controller consumes visible amount of RAM :(

martgras, 2022-01-10:
Use an esp32. Not only has it more memory it also has 2 Hardware uart's   (no problem with serial logging while using modbus... )  
And in my experience you run out of stack space before heap becomes a problem

closed 2022-01-12 by jesserockz