esphome/issues #2927: heatpumpir fan speeds mapping is hardcoded

jeffborg, 2022-01-10:
### The problem

Esphome appears to have only 3 fan speeds for climate, but some heatpumps via the arduino heatpumpir library have 3, 4 or 5, The mapping is hardcoded.

It appears the heatpumpir library seems to use fan speeds starting from 1 regardless of the number of speeds supported. 

see the code at https://github.com/esphome/esphome/blob/dev/esphome/components/heatpumpir/heatpumpir.cpp#L143

At the moment because I needed to fix https://github.com/esphome/esphome/pull/3025 I've just fixed the hard coding locally. But this isn't a suitable solution.

In my case if you choose fan speed high the air conditioner goes into auto mode. because speed 4 isn't mapped anywhere.

### Which version of ESPHome has the issue?

2021.12.3

### What type of installation are you using?

Docker

### What platform are you using?

ESP32

### Board

esp32 dev

### Component causing the issue

heatpumpir

### Additional information

The code block in question. 
```
  switch (this->fan_mode.value_or(climate::CLIMATE_FAN_AUTO)) {
    case climate::CLIMATE_FAN_LOW:
      fan_speed_cmd = FAN_2;
      break;
    case climate::CLIMATE_FAN_MEDIUM:
      fan_speed_cmd = FAN_3;
      break;
    case climate::CLIMATE_FAN_HIGH:
      fan_speed_cmd = FAN_4;
      break;
    case climate::CLIMATE_FAN_AUTO:
    default:
      fan_speed_cmd = FAN_AUTO;
      break;
  }
```

jeffborg, 2022-01-10:
@rob-deutsch My thoughts on mapping this is to add a config setting for number of fan speeds, 3 , 4 or 5 And map accordingly.
 Further to this each type of unit the number of fan speeds is already known via the code in heatpumpir library

OttoWinter, 2022-01-18:
There's [`custom_fan_mode`](https://github.com/esphome/esphome/blob/dev/esphome/components/climate/climate.h#L207) for that, It would just need to be implemented in the heatpumpir integration.

tomaswilhelmsson, 2024-02-12:
So any news on this? Installed the climate control from my IVT heatpump and noticed this error also and tracked it down to this, why doesnt LOW start on FAN_1 etc insted of FAN_2 ? with current source i cannot reach lowest fan speed on my heatpump.

N0m4n904, 2024-08-30:
Still no news on this? I am having the same issue. When I set the fan mode to middle, the climate is set to high. Same for Swing mode. The climate has only vertical swing, but I need to enable horizontal swing to activate it