esphome/issues #2929: Repeat turns into infinite loop if triggered again while first sequence still running.

MikeS1976, 2022-01-11:
### The problem

On one of my ESPs i have a service that opens a door, and flashes a light for 15 times, then closed the door.
If the service is re-triggered while the light flashing is still running, it turns into an endless loop.

`api:
  services:
    - service: service_open_door
      then:
        - switch.turn_on: open_door
        - repeat:
            count: 15
            then:
              - light.turn_on: light_led_blue
              - delay: 0.1s
              - light.turn_off: light_led_blue
              - delay: 0.1s
        - switch.turn_off: open_door`

if i replace the repeat by the snippet below and retrigger it does not end up in a loop.....:

`        - switch.turn_on: open_door
        - light.turn_on: light_led_blue
        - delay: 0.15s
        - light.turn_off: light_led_blue
        - delay: 0.15s
        - light.turn_on: light_led_blue
        - delay: 0.15s
        - light.turn_off: light_led_blue
        - delay: 0.15s
        - light.turn_on: light_led_blue
        - delay: 0.15s
        - light.turn_off: light_led_blue
        - delay: 0.15s
        - light.turn_on: light_led_blue
        - delay: 0.15s
        - light.turn_off: light_led_blue
        - delay: 0.15s
        - light.turn_on: light_led_blue
        - delay: 0.15s
        - light.turn_off: light_led_blue
        - delay: 0.15s
        - light.turn_on: light_led_blue
        - delay: 0.15s
        - light.turn_off: light_led_blue
        - delay: 0.15s
        - light.turn_on: light_led_blue
        - delay: 0.15s
        - light.turn_off: light_led_blue
        - delay: 0.15s
        - light.turn_on: light_led_blue
        - delay: 0.15s
        - light.turn_off: light_led_blue
        - delay: 0.15s
        - light.turn_on: light_led_blue
        - delay: 0.15s
        - light.turn_off: light_led_blue
        - delay: 0.15s
        - light.turn_on: light_led_blue
        - delay: 0.15s
        - light.turn_off: light_led_blue
        - delay: 0.15s
        - switch.turn_off: open_door`

### Which version of ESPHome has the issue?

2021.12.3

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2021.12.8

### What platform are you using?

ESP8266

### Board

wemos d1 mini

### Component causing the issue

-

### Example YAML snippet

```yaml
api:
  services:
    - service: service_open_door
      then:
        - switch.turn_on: open_door
        - repeat:
            count: 15
            then:
              - light.turn_on: light_led_blue
              - delay: 0.1s
              - light.turn_off: light_led_blue
              - delay: 0.1s
        - switch.turn_off: open_door
```

### Anything in the logs that might be useful for us?

```txt
-
```

### Additional information

-

jdrowell, 2022-12-29:
I'm getting the same thing when I use a repeat action that is reentered (by responding to a second event). Not finding anything in the documentation that would prevent this. If I have to manually track the repeat loop state, it is probably easier to just use wait_until with a variable.

jdrowell, 2022-12-29:
I looked at the generated code (which includes the YAML in comments, nice!) and it seems fine, it just sets up the calls to ESPHome code. My best guess is that in this check

https://github.com/esphome/esphome/blob/119a6920f2a82a9f77e42158c8fdf69755803a11/esphome/core/base_automation.h#L242

the iteration is being incremented twice if the automation is run again, so the == is never caught, and then it never ends. Maybe use >= there. I "fixed" my case with an "if" before the "repeat", plan to look back into this if I get some time.

ssieb, 2022-12-29:
You shouldn't have actions that take a long time in a trigger.  The behaviour is somewhat undefined.  You should use a script and trigger that instead.

jdrowell, 2022-12-29:
Ah good to know, I'll refactor that! Still, there seems to be reentrency issues with repeat, which should be fixed to avoid other corner cases.

ssieb, 2022-12-29:
Yes, I completely agree.

gerchikov, 2023-02-23:
+1 to this issue. Still happens on ESPHome 2023.2.2 running on ESP32. I also agree that the condition at [base_automation.h#L242](https://github.com/esphome/esphome/blob/119a6920f2a82a9f77e42158c8fdf69755803a11/esphome/core/base_automation.h#L242) does not seem robust.

gerchikov, 2023-02-23:
And by the way: the advice to use scripts is well taken, BUT: the same defect in the implementation of `repeat` also impacts scripts. If a script that contains `repeat` is allowed to run multiple instances (`mode: parallel`), then concurrent executions interfere with each others' iteration counter and still result in an infinite loop.

oxan, 2023-02-23:
esphome/esphome#4480.

closed 2023-02-26 by jesserockz