esphome/issues #2930: TCS34725 broke with recent changes, endian is wrong

ceandraka, 2022-01-12:
### The problem

Previously working code to determine color of LED (red or green) now fails, and return values for RGB and clear are very small.

Loading a copy of 2021.10.3 or older as a custom component (change name throughout) fixes the issue. Expected percentages are in the range of 10-40%, but with new version they are well under 0.1%. Using older version of TCS34725 as custom component restores proper operation.

Pull request https://github.com/esphome/esphome/pull/2445 indicates that endian was swapped because documentation showed little endian in ESP but big endian in I2C. this was implemented on lines 185-190 in TCS34725.cpp. However, this calls read_byte_16 in i2c.cpp, which in turn calls read_bytes_16, which has the endian swap already.

Please remove lines 185-190 in the newest TCS34725.cpp to fix this issue.

Thanks

### Which version of ESPHome has the issue?

2021.11.0 and later

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2021.12.7

### What platform are you using?

ESP32

### Board

d1 mini and TCS34725

### Component causing the issue

TCS34725

### Example YAML snippet

```yaml
- platform: tcs34725old
    red_channel:
      name: "TCS34725 Red"
      internal: true
      id: red_bright

    green_channel:
      name: "TCS34725 Green"
      internal: true
      id: green_bright

    clear_channel:
      name: "TCS34725 Clear"
      internal: true
      id: clear_bright

    blue_channel:
      name: "TCS34725 Blue"
      internal: true
      id: blue_bright
      on_value:
        then:
          - script.execute: read_colors

    gain: 60x
#    integration_time: 2.4ms
    integration_time: 101ms
    address: 0x29
    update_interval: 200ms
```

### Anything in the logs that might be useful for us?

```txt
No error message, just wrong values for RGB and Clear
```

ceandraka, 2022-01-12:
I took latest version (2021.12.3), copied files for tcs34725 to my external components directory, removed the 4 offending lines, tested, and it works properly. I will run this as an external component until it is fixed. Just lines 185-190 were commented out.

martgras, 2022-01-12:
This should work as well now 
```yaml
external_components:
  - source: github://pr#3037
    refresh: 1h
    components: ["tcs34725"]
```
My PR removed the duplicate conversion - exactly the change you described - Thanks for debugging it to that level
Can't test but I don't expect a problem

ceandraka, 2022-01-12:
I changed my external component to point to your PR, and it works as well.

Thanks!

martgras, 2022-01-15:
Actually I think the exiting code is correct
From https://cdn-shop.adafruit.com/datasheets/TCS34725.pdf

Table 14. ADC Channel Data Registers
| Register | ADDRESS | BITS | DESCRIPTION          |
|----------|---------|------|----------------------|
| CDATA    | 0x14    | 7:0  | Clear data low byte  |
| CDATAH   | 0x15    | 7:0  | Clear data high byte |
| RDATA    | 0x16    | 7:0  | Red data low byte |
| RDATAH   | 0x17    | 7:0  | Red data high byte  |
| GDATA    | 0x18    | 7:0  | Green data low byte |
| GDATAH   | 0x19    | 7:0  | Green data high byte |
| BDATA    | 0x1A    | 7:0  | Blue data low byte |
| BDATAH   | 0x1B    | 7:0  |  Blue data high byte  |

So this is actually little endian and not big endian  as used by most other i2c devices 

The reason for the low C RGB values is caused by the scaling approach taken. The assumption is that the max value is 65535 and the results for the values are returned as (raw_value/65535 ) * 100.  
But actually the max value depends on the integration time. The TTCS34725 accumulates 1024 counts per 2.4ms of integration time, up to a maximum values of 65535.  Therefore scaling should be based on `integration time * 1024 /  2.4ms `

I'm going to update the PR in the next days with this suggestion:

Since R + G + B  roughly equals C my proposal is returning R,G and B as percentage of C  `R = (raw_r / raw_c ) * 100`
C will be returned as `C = raw_c / (integration time * 1024 /  2.4) * 100`

In my tests this returns reasonable results

ceandraka, 2022-01-15:
@martgras Thanks for continuing to look at this. You appear to be on to something. Unfortunately the user base for this device appears small to non-existent.

My application (I have exactly one of these) is to look at an led on a pump controller. I built a housing with magnets to hold the board against the LED display and exclude external light. The three possible states are "On" (Green light on", "Running", green light blinking, and "Error", red light blinking, count blinks to determine error. 

I was able to find a combination that works, but indeed it could be struggling with reversed endian. When green is on, I also get a red indication, so I need to look for red and green together to indicate green. I thought it had to do with the spectrum of the led.

I had also noted, some time ago, that i2c was inconsistent in how endian was handled, based on trying to set a 2-byte register on this device. However, it well might be that i2c is correct, and this device (or code) was wrong. I forget what I was trying to do, but I had modified the TCS34725 code to set a 16-byte register. When I set it and then read it back, it was swapped. I never got things working and walked away. I was assuming read word was OK, and write was backward, but perhaps read was wrong and write was right!

I will look at the code later and see if I have any thoughts on your approach. Thanks for looking!

ceandraka, 2022-01-15:
It looks like I was, at one point, trying to set up interrupts instead of polling, which required setting 2-byte threshold registers. This is where I had the discrepancy. I believe i2c was actually writing one byte at a time, which is the only way that appears supported, but the read back was reading a word. I never got it working (likely I was putting the bytes in backward), and implemented polling approach in my project.

I do note that in the example at https://esphome.io/components/sensor/tcs34725.html, Red, Green, and Blue show percentage levels, while clear is 0%. This should not be the case! However, the wording in that page indicates the percentages are a "percentage of how strongly the color channel is activated", which would be consistent with the method used (divide by the max possible, 65535). But, as you state, this is dependent on integration time too. With smaller integration times, this can result in very small responses.

Your proposed method should work, and will give some indication of RGB relative content, but not overall signal strength. That is fine, and perhaps more useful, but the documentation may need to be updated to reflect this. The current method, dividing by saturation levels, results in very small percentages for dim light without full integration times. Even with 101ms integration time, the percentages reported were extremely small. My integration time is limited by the blink duration.

Again, thanks for working on this. Let me know if/when you have a test code available and I will link it in using external components and let you know how its working out.

ceandraka, 2022-01-15:
I think there is an issue with your approach. I played around with it a little bit.

I believe my light levels are so low that they are an issue. I have to work on how to resolve that, perhaps have to have a custom component with large digital gains (multipliers). I cannot have more integration time because of the timing of the blinks on the led.

Here is the issue: When the LED is off (blinking), raw_c is zero and the results are a NaN for R, G, B.

So I think we need to keep with the original intent of R, G, B, and C being the fraction of saturation, perhaps with your integration time adjustment.

ceandraka, 2022-01-15:
I was able to verify that you are correct on endian, and I was wrong. My signal levels are very low. With gain of 60x, and integration time of 101ms, my raw values for green LED on are 32, 43, 8, 94. The other byte is always 00. When converted to percentage, whatever method is used, these result in less than 0.1 percent, so show up as zeros. When endian was wrong, it basically multiplied all of these by 1024.

So I am going to have to rethink this. I may need a custom component. Or we can add an output to ESPHome version that provides the raw integer data from the sensor, and I can do my analysis based on that. Even though my levels are in the noise, they are stable and repeatable, since I block out ambient light. So it works for me even though the levels are so low.

For now I am running a version with the endian correct, but with the color values equal to the raw values (no scaling).

Thanks again

ceandraka, 2022-01-15:
Looking at the Adafruit code (which was used in the recent update), they provide RGB color in pretty much the way you proposed (they normalize to 255 rather than percent). However, they check the clear value, and if 0 they set all channels to 0. They do not return a clear value, just RGB.

I suppose if you did it the way you proposed, and in addition set to 0 if clear is 0, then it should work, and would still work in my case where values are very small.

Thanks
Chuck

martgras, 2022-01-16:
I pushed the new version to the PR. Please let me know if the results make sense for you

ceandraka, 2022-01-16:
I compiled and tested on my limited application, and it appears to be working.

It should be noted that the new RGB output differs from documentation. It is now consistent with Adafruit version. Clear channel is also changed, and appears more appropriate. However, these changes need to be in the documentation, and may result in this being a "breaking" change.

Thanks for your work on this.

closed 2022-01-24 by jesserockz