esphome/issues #2931: ESP32 C3 devkit is not working

TAMCTec, 2022-01-12:
### The problem

I download esphome from hassio, and run the first test with esp32 c3, with esp-idf framework. I haven't add any component yet, just the simple setup. After installed, it seem keep rebooting, printing the following information again and again.
```
[12:33:54]ESP-ROM:esp32c3-api1-20210207
[12:33:54]Build:Feb  7 2021
[12:33:54]rst:0x7 (TG0WDT_SYS_RST),boot:0xc (SPI_FAST_FLASH_BOOT)
[12:33:54]Saved PC:0x40049a42
[12:33:54]SPIWP:0xee
[12:33:54]mode:QIO, clock div:1
[12:33:54]load:0x3fcd6100,len:0x17f4
[12:33:54]ets_loader.c 78
```

### Which version of ESPHome has the issue?

2021.12.3 dev

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

Home Assistant 2021.12.8

### What platform are you using?

ESP32-IDF

### Board

esp32-c3-devkitM-1

### Component causing the issue

None

### Example YAML snippet

```yaml
esphome:
  name: c3test

esp32:
  board: esp32-c3-devkitm-1
  variant: ESP32C3
  framework:
    type: esp-idf

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:
  password: "*********************"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "C3Test Fallback Hotspot"
    password: "V6OmdKMap4I5"

# captive_portal:
```
```

martgras, 2022-01-12:
Can you try if this works for you
```yaml
esphome:
  name: c3test
  platformio_options:
    board_build.flash_mode: dio  
esp32:
  board: esp32-c3-devkitm-1
  variant: ESP32C3
  framework:
    type: esp-idf

```

martgras, 2022-01-12:
I think this is https://github.com/platformio/platform-espressif32/issues/622   because the fix hasn't merged into master

markusg1234, 2022-01-13:
I have a similar issue if using ESP flasher to load the firmware. Also in the console it displays invalid header 0xffffffff.

Connecting the ESP32C3 to my Hass server and programing via USB works correctly.

TAMCTec, 2022-01-14:
@martgras Yes it totally fix my problem! Thank you!

TAMCTec, 2022-01-14:
@markusg1234 i have the similar problem before, have you try to erase flash before install?

TAMCTec, 2022-01-14:
As my problem solved, I’m closing this issue.

closed 2022-01-14 by TAMCTec

platima, 2022-05-04:
Yeah making it DIO worked for me

mdaviesnz, 2022-05-12:
@platima Did you do anything else? Setting DIO didn't help in my case with a Wemos ESP32-C3

platima, 2022-05-12:
For that exact issue all I did was DIO, but for my ESP32-C3 I ended up using a pile of changes in my YAML. Here's my header;

```
external_components:
  - source: github://pr#3454
    components:
      - deep_sleep

esphome:
  name: s01-blue
  comment: DFRobot Beetle monitoring garden status, solar powered w/ battery. Labelled Sensor01
  platformio_options:
    board_build.flash_mode: dio
    platform_packages:
      - espressif/toolchain-riscv32-esp @ https://github.com/Jason2866/crosstool-NG/releases/download/8.4.0/riscv32-esp-elf.tar.gz
      - espressif/toolchain-xtensa-esp32s2 @ https://github.com/Jason2866/crosstool-NG/releases/download/8.4.0/xtensa-esp32s2-elf.tar.gz
      
# Product page: https://wiki.dfrobot.com/SKU_DFR0868_Beetle_ESP32_C3#target_0
# Chip data sheet: https://www.espressif.com/sites/default/files/documentation/esp32-c3_hardware_design_guidelines_en.pdf

esp32:
  board: esp32-c3-devkitm-1
  framework:
    type: esp-idf
```

mdaviesnz, 2022-05-19:
I'm still struggling with this and the Wemos ESP32-C3, possibly related to this? https://github.com/esphome/esphome/pull/3405

mdaviesnz, 2022-05-19:
Tried building with the esphome dev version but still not functioning. 
Seems to flash alright but never comes back after resetting:
`esptool.py v3.3.1
Serial port COM7
Connecting...
Chip is ESP32-C3 (revision 3)
Features: Wi-Fi
Crystal is 40MHz
MAC: 60:55:f9:28:36:c0
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
Auto-detected Flash size: 4MB
Flash will be erased from 0x00010000 to 0x000bdfff...
Flash will be erased from 0x00000000 to 0x00005fff...
Flash will be erased from 0x00008000 to 0x00008fff...
Flash will be erased from 0x0000d000 to 0x0000efff...
Compressed 711088 bytes to 421566...
Wrote 711088 bytes (421566 compressed) at 0x00010000 in 10.4 seconds (effective 548.9 kbit/s)...
Hash of data verified.
Compressed 20656 bytes to 12205...
Wrote 20656 bytes (12205 compressed) at 0x00000000 in 0.5 seconds (effective 306.9 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 129...
Wrote 3072 bytes (129 compressed) at 0x00008000 in 0.1 seconds (effective 227.3 kbit/s)...
Hash of data verified.
Compressed 8192 bytes to 31...
Wrote 8192 bytes (31 compressed) at 0x0000d000 in 0.2 seconds (effective 392.1 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
←[32mINFO Successfully uploaded program.←[0m
←[32mINFO Starting log output from COM7 with baud rate 115200←[0m
`

mdaviesnz, 2022-05-19:
Managed to get it to install/boot with nothing in the esphome yaml file, just:

> esphome:
>   name: officec3
>   platformio_options:
>     board_build.flash_mode: dio  
> 
> esp32:
>   board: esp32-c3-devkitm-1
>   variant: ESP32C3  
>   framework:
>     type: esp-idf
>     version: latest

> ESP-ROM:esp32c3-api1-20210207
> Build:Feb  7 2021
> rst:0x15 (USB_UART_CHIP_RESET),boot:0xd (SPI_FAST_FLASH_BOOT)
> Saved PC:0x4200b900
> SPIWP:0xee
> mode:DIO, clock div:1
> load:0x3fcd6100,len:0x19c8
> load:0x403ce000,len:0x8d4
> load:0x403d0000,len:0x2dc0
> entry 0x403ce000
> I (24) boot: ESP-IDF 4.3.2 2nd stage bootloader
> I (24) boot: compile time 22:23:47
> I (25) boot: chip revision: 3
> I (26) boot_comm: chip revision: 3, min. bootloader chip revision: 0
> I (33) boot.esp32c3: SPI Speed      : 80MHz
> I (38) boot.esp32c3: SPI Mode       : DIO
> I (42) boot.esp32c3: SPI Flash Size : 4MB
> I (47) boot: Enabling RNG early entropy source...
> I (53) boot: Partition Table:
> I (56) boot: ## Label            Usage          Type ST Offset   Length
> I (63) boot:  0 nvs              WiFi data        01 02 00009000 00004000
> I (71) boot:  1 otadata          OTA data         01 00 0000d000 00002000
> I (78) boot:  2 phy_init         RF data          01 01 0000f000 00001000
> I (86) boot:  3 app0             OTA app          00 10 00010000 001c0000
> I (93) boot:  4 app1             OTA app          00 11 001d0000 001c0000
> I (101) boot: End of partition table
> I (105) boot_comm: chip revision: 3, min. application chip revision: 0
> I (112) esp_image: segment 0: paddr=00010020 vaddr=3c020020 size=0cbe8h ( 52200) map
> I (129) esp_image: segment 1: paddr=0001cc10 vaddr=3fc89600 size=01aech (  6892) load
> I (130) esp_image: segment 2: paddr=0001e704 vaddr=40380000 size=01914h (  6420) load
> I (139) esp_image: segment 3: paddr=00020020 vaddr=42000020 size=194d0h (103632) map
> I (162) esp_image: segment 4: paddr=000394f8 vaddr=40381914 size=07b98h ( 31640) load
> I (168) esp_image: segment 5: paddr=00041098 vaddr=50000000 size=00010h (    16) load
> I (172) boot: Loaded app from partition at offset 0x10000
> I (174) boot: Disabling RNG early entropy source...
> I (190) cpu_start: Pro cpu up.
> I (850) cpu_start: Pro cpu start user code
> I (850) cpu_start: cpu freq: 160000000
> I (850) cpu_start: Application information:
> I (853) cpu_start: Project name:     officec3
> I (858) cpu_start: App version:      2022.6.0-dev
> I (863) cpu_start: Compile time:     May 19 2022 22:23:03
> I (869) cpu_start: ELF file SHA256:  c8556d3abd473a9d...
> I (875) cpu_start: ESP-IDF:          4.3.2
> I (880) heap_init: Initializing. RAM available for dynamic allocation:
> I (887) heap_init: At 3FC8C060 len 00033FA0 (207 KiB): DRAM
> I (893) heap_init: At 3FCC0000 len 0001F060 (124 KiB): STACK/DRAM
> I (900) heap_init: At 50000010 len 00001FF0 (7 KiB): RTCRAM

Not very useful though...