esphome/issues #2933: Modbus controller stops working in case of multiple devices and one stopping responding

anatoly-savchenkov, 2022-01-12:
### The problem

I've re-connected my HVAC room units from the vendor's controller to ESPHome powered DYI controller prototype.  Good news is that basic things for all 7 devices work ok. However, write request/read response serialization logic stops working if one device stops responding. Log starts to look really weird. In the log below device with address 1 is disconnected. But you can see sporadic communication problems with other devices as well. Sometimes commands are successful but success rate is quite low. Sequential write raw log entries are very suspicious.

### Which version of ESPHome has the issue?

dev with modbus queue PR

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2021.12.8

### What platform are you using?

ESP8266

### Board

Wemos D1 mini

### Component causing the issue

Modbus Controller

### Example YAML snippet

```yaml
modbus_controller:
  - id: jl204c5m
    ## the Modbus device addr
    address: 1
    modbus_id: modbus1
    setup_priority: -10
    update_interval: 5s

  - id: jl201_10
    ## the Modbus device addr
    address: 10
    modbus_id: modbus1
    setup_priority: -10
    update_interval: 5s

  - id: jl201_11
    ## the Modbus device addr
    address: 11
    modbus_id: modbus1
    setup_priority: -10
    update_interval: 5s
  
  - id: jl201_12
    ## the Modbus device addr
    address: 12
    modbus_id: modbus1
    setup_priority: -10
    update_interval: 5s

  - id: jl201_13
    ## the Modbus device addr
    address: 13
    modbus_id: modbus1
    setup_priority: -10
    update_interval: 5s

  - id: jl201_14
    ## the Modbus device addr
    address: 14
[... 538 lines cut ...]
[23:27:46][V][modbus:075]: Modbus received Byte  2 (0X2)
[23:27:46][V][modbus:075]: Modbus received Byte  91 (0X5b)
[23:27:46][V][modbus:075]: Modbus received Byte  217 (0Xd9)
[23:27:46][V][modbus:075]: Modbus received Byte  83 (0X53)
[23:27:46][V][modbus:075]: Modbus received Byte  155 (0X9b)
[23:27:46][V][modbus_controller:035]: Sending next modbus command to device 1 register 0x13 count 4
[23:27:46][V][modbus_controller:461]: Command sent 4 0x13 4
[23:27:46][V][modbus_controller:035]: Sending next modbus command to device 15 register 0x09 count 1
[23:27:46][V][modbus_controller:461]: Command sent 4 0x9 1
[23:27:46][V][modbus:208]: Modbus write raw: 0E.04.00.09.00.01 (6)
[23:27:46][V][modbus:075]: Modbus received Byte  14 (0Xe)
[23:27:46][V][modbus:075]: Modbus received Byte  4 (0X4)
[23:27:46][V][modbus:075]: Modbus received Byte  2 (0X2)
[23:27:46][V][modbus:075]: Modbus received Byte  92 (0X5c)
[23:27:46][V][modbus:075]: Modbus received Byte  13 (0Xd)
[23:27:46][V][modbus:075]: Modbus received Byte  21 (0X15)
[23:27:46][V][modbus:075]: Modbus received Byte  244 (0Xf4)
[23:27:46][V][modbus_controller:035]: Sending next modbus command to device 1 register 0x13 count 4
[23:27:46][V][modbus_controller:461]: Command sent 4 0x13 4
[23:27:46][V][modbus_controller:035]: Sending next modbus command to device 15 register 0x09 count 1
[23:27:46][V][modbus_controller:461]: Command sent 4 0x9 1
[23:27:46][V][modbus_controller:158]: Updating modbus component
[23:27:46][V][modbus_controller:124]: Range : 9 Size: 1 (4) skip: 0
[23:27:46][V][modbus:208]: Modbus write raw: 01.04.79.18.00.01 (6)
```

nagyrobi, 2022-01-14:
Did you try with an ESP32?

lukedukeus, 2022-04-14:
I am having the same issue. When I see the log:
```
[18:14:56][V][modbus:184]: Modbus write: 02.04.10.0F.00.01.05.3A (8)
[18:14:56][V][modbus_controller:487]: Command sent 4 0x100F 1
[18:14:56][V][modbus:058]: Modbus received Byte  2 (0X2)
[18:14:56][V][modbus:058]: Modbus received Byte  4 (0X4)
[18:14:56][V][modbus:058]: Modbus received Byte  16 (0X10)
[18:14:56][V][modbus:058]: Modbus received Byte  15 (0Xf)
[18:14:56][V][modbus:058]: Modbus received Byte  0 (0X0)
[18:14:56][V][modbus:058]: Modbus received Byte  1 (0X1)
[18:14:56][V][modbus:058]: Modbus received Byte  5 (0X5)
[18:14:56][V][modbus:058]: Modbus received Byte  58 (0X3a)
```

It looks like it is recieving what it sent out? What is the underlying issue here?