esphome/issues #2934: ESP32-C3 remote receiver fails to initialize

rpatel3001, 2022-01-13:
### The problem

I'm trying out the M5Stack IR receiver/transmitter combo. Setting up the receiver to dump all received messages fails to set up the RMT peripheral. I have a hunch it's related to the channel selection, as the config dump shows channel 0, but the [docs](https://docs.espressif.com/projects/esp-idf/en/latest/esp32c3/api-reference/peripherals/rmt.html) say 0/1 are TX only and 2/3 are RX only.

### Which version of ESPHome has the issue?

2021.12.3

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

n/a

### What platform are you using?

ESP32-IDF

### Board

M5Stack Stamp C3

### Component causing the issue

remote_receiver

### Example YAML snippet

```yaml
esphome:
  name: env

esp32:
  board: esp32-c3-devkitm-1
  variant: ESP32C3
  framework:
    type: esp-idf
    version: recommended

remote_receiver:
  pin: GPIO1
  dump: all
```

### Anything in the logs that might be useful for us?

```txt
[I][app:029]: Running through setup()...
[C][remote_receiver.esp32:013]: Setting up Remote Receiver...
[D][esp-idf:000]: I (357) gpio: GPIO[1]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0

[D][esp-idf:000]: E (364) rmt: rmt_set_pin(544): RMT CHANNEL ERR

[D][esp-idf:000]: E (364) rmt: rmt_config(691): set gpio for RMT driver failed

[E][component:112]: Component remote_receiver was marked as failed.

...

[I][app:062]: setup() finished successfully!
[I][app:102]: ESPHome version 2021.12.3 compiled on Jan 12 2022, 22:11:01

...

[C][remote_receiver.esp32:054]: Remote Receiver:
[C][remote_receiver.esp32:055]:   Pin: GPIO1
[C][remote_receiver.esp32:060]:   Channel: 0
[C][remote_receiver.esp32:061]:   RMT memory blocks: 3
[C][remote_receiver.esp32:062]:   Clock divider: 80
[C][remote_receiver.esp32:063]:   Tolerance: 25%
[C][remote_receiver.esp32:064]:   Filter out pulses shorter than: 50 us
[C][remote_receiver.esp32:065]:   Signal is done after 10000 us of no changes
[E][remote_receiver.esp32:067]: Configuring RMT driver failed: ESP_ERR_INVALID_ARG
```

rpatel3001, 2022-01-14:
I was able to work around this issue by creating 2 dummy transmitters to fill up the TX only channels before the receiver gets a channel allocated.

rpatel3001, 2022-01-14:
I've just found that this workaround doesn't work with the neopixelbus component. The neopixel light is setup before the dummy transmitter regardless of the YAML order, so the dummy transmitter reconfigures the RMT channel.

At a minimum the auto channel selection needs to take the channel restrictions into account, and allowing users to choose which RMT channel to use in the YAML file would be useful.

rpatel3001, 2022-05-19:
not stale

Jorre05, 2022-08-09:
Hi,
Same problem here. Adding 2 dummy transmitters makes the remote_receiver work, but breaks the Neopixel bus.

Jorre05, 2022-08-09:
@rpatel3001 For the time being I've created a quick hack in a separate repository to work around this issue.
https://github.com/Jorre05/remote_receiver

diginfo, 2022-10-11:
I have same problem, this is not stale :-)

DjordjeMandic, 2022-12-04:
I have neopixelbus also and both receiver and transmitter throw same error.

Jorre05, 2022-12-11:
@RENOxDECEPTION Thanks for the update. I've updated the README accordingly

DjordjeMandic, 2022-12-11:
I made short video showcasing the problem. If i find free time i will try to implement something similar in remote_receiver and remote_transmitter like in NeoPixelBus component.

https://youtu.be/gAMdgxzktQk

I really thing that this can be fixed in few days max. I just don't have ESP32-S3 for testing.

nixmaster2000, 2023-01-05:
I think I'm experiencing the same issue with ESP32 (8 RMT channels). The project I'm working on requires 3x IR receivers and up to 5x IR transmitters (I could reduce transmitters if necessary). Using @Jorre05's repo I've got the receivers loading correctly but it's auto-assigning channels 9 - 13 for the transmitters which is obviously failing. Really need to be able to specify the channels for the transmitter component also.

`[20:49:33][C][remote_transmitter:015]: Remote Transmitter...
[20:49:33][C][remote_transmitter:016]:   Channel: 9
[20:49:33][C][remote_transmitter:017]:   RMT memory blocks: 255
[20:49:33][C][remote_transmitter:018]:   Clock divider: 80
[20:49:33][C][remote_transmitter:019]:   Pin: GPIO5
[20:49:33][C][remote_transmitter:022]:     Carrier Duty: 50%
[20:49:33][E][remote_transmitter:026]: Configuring RMT driver failed: ESP_ERR_INVALID_ARG
[20:49:33][C][remote_transmitter:015]: Remote Transmitter...
[20:49:33][C][remote_transmitter:016]:   Channel: 10
[20:49:33][C][remote_transmitter:017]:   RMT memory blocks: 254
[20:49:33][C][remote_transmitter:018]:   Clock divider: 80
[20:49:33][C][remote_transmitter:019]:   Pin: GPIO4
[20:49:33][C][remote_transmitter:022]:     Carrier Duty: 50%
[20:49:33][E][remote_transmitter:026]: Configuring RMT driver failed: ESP_ERR_INVALID_ARG
[20:49:33][C][remote_transmitter:015]: Remote Transmitter...
[20:49:33][C][remote_transmitter:016]:   Channel: 11
[20:49:33][C][remote_transmitter:017]:   RMT memory blocks: 253
[20:49:33][C][remote_transmitter:018]:   Clock divider: 80
[20:49:33][C][remote_transmitter:019]:   Pin: GPIO16
[20:49:33][C][remote_transmitter:022]:     Carrier Duty: 50%
[20:49:33][E][remote_transmitter:026]: Configuring RMT driver failed: ESP_ERR_INVALID_ARG
[20:49:33][C][remote_transmitter:015]: Remote Transmitter...
[20:49:33][C][remote_transmitter:016]:   Channel: 12
[20:49:33][C][remote_transmitter:017]:   RMT memory blocks: 252
[20:49:33][C][remote_transmitter:018]:   Clock divider: 80
[20:49:33][C][remote_transmitter:019]:   Pin: GPIO15
[20:49:33][C][remote_transmitter:022]:     Carrier Duty: 50%
[20:49:33][E][remote_transmitter:026]: Configuring RMT driver failed: ESP_ERR_INVALID_ARG
[20:49:33][C][remote_transmitter:015]: Remote Transmitter...
[20:49:33][C][remote_transmitter:016]:   Channel: 13
[20:49:33][C][remote_transmitter:017]:   RMT memory blocks: 251
[20:49:33][C][remote_transmitter:018]:   Clock divider: 80
[20:49:33][C][remote_transmitter:019]:   Pin: GPIO14
[20:49:33][C][remote_transmitter:022]:     Carrier Duty: 50%
[20:49:33][E][remote_transmitter:026]: Configuring RMT driver failed: ESP_ERR_INVALID_ARG`

cg6, 2023-01-05:
@Nixmaster2000

This issue is very specific to the ESP32-C3 variant, which has only 2
receive and 2 transmit channels. Don't use the custom repo for the original
ESP32

nixmaster2000, 2023-01-05:
I guess I'd like to report that auto channel assignment is also not working properly with the original ESP32 as channels 9 through 13 aren't valid choices either.

zwgmy, 2023-05-28:
how to chose the transmit channels of the esp32? The system auto slect the transmit channels 3, report error

jtenniswood, 2023-06-10:
I used @Jorre05 guide but on an m5 atom s3, it threw an error but I picked channel 4 and it worked perfectly. 
I wish this was fixed in the core platform.
`
external_components:
  - source: github://Jorre05/remote_receiver
    components: [ remote_receiver ]

# IR Receiver
remote_receiver:
  pin:
    number: GPIO1
    inverted: true
  rmt_channel: 4
  dump: all
`

DjordjeMandic, 2023-06-12:
> how to chose the transmit channels of the esp32c3? The system auto slect the transmit channels 3, report error

Make them unavailable by using another components if you have ram/flash/pins to waste or change the order of components in yaml code

baylf2000, 2023-09-06:
> @nixmaster2000 This issue is very specific to the ESP32-C3 variant, which has only 2 receive and 2 transmit channels. Don't use the custom repo for the original ESP32

That is not correct. This issue still exists and also affects ESP-S3 variants, and possibly others.

pbadmin, 2023-09-09:
Wow many thanks!

Using external component from github://Jorre05/remote_receiver worked for me.
I replaced a Tuya CBU module with a esp8685-wroom-06 in a S16 Smart Remote IR Controller and now it's working.

Here's the config just in case someone needs it.

`remote_transmitter:
  pin: GPIO7
  carrier_duty_percent: 30%

remote_receiver:
  id: rcvr
  pin:
    number: GPIO3
    inverted: True
  rmt_channel: 2
  tolerance: 55%
  dump: all

climate:
  - platform: daikin
    name: "$upper_devicename Living AC"
    receiver_id: rcvr`

ferbulous, 2023-11-15:
Thanks @Jorre05 for the fix
@pbadmin can your device boot up if you're using esp-idf framework? mine didn't

pbadmin, 2023-11-15:
> Thanks @Jorre05 for the fix @pbadmin can your device boot up if you're using esp-idf framework? mine didn't

@ferbulous I'm using ESPhome plugin from HASSIO (HomeAssistant OS)

predam, 2024-01-13:
Thanks @Jorre05 
I also needed this feature for `remote_transmitter` If anyone else needs it, it is available here:
https://github.com/predam/remote_transmitter

NeoMatrixJR, 2024-01-19:
The Jorre05 workaround seems to have broken on_XXXX calls from the device for me (for IR Receive anyway).  Anyone else experiencing this?  I have an on_magiquest callout to a homeassistant service I was trying to set up for my kids after gong to Great Wolf Lodge.  I have this working on an ESP8266 device (or did....haven't tested recently) but M5Atoms don't work.  I'll get the ir magiquest output in the logs, but no event call.

easybe, 2024-04-08:
@Jorre05 have you tried upstreaming the fix? Is there a PR?

Jorre05, 2024-04-09:
> @Jorre05, @predam have you tried upstreaming the fixes? Is there a PR?

No, I did a quick and dirty fix to get my receiver working. A proper implementation that works for other variants is a lot more work. My fix may even break other things that I'm not aware of.

Jorre05, 2024-04-17:
The latest ESPHome release contains a PR ( https://github.com/esphome/esphome/pull/6497 ) that seems to fix this issue for me.

rpatel3001, 2024-04-23:
closed by https://github.com/esphome/esphome/pull/6497

closed 2024-04-23 by rpatel3001