esphome/issues #2935: ESP8288 hardware UART not receiving data

escoand, 2022-01-13:
### The problem

I tried to use the first hardware UART on ESP8266 but not receiving any data. Using the same configuration with any other GPIO is working without problems.

### Which version of ESPHome has the issue?

2021.12.3

### What type of installation are you using?

Docker

### What platform are you using?

ESP8266

### Board

Wemos D1 mini

### Component causing the issue

uart

### Example YAML snippet

```yaml
esphome:
  platform: ESP8266
  board: d1_mini

logger:
  hardware_uart: UART1

uart:
- id: uart1   # not working
  rx_pin: RX
  baud_rate: 4800
  data_bits: 8
  stop_bits: 1
  parity: ODD
- id: uart2    # working
  rx_pin: D1
  baud_rate: 4800
  data_bits: 8
  stop_bits: 1
  parity: ODD
```

martgras, 2022-01-26:
Please see https://esphome.io/components/logger.html?highlight=logger#hardware-uarts

HW Uart1 can only transmit
`UART1 - TX: GPIO2, RX: None (Only on ESP8266)`

escoand, 2022-01-26:
@martgras I know, the here used `id: uart1` is just a string and has nothing to do with the HW UART1.

martgras, 2022-01-26:
According to the logs this is HW Serial because RX == GPIO3
```
[15:48:51][C][logger:233]: Logger:
[15:48:51][C][logger:234]:   Level: DEBUG
[15:48:51][C][logger:235]:   Log Baud Rate: 115200
[15:48:51][C][logger:236]:   Hardware UART: UART1
[15:48:51][C][uart.arduino_esp8266:102]: UART Bus:
[15:48:51][C][uart.arduino_esp8266:104]:   RX Pin: GPIO3
[15:48:51][C][uart.arduino_esp8266:106]:   RX Buffer Size: 256
[15:48:51][C][uart.arduino_esp8266:108]:   Baud Rate: 4800 baud
[15:48:51][C][uart.arduino_esp8266:109]:   Data Bits: 8
[15:48:51][C][uart.arduino_esp8266:110]:   Parity: ODD
[15:48:51][C][uart.arduino_esp8266:111]:   Stop bits: 1
[15:48:51][C][uart.arduino_esp8266:113]:   Using hardware serial interface. <==
[15:48:51][C][uart.arduino_esp8266:102]: UART Bus:
[15:48:51][C][uart.arduino_esp8266:104]:   RX Pin: GPIO5
[15:48:51][C][uart.arduino_esp8266:106]:   RX Buffer Size: 256
[15:48:51][C][uart.arduino_esp8266:108]:   Baud Rate: 4800 baud
[15:48:51][C][uart.arduino_esp8266:109]:   Data Bits: 8
[15:48:51][C][uart.arduino_esp8266:110]:   Parity: ODD
[15:48:51][C][uart.arduino_esp8266:111]:   Stop bits: 1
[15:48:51][C][uart.arduino_esp8266:115]:   Using software serial
```

If I'm not misunderstanding  the code  defining  `tx_pin` to anything except GPIO1 will force software serial

martgras, 2022-01-26:
Sorry I mixed things up uart1 maps to HW Serial 0 and your pin definition is correct

Samoth47, 2023-12-05:
Answering to an old issue, but it could be caused by a very high impedance output from your UART peripheral.
Wemos D1 mini boards have a resistor between the CH340G USB to UART chip, and the RX and TX pins. The CH340G is apparently always pulling the pin high with a very small impedance. Even with the usual 470 ohms resistor, it could be enough to completely overcome the small signals put out by your peripheral.
A temporary fix that successfully fixed the issue for me was to remove this resistor, cutting communication through the USB port, but removing the parasitic strong pull-up. I just soldered it on one of the pad to put it back if I need to.

For context, my UART peripheral is a CurrentCost screen. I'm trying to hack the UART port on the back to "smartify" it and get the real power in realtime in the HA UI.
The TX impedance from the CurrentCost screen is incredibly high, roughly 10 k ohms.

[Source Arduino forum](https://forum.arduino.cc/t/cant-use-hardware-serial-on-wemos-d1-mini/575304/8)