esphome/issues #2938: I2C device with address 0x05 not detected

stephan11111, 2022-01-14:
### The problem

I made some component that communicates with ATTINY84 with I2C. The Tiny uses 0x05 as address. That worked well with the latest ESPHome version in June/July 2021.

When I tried to use the same setting with version 2021.12.0, the I2C device got not detected. (no devices found). In very verbose mode, I noticed that the address range of 0x05 did not get scanned - I think there was no address below 0x8 listed.

After changing the address to 0x12, the component worked like it should and got detected on ESPHome start.

I don't know the details on I2C protocol so sorry in case that behaviors is intended. But it looks like I2C devices below 0x8 can't be used with ESPHome.
For reference, the communication between a PI and the Tiny worked well with address 0x05.

Hope that helps and let me know if additional information is required. 
Thank you for this amazing project!

### Which version of ESPHome has the issue?

2021.12.0

### What type of installation are you using?

pip

### What platform are you using?

ESP32

### Board

nodemcu-32s

### Component causing the issue

I2C

### Example YAML snippet

```yaml
i2c:
  - id: bus_a
    sda: 32
    scl: 33
    scan: True
  - id: bus_b # ATTINY was connected to bus_b
    sda: 19
    scl: 21
    scan: True
    frequency: 10kHz
```

### Anything in the logs that might be useful for us?

```txt
[08:49:31][I][i2c.arduino:048]: Scanning i2c bus for active devices...
[08:49:31][VV][i2c.arduino:121]: 0x08 TX
[08:49:31][VV][i2c.arduino:145]: TX failed: not acknowledged
[08:49:31][VV][i2c.arduino:121]: 0x09 TX
[08:49:31][VV][i2c.arduino:145]: TX failed: not acknowledged
[08:49:31][VV][i2c.arduino:121]: 0x0A TX
[08:49:31][VV][i2c.arduino:145]: TX failed: not acknowledged
[08:49:31][VV][i2c.arduino:121]: 0x0B TX
[08:49:31][VV][i2c.arduino:145]: TX failed: not acknowledged
[08:49:31][VV][i2c.arduino:145]: TX failed: not acknowledged
[08:49:31][VV][i2c.arduino:121]: 0x0D TX
[08:49:31][VV][i2c.arduino:145]: TX failed: not acknowledged
[08:49:31][VV][i2c.arduino:121]: 0x0E TX
[08:49:31][VV][i2c.arduino:145]: TX failed: not acknowledged
[08:49:31][VV][i2c.arduino:121]: 0x0F TX
[08:49:31][VV][i2c.arduino:145]: TX failed: not acknowledged
[08:49:31][VV][i2c.arduino:121]: 0x10 TX
[08:49:31][VV][i2c.arduino:145]: TX failed: not acknowledged
[08:49:31][VV][i2c.arduino:121]: 0x14 TX
[08:49:31][VV][i2c.arduino:145]: TX failed: not acknowledged
[08:49:31][VV][i2c.arduino:121]: 0x15 TX
[08:49:31][VV][i2c.arduino:145]: TX failed: not acknowledged
[08:49:31][VV][i2c.arduino:121]: 0x16 TX
[08:49:31][VV][i2c.arduino:145]: TX failed: not acknowledged
[08:49:31][VV][i2c.arduino:121]: 0x17 TX
[08:49:31][VV][i2c.arduino:145]: TX failed: not acknowledged
[08:49:31][VV][i2c.arduino:121]: 0x18 TX
[08:49:31][VV][i2c.arduino:145]: TX failed: not acknowledged
[08:49:31][VV][i2c.arduino:121]: 0x19 TX
[08:49:31][VV][i2c.arduino:145]: TX failed: not acknowledged
[08:49:31][VV][i2c.arduino:121]: 0x1A TX
[08:49:31][VV][i2c.arduino:145]: TX failed: not acknowledged
[08:49:31][VV][i2c.arduino:145]: TX failed: not acknowledged
[08:49:31][VV][i2c.arduino:121]: 0x23 TX
[08:49:31][VV][i2c.arduino:145]: TX failed: not acknowledged
[08:49:31][VV][i2c.arduino:121]: 0x24 TX
[08:49:31][VV][i2c.arduino:145]: TX failed: not acknowledged
[08:49:31][VV][i2c.arduino:121]: 0x25 TX
[08:49:31][VV][i2c.arduino:145]: TX failed: not acknowledged
[08:49:31][VV][i2c.arduino:121]: 0x26 TX
[08:49:31][VV][i2c.arduino:145]: TX failed: not acknowledged
[08:49:31][VV][i2c.arduino:121]: 0x27 TX
[08:49:31][VV][i2c.arduino:145]: TX failed: not acknowledged
[08:49:31][VV][i2c.arduino:121]: 0x2E TX
[08:49:31][VV][i2c.arduino:145]: TX failed: not acknowledged
[08:49:31][VV][i2c.arduino:121]: 0x2F TX
[08:49:31][VV][i2c.arduino:145]: TX failed: not acknowledged
[08:49:31][VV][i2c.arduino:121]: 0x30 TX
[08:49:31][VV][i2c.arduino:145]: TX failed: not acknowledged
[08:49:31][VV][i2c.arduino:121]: 0x31 TX
[08:49:31][VV][i2c.arduino:145]: TX failed: not acknowledged
[08:49:31][VV][i2c.arduino:145]: TX failed: not acknowledged
[08:49:31][VV][i2c.arduino:121]: 0x39 TX
[08:49:31][VV][i2c.arduino:145]: TX failed: not acknowledged
[08:49:31][VV][i2c.arduino:121]: 0x3A TX
[08:49:31][VV][i2c.arduino:145]: TX failed: not acknowledged
[08:49:31][VV][i2c.arduino:121]: 0x3B TX
[08:49:31][VV][i2c.arduino:145]: TX failed: not acknowledged
[08:49:31][VV][i2c.arduino:121]: 0x3C TX
[08:49:31][VV][i2c.arduino:145]: TX failed: not acknowledged
[08:49:31][VV][i2c.arduino:121]: 0x3D TX
[08:49:31][VV][i2c.arduino:145]: TX failed: not acknowledged
[08:49:31][VV][i2c.arduino:121]: 0x3E TX
[08:49:31][VV][i2c.arduino:145]: TX failed: not acknowledged
[08:49:31][VV][i2c.arduino:121]: 0x3F TX
[08:49:31][VV][i2c.arduino:145]: TX failed: not acknowledged
[08:49:31][VV][i2c.arduino:121]: 0x40 TX
[08:49:31][VV][i2c.arduino:145]: TX failed: not acknowledged
[08:49:31][VV][i2c.arduino:121]: 0x41 TX
[08:49:31][VV][i2c.arduino:145]: TX failed: not acknowledged
[08:49:31][VV][i2c.arduino:121]: 0x42 TX
[08:49:31][VV][i2c.arduino:145]: TX failed: not acknowledged
[08:49:31][VV][i2c.arduino:121]: 0x43 TX
[08:49:31][VV][i2c.arduino:145]: TX failed: not acknowledged
[08:49:31][VV][i2c.arduino:121]: 0x44 TX
[08:49:31][VV][i2c.arduino:145]: TX failed: not acknowledged
[08:49:31][VV][i2c.arduino:121]: 0x45 TX
[08:49:31][VV][i2c.arduino:145]: TX failed: not acknowledged
[08:49:31][VV][i2c.arduino:121]: 0x46 TX
[08:49:31][VV][i2c.arduino:145]: TX failed: not acknowledged
[08:49:31][VV][i2c.arduino:121]: 0x47 TX
[08:49:31][VV][i2c.arduino:145]: TX failed: not acknowledged
[08:49:31][VV][i2c.arduino:121]: 0x48 TX
[08:49:31][VV][i2c.arduino:145]: TX failed: not acknowledged
[08:49:31][VV][i2c.arduino:121]: 0x4F TX
[08:49:31][VV][i2c.arduino:145]: TX failed: not acknowledged
[08:49:31][VV][i2c.arduino:121]: 0x50 TX
[08:49:31][VV][i2c.arduino:145]: TX failed: not acknowledged
[08:49:31][VV][i2c.arduino:121]: 0x51 TX
[08:49:31][VV][i2c.arduino:145]: TX failed: not acknowledged
[08:49:31][VV][i2c.arduino:121]: 0x52 TX
[08:49:31][VV][i2c.arduino:145]: TX failed: not acknowledged
[08:49:31][VV][i2c.arduino:121]: 0x53 TX
[08:49:31][VV][i2c.arduino:145]: TX failed: not acknowledged
[08:49:31][VV][i2c.arduino:121]: 0x54 TX
[08:49:31][VV][i2c.arduino:145]: TX failed: not acknowledged
[08:49:31][VV][i2c.arduino:121]: 0x55 TX
[08:49:31][VV][i2c.arduino:145]: TX failed: not acknowledged
[08:49:31][VV][i2c.arduino:145]: TX failed: not acknowledged
[08:49:31][VV][i2c.arduino:121]: 0x5A TX
[08:49:31][VV][i2c.arduino:145]: TX failed: not acknowledged
[08:49:31][VV][i2c.arduino:121]: 0x5B TX
[08:49:31][VV][i2c.arduino:145]: TX failed: not acknowledged
[08:49:31][VV][i2c.arduino:121]: 0x5C TX
[08:49:31][VV][i2c.arduino:145]: TX failed: not acknowledged
[08:49:31][VV][i2c.arduino:121]: 0x5D TX
[08:49:31][VV][i2c.arduino:145]: TX failed: not acknowledged
[08:49:31][VV][i2c.arduino:121]: 0x5E TX
[08:49:31][VV][i2c.arduino:145]: TX failed: not acknowledged
[08:49:31][VV][i2c.arduino:121]: 0x5F TX
[08:49:31][VV][i2c.arduino:145]: TX failed: not acknowledged
[08:49:31][VV][i2c.arduino:121]: 0x60 TX
[08:49:31][VV][i2c.arduino:145]: TX failed: not acknowledged
[08:49:31][VV][i2c.arduino:121]: 0x61 TX
[08:49:31][VV][i2c.arduino:145]: TX failed: not acknowledged
[08:49:31][VV][i2c.arduino:121]: 0x62 TX
[08:49:31][VV][i2c.arduino:145]: TX failed: not acknowledged
[08:49:31][VV][i2c.arduino:121]: 0x63 TX
[08:49:31][VV][i2c.arduino:145]: TX failed: not acknowledged
[08:49:31][VV][i2c.arduino:121]: 0x64 TX
[08:49:31][VV][i2c.arduino:145]: TX failed: not acknowledged
[08:49:31][VV][i2c.arduino:121]: 0x65 TX
[08:49:31][VV][i2c.arduino:145]: TX failed: not acknowledged
[08:49:31][VV][i2c.arduino:121]: 0x66 TX
[08:49:31][VV][i2c.arduino:145]: TX failed: not acknowledged
[08:49:31][VV][i2c.arduino:121]: 0x67 TX
[08:49:31][VV][i2c.arduino:145]: TX failed: not acknowledged
[08:49:31][VV][i2c.arduino:121]: 0x68 TX
[08:49:31][VV][i2c.arduino:145]: TX failed: not acknowledged
[08:49:31][VV][i2c.arduino:121]: 0x69 TX
[08:49:31][VV][i2c.arduino:145]: TX failed: not acknowledged
[08:49:31][VV][i2c.arduino:121]: 0x6A TX
[08:49:31][VV][i2c.arduino:145]: TX failed: not acknowledged
[08:49:31][VV][i2c.arduino:121]: 0x6B TX
[08:49:31][VV][i2c.arduino:145]: TX failed: not acknowledged
[08:49:31][VV][i2c.arduino:121]: 0x6C TX
[08:49:31][VV][i2c.arduino:145]: TX failed: not acknowledged
[08:49:31][VV][i2c.arduino:121]: 0x6D TX
[08:49:32][VV][i2c.arduino:145]: TX failed: not acknowledged
[08:49:32][VV][i2c.arduino:121]: 0x6E TX
[08:49:32][VV][i2c.arduino:145]: TX failed: not acknowledged
[08:49:32][VV][i2c.arduino:121]: 0x6F TX
[08:49:32][VV][i2c.arduino:145]: TX failed: not acknowledged
[08:49:32][VV][i2c.arduino:121]: 0x70 TX
[08:49:32][VV][i2c.arduino:145]: TX failed: not acknowledged
[08:49:32][VV][i2c.arduino:121]: 0x71 TX
[08:49:32][VV][i2c.arduino:145]: TX failed: not acknowledged
```

martgras, 2022-01-14:
The i2c adresses 0-7 are reserved.
5 is used for High Speed mode (not used in esphome afaik) 
See https://www.i2c-bus.org/addressing/

OttoWinter, 2022-01-18:
As @martgras mentioned, the address is reserved, so we don't scan for them. This is the right behavior, as no device should be using these as addresses.

closed 2022-01-18 by OttoWinter