esphome/issues #2940: MODBUS bus writing to register

worker72, 2022-01-14:
### The problem

I've defined 2 sequential registers 0x00D2 and 0x00D3 as modbus numbers. 
These numbers are succesfully read and communicated to HA, so I see 2 fields that I can edit.
When I change in HA the number of address 0xD2, the number is succesfully written to the modbus device. 
But when I change the number that corresponds to 0xD3, this number is NOT succesfully written to the modbus device.

I have changed the logging to VERY_VERBOSE and I see the cause: ESPhome is writing to address 0xD4.
[image]

Because I use 2 addresses that are sequential, the modbus library combines reading and writing to those addresses. I think that is why the screenshot says 0xD2 in the cyan line. But I think there is a bug in the code that causes ESPhome to address 0xD4 where it should address 0xD3

In the log snippet below, I first changed in HA the number that corresponds to address 0xD2, and right after I changed the number that corresponds to 0xD3. In the log you see it first writes to 0xD2 (correct) and then to 0xD4 (not correct)

### Which version of ESPHome has the issue?

2021.12.3

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2021.12.3

### What platform are you using?

ESP8266

### Board

nodemcuv3

### Component causing the issue

modbus

### Example YAML snippet

```yaml
number:
  - platform: modbus_controller
    modbus_controller_id: deye
    id: max_charge
    name: "max_charge"
    address: 0x00D2
    use_write_multiple: true
    value_type: U_WORD
  - platform: modbus_controller
    modbus_controller_id: deye
    id: max_discharge
    name: "max_discharge"
    address: 0x00D3
    use_write_multiple: true
    value_type: U_WORD
```

### Anything in the logs that might be useful for us?

```txt
[23:59:06][D][number:010]: 'max_charge' - Setting
[23:59:06][D][number:029]:   Value: 30.000000
[23:59:06][D][modbus.number:055]: Updating register: connected Sensor=max_charge start address=0xD2 register count=1 new value=30.00 (val=30.00)
[23:59:06][V][modbus_controller:028]: Sending next modbus command to device 1 register 0xD2 count 1
[23:59:06][VV][uart.arduino_esp8266:178]:     Flushing...
[23:59:06][V][modbus:184]: Modbus write: 01.10.00.D2.00.01.02.00.1E.34.2A (11)
[23:59:06][V][modbus_controller:447]: Command sent 16 0xD2 1
[23:59:06][V][modbus:058]: Modbus received Byte  1 (0X1)
[23:59:06][V][modbus:058]: Modbus received Byte  16 (0X10)
[23:59:06][V][modbus:058]: Modbus received Byte  0 (0X0)
[23:59:06][V][modbus:058]: Modbus received Byte  210 (0Xd2)
[23:59:06][V][modbus:058]: Modbus received Byte  0 (0X0)
[23:59:06][V][modbus:058]: Modbus received Byte  1 (0X1)
[23:59:06][V][modbus:058]: Modbus received Byte  161 (0Xa1)
[23:59:06][V][modbus:058]: Modbus received Byte  240 (0Xf0)
[23:59:06][V][modbus_controller:048]: Modbus response queued
[23:59:06][V][modbus_controller:056]: Process modbus response for address 0xD2 size: 4
[23:59:06][V][modbus_controller:294]: Command ACK 0xD2 -11776 
[23:59:06][D][number:036]: 'max_charge': Sending state 30.000000
[23:59:06][VV][api.service:260]: send_number_state_response: NumberStateResponse {
  key: 2265868028
  state: 30
  missing_state: NO
}
[23:59:08][VV][api.service:512]: on_number_command_request: NumberCommandRequest {
  key: 3220244840
  state: 29
}
[23:59:08][D][number:010]: 'max_discharge' - Setting
[23:59:08][D][number:029]:   Value: 29.000000
[23:59:08][D][modbus.number:055]: Updating register: connected Sensor=max_discharge start address=0xD2 register count=1 new value=29.00 (val=29.00)
[23:59:08][V][modbus_controller:028]: Sending next modbus command to device 1 register 0xD4 count 1
[23:59:08][VV][uart.arduino_esp8266:178]:     Flushing...
[23:59:08][V][modbus:184]: Modbus write: 01.10.00.D4.00.01.02.00.1D.74.4D (11)
[23:59:08][V][modbus_controller:447]: Command sent 16 0xD4 1
[23:59:08][V][modbus:058]: Modbus received Byte  1 (0X1)
[23:59:08][V][modbus:058]: Modbus received Byte  16 (0X10)
[23:59:08][V][modbus:058]: Modbus received Byte  0 (0X0)
[23:59:08][V][modbus:058]: Modbus received Byte  212 (0Xd4)
[23:59:08][V][modbus:058]: Modbus received Byte  0 (0X0)
[23:59:08][V][modbus:058]: Modbus received Byte  1 (0X1)
[23:59:08][V][modbus:058]: Modbus received Byte  65 (0X41)
[23:59:08][V][modbus:058]: Modbus received Byte  241 (0Xf1)
[23:59:08][V][modbus_controller:048]: Modbus response queued
[23:59:08][V][modbus_controller:056]: Process modbus response for address 0xD4 size: 4
[23:59:08][V][modbus_controller:294]: Command ACK 0xD4 -11264 
[23:59:08][D][number:036]: 'max_discharge': Sending state 29.000000
[23:59:08][VV][api.service:260]: send_number_state_response: NumberStateResponse {
  key: 3220244840
  state: 29
  missing_state: NO
}
```

martgras, 2022-01-15:
This should be fixed in https://github.com/esphome/esphome/pull/3031
You can test if the fix works for you with 
```yaml
external_components:
  - source: github://pr#3031
    components: ["modbus_controller"]
```

worker72, 2022-01-16:
I get error when trying that. 
I copied your text (external components: ...) into the yaml file of my ESPnode, is that correct?

[image]

martgras, 2022-01-16:
The problem is that you also need https://github.com/esphome/esphome/pull/2882 but since it is core function  external_component doesn't work. (format_hex_pretty was implemented in that commit)
So the dev branch of esphome is required for testing

worker72, 2022-01-16:
Right. 
I can live with the current situation, so I can wait until this PR ends up in an upgrade. Any idea when it will be released into a stable release of ESPhome?

And may I ask an unrelated question? 
By default the modbus sensors that I defined in de yaml files are updated every minute or so. There is one specific register that I want to update more often. Can I set the update frequency, and can this be done per sensor? (I read somewhere thatI can set skip_updates, so maybe that's the way to make all other sensors slower)

martgras, 2022-01-16:
It's included in beta already and should be in the next release

worker72, 2022-01-16:
thanks a lot

closed 2022-01-16 by worker72